gyroflow/gyroflow #643: Add core library usage example

ralfbiedert, 2023-05-15:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

First of all, thank you for Gyroflow! I've been using it for a while now and it works really well. That it's written in Rust makes it even better :)

I recently tried integrating the `gyroflow-core` library into a project of mine to batch process videos along with other sensor data, but I've been struggling to identify the "absolute MVP" I'd need to use to get started. The documentation mentions the core library, but then just says to read the frontend / CLI code.

Attempting that I ended up with a bunch of questions like:

- Do i really need the `StabilizationManager`, or is that more a UI helper?
- If not, what structs / components should I use?
- The CLI code path seems to rely on Qt code, (for example I can't get a `RenderQueue` without it) which in turns seems to hold the decoder(?) What's the preferred way to get a decoder using only core?
- Similarly, what's the preferred way to get `Buffers`? 
- Where would I get information like for `init_from_video_data` best get from?

I think it would be nice to have an `examples/basic.rs`, only relying on `gyroflow-core`, that shows how to process a single video in the most basic way. From there it would be much easier to extrapolate how to get started, or add more functionality.

AdrianEddy, 2023-05-15:
The core purposely excludes ffmpeg and thus decoding and encoding of data. This also includes render queue, as render queue without a decoder and encoder doesn't make sense.

Examples of pure core usage are here:
https://github.com/gyroflow/gyroflow-ofx/blob/main/src/gyroflow.rs
https://github.com/latenitefilms/GyroflowToolbox/blob/main/Source/Frameworks/gyroflow/src/lib.rs

Some documentation/info is here: https://docs.gyroflow.xyz/app/technical-details/gyroflow-core

- `StabilizationManager` is the core struct of the gyroflow-core and it's needed and is the main interface of the gyroflow-core
- CLI includes Qt and ffmpeg so it's not really core-only

What's your use case? If you need ffmpeg and render queue, then why not just use CLI?

> I think it would be nice to have an examples/basic.rs, only relying on gyroflow-core, that shows how to process a single video in the most basic way.
That's not really the purpose of core. Note that `gyroflow-core` is not `gyroflow-cli`, but rather "pixels" processing library. Read more on the architecture in the docs link I provided

AdrianEddy, 2023-09-21:
I think this can be closed, the documentation describes the usage of core sufficiently. While there isn't a ready to use "basic" example of pure core usage, writing one is not something I plan doing and the current docs + plugins code is enough to get an idea of how to use it

closed 2023-09-21 by AdrianEddy