gyroflow/gyroflow #644: imu_integration: prev_time issue

Lydia43, 2023-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

/

### What operating system are you using?

/

### What GPU are you using?

/

### What happened?

there is a issue confused me in the first integral time.
current code:
let sample_time_ms = duration_ms / 1000.0 / imu_data.len() as f64;
let mut prev_time = imu_data[0].timestamp_ms - sample_time_ms;
let dt = (v.timestamp_ms - prev_time) / 1000.0;

the unit of imu_data[0].timestamp_ms is 'ms';
the unit of sample_time_ms is 's';
the unit of dt will not be 's'.

please check it.

AdrianEddy, 2023-05-17:
Indeed, good catch! fixed now
Although it only affects first sample, because `prev_time` is overwritten in the loop so it didn't cause much trouble

closed 2023-05-17 by AdrianEddy

Lydia43, 2023-05-18:
There's one more problem in the integration process.
The integral process is "quat(n)=quat(n-1)*delta_quat". As we understand, the new rotation state should be rotated by the original state, so it should be 'delta_quat*quat(n-1)'.
I look forward to your reply!

AdrianEddy, 2023-05-18:
I don't think so, the "current state" should be rotated by "delta state", just like it is implemented now.
It doesn't work if I switch it

Lydia43, 2023-05-18:
Thanks for your reply!
Currently, the integral process is right-multiplying 'quat_delta' from intial state.
Shouldn't the estimated trajectory represent a change from the initial state to the current state?
Why the process is 'quat_current=quat_intial*quat_delta" rather than "quat_current=quat_delta*quat_initial"?

'eq0： quat_current = quat_delta * quat_pre '
By definition, "eq0" means that the rotation of "quat_pre" to "quat_current" can be achieved by multiplying "quat_delta" left.
So the current timestamp state can also be obtained by left-multiplying "quat_delta" from the previous timestamp state.
'quat_current=quat_delta*quat_pre'

If there are any misunderstandings, please let me know, thank you very much!

AdrianEddy, 2023-05-18:
The process is `quat_current = quat_current * quat_delta`, not `quat_current = quat_intial * quat_delta`
I'm not sure I understand what exactly you're confused about. @ElvinC do you have any insight?

ElvinC, 2023-05-18:
Been a while since I looked at this, but iirc the reason is that the quat_delta based on the gyroscope is given in the local body frame instead of the global frame, so the multiplication order is reversed