gyroflow/gyroflow #652: Behaviour when preset has not "Output path"defined

Fedess1986, 2023-05-25:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

When "Output path" is not checked when creating a preset, at the moment of applying that preset, processed files go to the latest used path.

If not output path is defined, could you add "ask when exporting" or give an advice to manually set the path? This would be more useful when multiple files are in the queue.

AdrianEddy, 2023-06-03:
Could you upload that preset and write steps to reproduce? This doesn't happen to me.
When preset doesn't have output path, applying it in the queue doesn't change any paths.

Maybe you're talking about dropping video files directly into render queue? If yes, then it works like this:
- If there's no file opened in main UI - the output path will be the same as the dropped file
- If there's any file opened in the main UI and "Output path" is set at the bottom - the output path of the dropped file will be the one used in the main window input field.

Fedess1986, 2023-06-05:
Yes, I was talking about dropping video files directly into render queue.

Tryed what you said and it is ok then. 
I though that render queue will override everything seen in the main UI, but I got it now.

A question related to this: Is there a way to "close" an opened file in the main UI? My actual workaround is to close and reopen Gyroflow.

AdrianEddy, 2023-06-05:
You can simply clear the "Output path" at the bottom, just make it empty, and it will not affect the render queue

Fedess1986, 2023-06-05:
That's going to be the way then, thanks!

AdrianEddy, 2023-06-11:
Added in 5a6ad6d96b1b6ac14aead3b5ef9269c0dd3d1838, it will now ask for the output location when dropping files into the render queue

closed 2023-06-11 by AdrianEddy