gyroflow/gyroflow #653: translations in dropdown menues are not shown (worked in previous versions)

3dkopfkino, 2023-05-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.0 (gh1232)

### What operating system are you using?

Windows 10 22H2

### What happened?

[image]
when expanding these menus, the english text is shown.
after selection, the field value is the translated text

AdrianEddy, 2023-05-27:
Not all translations are finished, if you want to help, they are on crodwin https://crowdin.com/project/gyroflow

closed 2023-05-27 by AdrianEddy

3dkopfkino, 2023-05-27:
these texts are included in resources/translations/de.ts and have been shown in previous builds
[image]

AdrianEddy, 2023-05-27:
You're right, sorry. Fixed on latest commit