gyroflow/gyroflow #656: Add "Max sync points every n-th seconds"

Fedess1986, 2023-05-29:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Instead of max sync points, It could be better to have max sync points every X time of video (for example, every 10 seconds)

I feel the longer the clip, more sync points are needed.

Perhaps, it's an issue of the gear I'm using (Gopro hero5 black). Anyway... if that's the case, the option would be really useful.

AdrianEddy, 2023-05-29:
I won't add this, it shouldn't be normally needed and there's already too much stuff in the UI. If you really need this you can use `custom_sync_pattern` like 
```json
"custom_sync_pattern": [
  {
      "start": "0",
      "interval": "10s"
  }
]
```
it will add a sync point every 10 seconds

You can add this to your preset or lens profile (in `"sync_settings"` section). For an example, see [here](https://github.com/gyroflow/gyroflow/blob/master/resources/camera_presets/RunCam/RunCam_5_Orange_1080p60XV_1920x1080.json#L61-L80)

closed 2023-05-29 by AdrianEddy

Fedess1986, 2023-05-29:
Thank you! I didn't know that, and it's really enough for what I need. It is great that those custom settings could be added. I would like to know more about this. 

And sorry I couldn't find this by myself, I searched in the docs, but found a lot of "to do" writing in the "Synchronization" section. I know and appreciate a lot how you put a all the effort and time required on this project.