gyroflow/gyroflow #657: RED SDK not downloading in Gyroflow 1.5.0

deafdirector, 2023-06-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.0

### What operating system are you using?

MacOS Ventura 13.0.1

### What GPU are you using?

Radeon Pr 580 8GB

### What happened?

I am trying to create lens profiles for my Red Komodo (firmware 1.7.5), but when I begin the process it tells me that it requires an external SDK (RED SDK). I am asked if I want to download, I click YES and the RED SDK begins downloading. The download is fine until about 50% and then I get a repeated failure to download notice. The failure notice is a big redd x in a circle and the says: "failed to unpack'/Volumes/Gyroflow v1.5.0/Gyrflow.app/Contents/MacOs/../Fraemworks/._REDDecoder.dylib'

I have reinstalled GyroFlow, shut the Program down and restarted, and even rebooted my computer. Nothing I do allows me to download the RED SDK. 

In addition to the in-program download, I have downloaded a RED SDK (RED SDK 8.3.1) From the RED Website, but I do not know if this is correct or how to get GyroFlow to read this. 

Is there a way to get the RED SDK download into gyroflow?

AdrianEddy, 2023-06-05:
You have to copy Gyroflow somewhere first, as the Volume is read-only.
Copy Gyroflow to Applications and then run it

closed 2023-06-05 by AdrianEddy

deafdirector, 2023-06-05:
Copying GyroFlow to Applications was the first thing I did. It did not work

AdrianEddy, 2023-06-05:
well the message clearly says that you are running it from the Volume, not Applications, please double check that 
"failed to unpack' **/Volumes/Gyroflow v1.5.0/`**...

AdrianEddy, 2023-06-07:
So you're opening R3D file and it says the file is unsupported? 
Please try latest dev build: https://gyroflow.xyz/devbuild/?autodownload

If it still happens there, please send me gyroflow.log file (it is created after you close the program, in Show package Contents -> Contents -> MacOS folder)