gyroflow/gyroflow #658: Dynamic Smothness - zoom to "not more than XX%"

temp-64GTX, 2023-06-08:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Good morning. Me again.

I have some idea. Let me explain.
For example - i have a footage. And i set this smoothing parameters for it:

[image]

Ok. As we can see below - footage have a moment, when camera making 90-degrees turn, it visible at Yaw-axis graphic, and we also see that zoom rises at this range.
So that - on a quiet-footage-part we have Zoom 65%, and on a 90-degrees-turn-part we have Zoom 182%.

[image]

[image]

Of course i can enable keyframing, and leave smoothness=1 on a quiet-parts, and lower it to 0.01 on a turn-part. In that way we will align zoom-graphic to more or less horizontal line. Zoom 65% on a quiet-part, and Zoom 72% on a turn-part. But when i do it "by hands" - it will be so much inaccurate, messy, and takes much of time. 

[image]

In previous screenshots was just one peak. But one footage can have dozens of peaks with different hight, and it can be dozens of that footages.

[image]

Bottom line. My main idea is to make an option to "set zoom - no more than XXX%, and program will automatically adjust Smoothness parameter". For example - set soom to "not more than 75%". And Gyroflow will do almost same as i did with that keyframes, but muuuuuch more accurate, with much more iterations, and much more smooth. Yes, it will need to do some calculations. But  that's what computers are made for - make calculations. 

[image]

So. In that way - complex footages will look suuuuuper smooth on low bumps. And zoom will not goes to extreme 999% values on a high bumps and sharp turns. I think it will be a great feature.

AdrianEddy, 2023-06-11:
If you reset everything to default and use just the "Max smoothness at high velocity" until Max zoom displays your desired max zoom level, it will be exactly what you described. I understand that you want to input max zoom level directly but using just this one slider will be equivalent

relaxxgit, 2023-08-10:
+1 for this
i always have this situation and Max smoothness at high velocity don't do exactly what we want.
Please support it.

MR444, 2023-08-15:
+2 for this
I also wish to have that function.

temp-64GTX, 2023-08-15:
Yep, i think it can be a great killer-feature. One-button solution. 
And it will be no longer need to create this amount of keyframes:

[image]

MR444, 2023-08-18:
> "Max smoothness at high velocity"

Where can I find that setting?

AdrianEddy, 2023-08-18:
Expand the "Advanced" section
[image]

relaxxgit, 2023-08-19:
> 
Yep, but
"It works only for biggest peak" - as temp wrote.

millerk310, 2023-10-13:
this is similar to a feature I've wanted... Minimum Zoom. Or max zoom ratio. If it has to zoom in a lot on a 4k video to stabilize that's fine, but I don't want it zooming ALL the way out, then ALL the way back in for shaky sections. Seems like there should be a way I can just zoom in the whole video so it never goes wider than that. If I change FOV it changes the whole zoom meaning instead of zooming to 150% at max, now it will zoom to 200% at max. Isn't there a way to just set the minimum zoom or lock it at the max zoom, so it doesn't zoom in and out at all? Like if it needs to zoom to 150% then just leave the entire video length at that zoom level. Seems like I must be just not finding the feature?? If so, can someone please let me know?

AdrianEddy, 2023-10-13:
> Isn't there a way to just set the minimum zoom or lock it at the max zoom, so it doesn't zoom in and out at all

That option is called Static zoom

AdrianEddy, 2024-07-05:
Zoom limit is implemented in latest dev build. It defaults to 140%
The limit is approximated so it won't always be exact. If you need more accuracy, you can increase the number of iterations (but it will make it slower)

Try it and let me know

temp-64GTX, 2024-07-06:
Does it need some kind of latest Net Desktop Runtime or something? Currently it giving error (0xc0000142) at the start. 
Version from May-28 is starting fine.

temp-64GTX, 2024-07-06:
> Try it and let me know

Well, it's seems to kind of working, but i can't be sure. Lower zoom limit is limited to 110%, and my typical everyday-zooms are around 70%-90%. So if this 110% threshold can be lowered to something like 70% - it would be nice.

AdrianEddy, 2024-07-06:
What's the use case to have it below 100? Is is because of lens profile?

temp-64GTX, 2024-07-06:
For example. Gopro 10. Wide. 4000x3000 at 59.97fps. When i am driving at the straight line - zoom can drops to 65%. 
In case of usual mountain-bike riding, with same lens profile - it's around 70%-90%.
[image]

AdrianEddy, 2024-07-07:
@temp-64GTX I updated latest dev build to be able to set the value from 50. The slider is still starting at 110, but you can manually enter the value above 50 in the number field

closed 2024-07-07 by AdrianEddy

spirin, 2024-08-15:
> @temp-64GTX I updated latest dev build to be able to set the value from 50. The slider is still starting at 110, but you can manually enter the value above 50 in the number field

this dev build is a game changer🔥 but davinci plugin seems to work slower (macos m1)

AdrianEddy, 2024-08-24:
> this dev build is a game changer🔥 but davinci plugin seems to work slower (macos m1)

@spirin try latest build of the plugin, should be faster

spirin, 2024-08-24:
> > this dev build is a game changer🔥 but davinci plugin seems to work slower (macos m1)
> 
> @spirin try latest build of the plugin, should be faster

Works great! 👍

Shadetail, 2024-09-14:
@AdrianEddy , I've spent over 10 hours experimenting with the zoom limit feature, and no matter what I do, it doesn't behave as I hoped.

I use the "Static zoom" option instead of "Dynamic zoom," but the feature behaves the same regardless. Looking at the curves on the graph, they are identical. Note: Adjusting the "FOV" slider is necessary for the curves to visually update after changing the zoom limit or iterations values.

At 300%, this feature is disabled; the video is as smooth as it would have been without it. In my case, I'm using 100% smoothness with a 5-second max and 0.1 max at high velocity.

I'm using 15 iterations and have discovered, by examining the XYZ curves on the graph, that a higher iteration number is only useful for lower zoom limit values. For example, if I set the zoom limit to 150%, iterations beyond 4 stop making a difference. Therefore, higher iteration counts and longer processing times aren't always necessary. It seems this feature has to "work harder" the lower the zoom limit is, which makes sense since at 300%, it's inactive.

This feature reduces smoothness everywhere as its value is reduced from 300% down to 51%. What I expected it to do—and haven't been able to achieve—is to leave smoothness untouched in parts of the video where no zooming is necessary (i.e., where at static zoom with an FOV of 0.9 there's no clipping into canvas borders even with insanely high smoothness value). I would expect it to only reduce smoothness in parts where clipping occurs, or where additional zoom would normally be required to prevent clipping. Instead, it reduces smoothness everywhere.

Additionally, I would expect the "zoom limit" feature to make the "max smoothness at high velocity" feature obsolete because, rather than switching (or lerping) between two smoothness values depending on the velocity threshold, it's supposed to dynamically change the smoothness at each part of the video to achieve maximum smoothness that doesn't cause the camera FOV to depart the canvas without the ability to zoom in further than the set amount (or without the ability to zoom at all, in the simpler case).

Currently, I notice that I can set a very high smoothness as a baseline and then manually add keyframes to the video, subdividing it further, and reduce the smoothing value on each keyframe as needed until the camera doesn't clip anywhere (as indicated by the red warning lines/areas on the graph). The smoothness value is thus as high as possible in each part of the video. However, this process isn't an art form; it's a deterministic, labor-intensive task that could be completely automated.

For example, one could subdivide the entire timeline into, say, 1000 smoothness keyframes, each initially set to 0% smoothness. Then, iteratively increase each by 1%, one at a time, until the camera clips outside the canvas. If the camera clips as a result of that change, undo the change, lock the keyframe, and continue until no unlocked keyframes remain. This automated process would produce better results than manual keyframing due to the labor-intensive nature of doing it manually. Yet, it doesn't seem complicated at all.

I thought this was essentially what the "zoom limit" feature does, but upon closer inspection, it doesn't seem functionally equivalent. It keeps reducing smoothness in parts of the video where smoothness reduction isn't needed or desired.

I would expect such a feature to be a "one-click wonder" that doesn't require tinkering with settings beyond choosing your desired zoom level limit (or the desired static zoom). You're simply providing Gyroflow with one number, a single constraint, and letting it maximize smoothness within that constraint. Therefore, tweaking additional settings would, in theory, be pointless. Even if this process took 5 minutes to produce these 1000 keyframes, it would still be totally worth it.

I'm not sure if the zoom limit feature can be improved to function as I've described or if this should be considered a separate feature. However, even if it were separate, I believe it would render the current implementation obsolete.

AdrianEddy, 2024-09-14:
The way it works is that it calculates all the smoothing and zooming without any limit first. Then it goes through every frame, and lowers the gyro sample velocity value on that frame by the ratio of (calculated_zoom / target_zoom). If the zoom does not exceed the limit, that frame uses the original velocity value. 

Therefore, all 3 sliders are still used for the calculations, and the "Max smoothness at high velocity" and "Max smoothness" slider values are affecting this process.

This is done like this because the limit needs to be smoothed out before and after hitting that limit, it can't be a hard limit

> For example, one could subdivide the entire timeline into, say, 1000 smoothness keyframes, each initially set to 0% smoothness. Then, iteratively increase each by 1%, one at a time, until the camera [...]

This may sound like a good idea in theory, but it will have the same inaccuracy as the current algorithm, simply because changing one keyframe affects the neighboring ones in unpredictable ways

I've experimented with other approaches but I wasn't able to figure out a better algorithm, everything else I tried was much worse than this. 

I'm not seeing the behavior you're describing in my tests though

AdrianEddy, 2024-09-14:
You can compile from source and experiment with multiplying different values than the velocity (which is essentially multiplying the "smoothness")
The calculation is here 
https://github.com/gyroflow/gyroflow/blob/master/src/core/smoothing/default_algo.rs#L306-L308

I'm open to PRs if you want to improve this

Shadetail, 2024-09-15:
> I'm open to PRs if you want to improve this

Thank you quick answer and all the info! I'll try to see if I can figure out how to implement this, though I'm not very experienced so it might take me a while.

Shadetail, 2024-11-18:
I'm not finished yet, so I haven't made a PR, but here's update on my progress so far:

I implemented the first version of the feature, if anyone is impatient and wants to compile gyroflow themselves to us it right away, it's available here: https://github.com/Shadetail/gyroflow/tree/auto-keyframing

Dynamically controlling smoothness with keyframing was indeed more chaotic than I expected, but I made it work anyways. It's many orders of magnitude slower than the zoom limit implementation you did, but it's still usable and usually takes 5-30 minutes depending on video length and settings used. It's not compatible with zoom limit, so currently user has to chose to either use zoom limit or auto keyframing. I personally use gyroflow without dynamic zooming, so that's the use case I'm aiming the feature at and testing with.

So far results are mixed but promising, and more complicated than I expected. I blind tested it vs zoom limit feature on 3 videos, all with FOV set to 1, max smoothness at 5s and high velocity at 0.1s, for zoom limit using 11 iterations with "static zoom" and 101% zoom limit (this gave result closer to 100% zoom than actually setting it to 100%).
 - 16:9 GoPro (a bit tight): In 9 out of 11 cases where there was obvious difference in perceived smoothness, zoom limit feature won. This was a clear victory for zoom limit feature despite quaternion curves in the timeline appearing noticeably smoother on auto keyframing: https://imgur.com/a/eUhehJE , which really surprised me.
 - 8:7 GoPro (much less tight): In 3 out of 4 cases the auto keyframing feature won, again I was surprised to see a trade off situation rather than one approach being always better. Due to situation being much less tight, smoothing was easier and there were less non-smooth moments that allowed for clear comparisons, hence the unsatisfying low sample size, need to test this case more.
 - 4:3 Avata 2 (medium tight): I expected zoom limit to win here and I was ready to give up and pack my bags, but auto keyframing ended up winning by a landslide. Halfway through video I already had 18 clearly distinctive situations, and in all of them auto keyframing did better, in 4 cases much better. This footage unlike previous two was much faster and more aggressive flying, which might have played a role.

While with GoPro footage auto keyframing tool runs reasonably fast, with Avata footage it seems to be affected by the same issue I reported a while ago: https://github.com/gyroflow/gyroflow-plugins/issues/13 which since got moved over to gyroflow-ofx repository, but this actually confirms that this is not an issue specific to the plugin, but that it affects gyroflow itself. So much so, that with same length video (12 minute), my auto keyframing tool runs 24 times slower, which makes it not really usable for Avata 2 videos in a practical sense, which also happens to by my primary use case.

Before considering the tool finished and submitting a PR, I plan to:
 - improve UI implementation to only make the tool appear when "no zooming" is selected
 - add a feature/checkbox enabled by default to automatically run "improve keyframes" after "generate keyframes" is finished
 - add small improvement to the second stage to make it slightly more efficient.
 - look into the bug I encountered where smoothing calculation is sometimes stronger than it should be, causing clipping, but as soon as fov slider is touched it fixes itself. Will probably try to trigger that workaround automatically, and then update the clipping warning in the description.
 - will try to have the tool support FOV values that are not 1, expecting this to be easy. 
 - will try to look into the Avata 2 performance bug since it's blocking me from using the tool with majority of my videos.
 - will upload some example videos and make a small demo video demoing the feature

I think all this will take me a few more weeks at minimum, though probably more.