gyroflow/gyroflow #659: Vertical Zooming controls broken on macOS

ezhd, 2023-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.0

### What operating system are you using?

macOS Big Sur 11.6.6

### What happened?

On mac OS(MacBook Pro): chart navigation pop-up hint currently says: "**Control+Shift+Scroll** to zoom vertically"
[image]
Same is documented [https://docs.gyroflow.xyz/app/getting-started/basic-usage/timeline-and-gyro-chart](https://docs.gyroflow.xyz/app/getting-started/basic-usage/timeline-and-gyro-chart)

However (at least in my case) it is wrong
Figured out that it is actually **Control+Shift+"Horizontal Scroll"**.
by "Horizontal Scroll" I mean literally scrolling one finger horizontally on magic mouse
or scrolling horizontally with two fingers on touchpad.

**P.S.** Personally finding current panning/zooming controls (on Mac) very confusing.
If you actually support horizontal scrolling in your code, why not utilize full power of multitouch?
Or at least use 2 hotkey buttons for 2 actions instead of 3 buttons.

AdrianEddy, 2023-06-09:
What would be your preferred shortcuts for these 3 functions? I never used magic mouse, does it have scrolling in both directions? is it different on a touchpad?

ezhd, 2023-06-09:
Touchpad just adds one more finger to any scroll in any direction. Other than that they are pretty similar. 

I would suggest making shortcuts as close to popular video editing applications as possible. 
(however they are not using vertical zoom very often and it's UI specific) 

**Horizontal Pan** - Horizontal Scroll  /  ⌘Cmd + mouse wheel
**Horizontal Zoom** - ⌥Alt + Horizontal Scroll / ⌥Alt + mouse wheel / Pinch-to-zoom (on touchpad)
**Vertical Zoom** - Vertical Scroll / Shift + ⌥Alt + mouse wheel / Shift + Pinch-to-zoom (on touchpad)

AdrianEddy, 2023-06-11:
Ok, so vertical scrolling is fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload
Now it's just option + vertical scroll

I'll leave the rest shortcuts as is, the most commonly used action in gyroflow is horizontal zoom, so it's not really like a video editor timeline. Also now in latest dev build, the need for autosync usage is reduced (now we have 5 formats without syncing: GoPro, DJI, Insta360, BRAW and Sony)

closed 2023-06-11 by AdrianEddy