gyroflow/gyroflow #661: Exporting PNG files for Meshroom, with less fps.

eirasys, 2023-06-16:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Objective is using meshroom to convert drone videos to 3D mesh drawings.
Can't find a way of making PNG files with steps from original fps. We don't need all frames to capture 3D, for example only 1 frame in each 10 frames of video, the idea is to export only a minimum necessary frames.
I know it can already export PNG inside a window of the full stabilized video, but not with a defined frame intervals, resulting in less PNG files exported.
Thanks

AdrianEddy, 2023-06-16:
You can already achieve that by setting Video speed to 1000% for example - it will simply skip the frames in between effectively doing what you're asking for. I don't think I should add a dedicated option for that, because it's a very niche use case

closed 2023-06-16 by AdrianEddy

eirasys, 2023-06-17:
Ok thanks! So I was trying with the wrong parameters, in the play back speed and the video fps. But all stabilization processing is affected since I suppose it is skipping frames for the calculations. I mean, the result are not the exact same frames as the processing with 100% speed. For high speed values zoom gets very high.

AdrianEddy, 2023-06-17:
Just uncheck these 2 checkboxes near the speed slider and it will not affect the stabilization or zooming