gyroflow/gyroflow #662: Error loading shared library libc++.so.1 on AppImage build

rjmalagon, 2023-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.1 AppImage

### What operating system are you using?

Fedora 38 - Gnome

### What GPU are you using?

Radeon R9 360

### What happened?

Fails to load

### Relevant log output

```shell
./Gyroflow-linux64.AppImage 
/tmp/.mount_GyroflJsnjrO//gyroflow: error while loading shared libraries: libc++.so.1: cannot open shared object file: No such file or directory
```

AdrianEddy, 2023-06-16:
you need to install `libc++-dev`, as mentioned in the [documentation ](https://docs.gyroflow.xyz/app/getting-started/installation)
In case of Fedora, it's `dnf install libcxx libcxx-devel`

closed 2023-06-16 by AdrianEddy

rjmalagon, 2023-06-16:
Thanks, usually I think of AppImage binaries as dependencyless. Have a nice day. (P.D a flatpak file could be awesome too.)

rjmalagon, 2023-06-16:
No dependency friendly on Fedora. After installing some required files, it outputs this:
`./Gyroflow-linux64.AppImage`
      
`/tmp/.mount_Gyrofl1TqJya//gyroflow: /tmp/.mount_Gyrofl1TqJya//lib/x86_64-linux-gnu/libselinux.so.1: no version information available (required by /lib64/libkrb5support.so.0)
*** stack smashing detected ***: terminated
zsh: IOT instruction (core dumped)  ./Gyroflow-linux64.AppImage
`

AdrianEddy, 2023-06-16:
try .tar.gz, AppImage is not a great format

Gyroflow is tricky because it uses multiple GPU and video APIs extensively. 
I know people run Gyroflow on Fedora just fine, so you have to figure out your hardware-specific deps