gyroflow/gyroflow #663: Cannot save lens profile to file: it tries to open empty filename

vi, 2023-06-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.5.1

### What operating system are you using?

Linux Debian Bookworm (AppImage)

### What GPU are you using?

Intel + NVidia

### What happened?

After doing the lens calibration I try to save it to a json file. It shows something like "OS error: No such file or directory".
Also file chooser dialog seems to be very buggy - I suspect chosen filename does not propagate to the actual code.

Also I tried to just export a video without loading the lens data or sync points (or whatever) as a smoke test and it just crashed (attached the dump).

### Relevant log output

```shell
strace shows this:

    openat(AT_FDCWD</tmp/shell-bash.prVnTEVd>, "", O_WRONLY|O_CREAT|O_TRUNC|O_CLOEXEC, 0666) = -1 ENOENT (No such file or directory)
```

AdrianEddy, 2023-06-16:
did you try both AppImage and .tar.gz? How does the file dialog look like? It's probably some missing package or AppImage issue, the file dialog code is not in gyroflow

vi, 2023-06-16:
Also tried nightly build, both AppImage and regular version.

File dialog looks relatively OK (though does not react to "Enter" keyboard key, only to explicit button), and default suggested filename is strange.

Attached the video session that shows how it all looks like. Also depicts the graphical glitch.
[screen.webm](https://github.com/gyroflow/gyroflow/assets/173219/bc49ee48-1aaa-49c0-a0e1-c18b349a05a8)

vi, 2023-06-16:
Workaround for export crash is disabling hardware encoding.

Unlike 1.5.1 (which crashes), Gyroflow 1.4.2 shows proper error message mentioning FFmpeg errors if hardware accelerated encoding cannot be used.

(I suggest to add a hint about disabling hardware encoding on that error message).

vi, 2023-06-16:
Hacky workaround:

dump memory of Gyroflow as if failed to export the file (`gcore`), search the core dump for things like `"distortion_coeffs": ` and manually construct the output file piece by piece using text editor.

AdrianEddy, 2023-06-16:
hm, the entire json should just be in the log, either in console or in gyroflow.log file, no need to reconstruct it from pieces

vi, 2023-06-17:
I see console output like this:

```
00:55:52 [DEBUG] (2) gyroflow::controller: rms: 0.4072699378352677, used_frames: [10, 70, 110, 140, 220, 260, 310, 340, 410, 420, 480, 530, 560, 640, 650, 730, 760, 800, 860, 890], camera_matrix:
  ┌                                                          ┐
  │ 12395.872396194642                  0 1004.5600567757053 │
  │                  0 12360.802920507009  680.5995015746504 │
  │                  0                  0                  1 │
  └                                                          ┘

, coefficients:
  ┌                    ┐
  │ 3.7872135848435904 │
  │ -2522.531353319025 │
  │  815707.7232352773 │
  │  -87840302.6828353 │
  └                    ┘
```

I don't see `gyroflow.log` anywhere (checked opened file descriptors or memory maps). Command line options help also does not mention log file.

AdrianEddy, 2023-06-17:
ah because the json content is printed to console and the log only when writing to a file, and it fails to write because it doesn't have a name. The dialog you're seeing is a fallback dialog from Qt. Normally it should use your system dialog, but that often fails due to insane mess with the Linux GUI stuff and it's dependencies and trying to the app as a single universal binary. Which desktop environment you're running? Qt should use GTK+ file dialog, maybe try to install `xdg-desktop-portal-gtk` or something

AppImage is not generally recommended, it's better to use .tar.gz

vi, 2023-06-17:
There is `/usr/libexec/xdg-desktop-portal-gtk`, but system is systemd-less and mostly dbus-less, so it probably won't work properly.

Anyway, fallback dialog should also be fixed to work properly.

Ideally there should be just a plain text file path edit box, just like for the path of exported video.

AdrianEddy, 2023-06-17:
Fallback dialog is entirely inside Qt, it's outside of my control

vi, 2023-06-17:
Trying to use `QT_QPA_PLATFORMTHEME=gtk3` workaround:

```
(gyroflow:18292): Gtk-WARNING **: 03:15:51.497: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../gtk/gtkiconhelper.c:495:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /tmp/.mount_Gyrofl9NGWid//usr/share/icons/Adwaita/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../gtk/gtkiconhelper.c:495:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /tmp/.mount_Gyrofl9NGWid//usr/share/icons/Adwaita/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Crashdump written to /tmp/shell-bash.pAnNumdG/528bd249-c371-41cb-d98bdf9f-fd732e06.dmp
Aborted
```

when trying to open a file chooser dialog.

AdrianEddy, 2023-09-20:
Can you check if it works any better in latest dev build? https://gyroflow.xyz/devbuild/

AdrianEddy, 2023-09-20:
under Linux: https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-linux.zip

vi, 2023-09-20:
Managed to save a lens profile even with a fallback Qt dialog.

But `QT_QPA_PLATFORMTHEME=gtk3 ./Gyroflow-linux64.AppImage` fails when I try to open a file:

```
17:39:50 [INFO] Latest version: v1.5.2, current version: 1.5.2 (gh1460)

(gyroflow:8999): Gtk-WARNING **: 19:39:50.698: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../gtk/gtkiconhelper.c:495:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /tmp/.mount_GyroflZqg3qi//usr/share/icons/Adwaita/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../gtk/gtkiconhelper.c:495:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /tmp/.mount_GyroflZqg3qi//usr/share/icons/Adwaita/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Crashdump written to /tmp/shell-bash.A74gLNNv/a580577e-5948-46e3-63ba03a0-536294d6.dmp
Aborted
```

[crashdump.zip](https://github.com/gyroflow/gyroflow/files/12676002/crashdump.zip)

AdrianEddy, 2023-09-20:
hmm does the same crash happen in .tar.gz?

AdrianEddy, 2023-09-20:
https://github.com/bottlesdevs/Bottles/issues/590#issuecomment-970596956 does this fix work for you?

vi, 2023-09-20:
With 

    XDG_DATA_DIRS="/usr/local/share:/usr/share" QT_QPA_PLATFORMTHEME=gtk3 ./Gyroflow-linux64.AppImage

it does not crash when I specify lens profile file and the file dialog is OK.

AdrianEddy, 2023-09-20:
How about now? https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-linux.zip
I added XDG_DATA_DIRS to AppImage

vi, 2023-09-20:
Now it uses good file open or save dialog even without environment variables.

AdrianEddy, 2023-09-20:
Cool! thanks for testing

closed 2023-09-20 by AdrianEddy