gyroflow/gyroflow #672: Failed to initialize OpenCL Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))

markusressel, 2023-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.1

### What operating system are you using?

Linux Silver 6.3.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 14 Jun 2023 20:10:31 +0000 x86_64 GNU/Linux

### What GPU are you using?

AMD Radeon RX 6700 XT (Vulkan)

### What happened?

Opening the application is fine, when loading a video, while or after (not quite sure) analysing the Gyro data, the application crashes.

Crash dump:
[d8879b04-6325-4820-2fdd4094-ffe09491.zip](https://github.com/gyroflow/gyroflow/files/11862661/d8879b04-6325-4820-2fdd4094-ffe09491.zip)

Note: I checked the box for "I have tried the latest build" to be able to create this report, but I (currently) cannot test this, because I am not able to build gyroflow from scratch to run it on my system. If there is an easier way to run a development build on an arch linux based system without building manually, please let me know, if necessary.

### Relevant log output

```shell
0.21.0 (git 34be505) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang16.0.5 (++20230602123304+185b81e034ba-1~exp1~20230602003315.101)GCC4.2.1; libc++160005; 12:20:05 Jun 11 2023
02:54:02 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
02:54:02 [DEBUG] (1) gyroflow: Gyroflow 1.5.1 (gh1266)
mdk plugin 'mdk-braw': 0x55874c0e62b0 from dir: /usr/share/gyroflow/lib/
mdk plugin 'mdk-r3d': 0x55874c0e67a0 from dir: /usr/share/gyroflow/lib/
02:54:03 [DEBUG] (2) gyroflow::util: Minidump uploaded: OK
02:54:13 [ERROR] thread 'main' panicked at 'Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))': /root/.cargo/git/checkouts/ocl-51b7c5264cbf134c/0c87db6/ocl/src/standard/platform.rs:49
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: main
  13: <unknown>
  14: __libc_start_main
  15: <unknown>

02:54:13 [ERROR] Failed to initialize OpenCL Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))
02:54:13 [DEBUG] (1) gyroflow_core::gpu::wgpu: WGPU adapter: AdapterInfo { name: "AMD Radeon RX 6700 XT", vendor: 4098, device: 29663, device_type: DiscreteGpu, driver: "AMD open-source driver", driver_info: "2023.Q2.2 (LLPC)", backend: Vulkan }
02:54:13 [DEBUG] (1) gyroflow::rendering: GPU type: Amd, from name: amd radeon rx 6700 xt
02:54:13 [INFO] Lens profiles directory: "/usr/share/gyroflow/camera_presets"
02:54:13 [INFO] Loaded 7055 lens profiles in 83.913ms
02:54:14 [INFO] Latest version: v1.5.1, current version: 1.5.1 (gh1266)
02:54:23 [ERROR] thread '<unnamed>' panicked at 'Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))': /root/.cargo/git/checkouts/ocl-51b7c5264cbf134c/0c87db6/ocl/src/standard/platform.rs:49
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
[... 30 lines cut ...]
02:54:28 [INFO] surface size: 3840x2160
02:54:29 [DEBUG] (2) telemetry_parser::dji: Metadata: Object {"clip_meta_header": Object {"proto_file_name": String("dvtm_wm169.proto"), "library_proto_version": String("02.00.00"), "product_proto_version": String("2.0.2"), "product_sn": String("5LSKL2N00201T0"), "product_firmware_version": String("01.02.00.00"), "meta_encryption_type": Number(0), "meta_compression_type": Number(0), "clip_timestamp": Number(116119083), "product_name": String("DJI FC8383")}, "clip_streams_meta": Object {"video_stream_num": Number(1), "audio_stream_num": Number(0)}, "distortion_coefficients": Object {"coeffients": Array [Number(0.24991768598556519), Number(0.0136057548224926), Number(-0.06208358332514763), Number(0.012193071655929089)]}, "sensor_readout_time": Object {"readout_time": Number(7476749)}, "sensor_read_direction": Object {"direction": Number(4)}, "digital_focal_length": Object {"focal_length": Number(1405.1292724609375)}, "eis_status": Object {"status": Number(0)}, "imu_sampling_rate": Object {"imu_sampling_rate": Number(2000)}, "sensor_fps": Object {"sensor_frame_rate": Number(120.27415466308594)}}
02:54:29 [DEBUG] (2) telemetry_parser::dji: Quaternions: [TimeQuaternion { t: -4.23053323394373, v: Quaternion { w: -0.15177729725837708, x: -0.18161460757255554, y: 0.7888426929712296, z: 0.5671921819448471 } }, TimeQuaternion { t: -3.7091785339775676, v: Quaternion { w: -0.1517719216644764, x: -0.18160418793559074, y: 0.7888440825045109, z: 0.5671950243413448 } }, TimeQuaternion { t: -3.187823834011406, v: Quaternion { w: -0.15176766365766525, x: -0.18159324675798416, y: 0.7888450101017952, z: 0.5671983286738396 } }, TimeQuaternion { t: -2.6664691340452444, v: Quaternion { w: -0.15176459774374962, x: -0.18158242478966713, y: 0.7888456843793392, z: 0.5672018863260746 } }, TimeQuaternion { t: -2.145114434079083, v: Quaternion { w: -0.15176236629486084, x: -0.1815720796585083, y: 0.7888458967208862, z: 0.5672053098678589 } }, TimeQuaternion { t: -1.6237597341129213, v: Quaternion { w: -0.1517607867717743, x: -0.18156281113624573, y: 0.7888460159301758, z: 0.5672085285186768 } }, TimeQuaternion { t: -1.1024050341467597, v: Quaternion { w: -0.1517595760524273, x: -0.18155496194958687, y: 0.7888460718095303, z: 0.5672113336622715 } }, TimeQuaternion { t: -0.5810503341805984, v: Quaternion { w: -0.1517585963010788, x: -0.1815488487482071, y: 0.7888460755348206, z: 0.5672135353088379 } }, TimeQuaternion { t: -0.05969563421443678, v: Quaternion { w: -0.15175770223140717, x: -0.1815444976091385, y: 0.7888460457324982, z: 0.567215234041214 } }, TimeQuaternion { t: 0.4616590657517248, v: Quaternion { w: -0.15175709128379822, x: -0.18154260516166687, y: 0.7888458967208862, z: 0.5672162175178528 } }, TimeQuaternion { t: 0.9830137657178862, v: Quaternion { w: -0.1517568603157997, x: -0.18154295533895493, y: 0.7888455167412758, z: 0.5672167167067528 } }, TimeQuaternion { t: 1.504368465684048, v: Quaternion { w: -0.15175720676779747, x: -0.1815456487238407, y: 0.7888448052108288, z: 0.5672167725861073 } }, TimeQuaternion { t: 2.0257231656502097, v: Quaternion { w: -0.15175830945372581, x: -0.18155056610703468, y: 0.7888437025249004, z: 0.5672165043652058 } }, TimeQuaternion { t: 2.5470778656163713, v: Quaternion { w: -0.15176014229655266, x: -0.1815575547516346, y: 0.7888420782983303, z: 0.5672158636152744 } }, TimeQuaternion { t: 3.0684325655825324, v: Quaternion { w: -0.15176276117563248, x: -0.18156643956899643, y: 0.7888401374220848, z: 0.5672150030732155 } }, TimeQuaternion { t: 3.589787265548694, v: Quaternion { w: -0.15176579728722572, x: -0.18157705292105675, y: 0.7888380028307438, z: 0.5672139786183834 } }]
02:54:29 [DEBUG] (21) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
02:54:29 [DEBUG] (21) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7fabb56b23d0) QSize(3840, 2160)
02:54:29 [DEBUG] (2) telemetry_parser::dji: Metadata: Object {"clip_meta_header": Object {"proto_file_name": String("dvtm_wm169.proto"), "library_proto_version": String("02.00.00"), "product_proto_version": String("2.0.2"), "product_sn": String("5LSKL2N00201T0"), "product_firmware_version": String("01.02.00.00"), "meta_encryption_type": Number(0), "meta_compression_type": Number(0), "clip_timestamp": Number(303989866), "product_name": String("D
JI FC8383")}, "clip_streams_meta": Object {"video_stream_num": Number(1), "audio_stream_num": Number(0)}, "distortion_coefficients": Object {"coeffients": Array [Number(0.24991768598556519), Number(0.0136057548224926), Number(-0.06208358332514763), Number(0.012193071655929089)]}, "sensor_readout_time": Object {"readout_time": Number(7476749)}, "sensor_read_direction": Object {"direction": Number(4)}, "digital_focal_length": Object {"focal_length": Number(1405.1292724609375)}, "eis_status": Object {"status": Number(0)}, "imu_sampling_rate": Object {"imu_sampling_rate": Number(2000)}, "sensor_fps": Object {"sensor_frame_rate": Number(120.27415466308594)}}
02:54:29 [DEBUG] (2) gyroflow_core::camera_identifier: Camera ID Brand: DJI, Model: FC8383, Metadata: Object {"clip_meta_header": Object {"proto_file_name": String("dvtm_wm169.proto"), "library_proto_version": String("02.00.00"), "product_proto_version": String("2 [... cut ...] sor_frame_rate": Number(120.27415466308594)}}
02:54:29 [INFO] CameraIdentifier {
    brand: "DJI",
    model: "FC8383",
    lens_model: "",
    lens_info: "",
    focal_length: None,
    camera_setting: "",
    fps: 119880,
    video_width: 3840,
    video_height: 2160,
    additional: "",
    identifier: "",
}
malloc(): unsorted double linked list corrupted
Crashdump written to /home/markus/d8879b04-6325-4820-2fdd4094-ffe09491.dmp
fish: Job 1, 'gyroflow' terminated by signal SIGABRT (Abort)
```

AdrianEddy, 2023-06-26:
Some OpenCL dependency or a GPU driver is missing for your system and hardware, you'll have to figure out which package to install for OpenCL, or you can disable OpenCL by setting `export NO_OPENCL=1` in the terminal before launching gyroflow.

List of dependencies for OpenCL is in the [documentation](https://docs.gyroflow.xyz/app/getting-started/installation) for debian-based distros, so you'll have to find equivalents for arch

You can also try different `Device for video processing` in Advanced and also make sure you have VDPAU and VAAPI dependencies installed. I don't know anything about arch, so I can't help you with exact packages, but it's definitely caused by that

> Note: I checked the box for "I have tried the latest build" to be able to create this report, but I (currently) cannot test this, because I am not able to build gyroflow from scratch to run it on my [...]

Latest build is always available at https://gyroflow.xyz/devbuild/?autodownload or in Github Actions

markusressel, 2023-06-26:
I think installing [rocm-opencl-runtime](https://archlinux.org/packages/?name=rocm-opencl-runtime) fixed the issue.
Gyroflow is no longer crashing. Thx!