gyroflow/gyroflow #674: Replace audio

victorlysak, 2023-06-26:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Please.
Would it be possible to add replace the video file audio track with another track, like you can do in ffmpeg with -map 1:a:0 ...

Something like this is described in more details in the following post:
[https://superuser.com/questions/1137612/ffmpeg-replace-audio-in-video](https://superuser.com/questions/1137612/ffmpeg-replace-audio-in-video)

I tried to add the imput.wav and map in custom encoder options, but it didn't work.

thank you.

victorlysak, 2023-06-26:
thank you for the reply

the thing is that I have a video file and a recorded audio at the same time, 
and I need to replace the orginal sound with the recorded one,

That means that I export the video in gyroflow,
and then I have to run ffmpeg on that file again in order to replace the audio,
and then edit the video file with the new audio

It's not a big deal,

thank you anyway.

AdrianEddy, 2023-06-26:
you would have to have the audio file already synced with the video, and to do that you'd need another software anyway. 
If you have to use another video/audio software, then do it all at once there

You can also use the OpenFX plugin with DaVinci Resolve or MAGIX Vegas, or gyroflow plugin for Final Cut on mac and do everything in one go without transcoding

Adding this option to gyroflow directly will add a lot more complexity while being very niche use case

closed 2023-06-26 by AdrianEddy

victorlysak, 2023-06-26:
thank you very much for your help