gyroflow/gyroflow #676: Issue: 3 Videos and GCSV are joined with file-joiner, but the joined video is not stabilized, while individually each video stabilizes ok

lesau, 2023-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.1

### What operating system are you using?

Windows 10 Enterprise 19045.3086

### What GPU are you using?

Intel® UHD Graphics for 10th Gen Intel® Processors

### What happened?

Hi, I tried joining 3 video file chapters from the split video made by CadDx Walnut camera using GF 1.5.1. GF joined them successfully and also created a joined .gcsv with the same name as described in https://docs.gyroflow.xyz/app/getting-started/file-joiner. However, the video doesn't get stabilized even around the green labeled sync points. In the GUI, I see 4 red and 2 green labeled sync points, gyro data appears empty/flat in the middle part of the joined video that corresponds to the 2nd gcsv file, and the video is not stabilized in either of the 3 joined videos time frames. 

In another test, each video stabilizes properly by itself without joining using the same stabilization settings. So, something is iffy with the joining. 

Video captured by Walnut camera in 4000x3000, 29.97fps, data rate 100kbps, bit rate 128kbps, flat color, 1/60 exposure, ISO 100. 
GF Settings: FOV 1.0, DEFAULT smoothing, smoothness 0.5-0.75, zooming speed 2.5, rolling shutter left at defaults see below.
Intel(R) Core(TM) i7-10610U CPU @ 1.80GHz

Picture: 3 videos joined w/ missing gyro data for the 2nd video:
[image]

Picture: zoomed in one sync point in the 1st video when joined, but the video still looks shaking.
[image]

1st gyro file of the 3-part video: https://drive.google.com/file/d/1vRdKt558ULQdTnXYM08-F_TvIxy3vopO/view?usp=drive_link
2nd gyro file of the 3-part video: https://drive.google.com/file/d/1kZA5uyJFaputDoGh7dze65B_bsw6sTrp/view?usp=drive_link

I can upload the videos if necessary.

AdrianEddy, 2023-07-11:
Should be fixed in 76853f81a0a13ed9f99462fb4de414654314dd72
https://gyroflow.xyz/devbuild/?autodownload

BTW: Is the Walnut on latest firmware?

closed 2023-07-11 by AdrianEddy

lesau, 2023-07-12:
@AdrianEddy , thanks, this fixed the issue! Thanks for the quick turnaround! Yes, used the latest Walnut firmware.