gyroflow/gyroflow #678: Gyroflow 1.5.1 no longer starts

netdisable, 2023-07-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.1

### What operating system are you using?

Windows 11 Home 22H2

### What GPU are you using?

nVidia GeForce RTX 3050 ti

### What happened?

I copied the Gyroflow folder to my documents folder. When I launch the executable file the program starts but gets stuck on the main screen with the loading logo (written Gyroflow in italics) superimposed.

[image]

AdrianEddy, 2023-07-12:
Try this:

1. Click `Windows+R` and type `regedit` or look for Registry Editor in Start menu
2. Go to `Computer\HKEY_CURRENT_USER\SOFTWARE\Gyroflow` (enter this in the top bar and hit Enter)
3. Delete the entire key

netdisable, 2023-07-12:
Thank you for fast reply...

This solution not working.

AdrianEddy, 2023-07-12:
hmm did it work before or it's the first time you're trying it?
Make sure to install latest GPU driver from nvidia.com and all latest windows updates

AdrianEddy, 2023-07-12:
yeah so I've seen this already - the driver for the integrated AMD card needs to be installed separately, even though Windows installs it. Download latest one from AMD.com website and it should work

AdrianEddy, 2023-07-12:
let me know if it worked, I'll be releasing a new version today and wanted to make sure there's no issue in gyroflow itself

netdisable, 2023-07-12:
it worked correctly.  Updating the AMD driver directly from the site and not through the Windows update service solved the problem.Thanks a lot.MGdroneSent from Android device

closed 2023-07-12 by AdrianEddy

stz0117, 2023-08-03:
> yeah so I've seen this already - the driver for the integrated AMD card needs to be installed separately, even though Windows installs it. Download latest one from AMD.com website and it should work

Works for me too👍. It might be a common problem for laptop users with two graphic cards. I spend a lot of time on finding solutions and finally find this issue😂. Maybe the document can mention the GPU driver stuff.