gyroflow/gyroflow #679: Can't launch AppImage

RafaelLinux, 2023-07-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.2 (latest)

### What operating system are you using?

opensuse Tumbleweed

### What GPU are you using?

Nvidia GTX 1050Ti

### What happened?

Can't see nothing launching from Dolphin.
From console, it show `APPRUN_ERROR: Permission denied` but issue has nothing to do with permissions at file level. I have about 17 programs in AppImage format and they all launch perfectly from Dolphin without the need to activate the run permission.

What's more, even with permissions enabled, the error is exactly the same.

### Relevant log output

```shell
APPRUN_ERROR: Permission denied
```

AdrianEddy, 2023-07-12:
Use the .tar.gz package, AppImage is a terrible format, nothing but issues with it. It might be fine for simple apps, but Gyroflow makes extensive use of GPU stuff and drivers and dependencies for that are a nightmare when it comes to distributing universal app across all distros, systems, devices and graphics environments

RafaelLinux, 2023-07-13:
It's true AppImage is not the easiest way for developers to distribute their creations. However, it's the easiest and flexible way for  any Linux distro user to get up-to-date versions.  Flatpak is too gnome-way work so very limited (it even don't let user to type a path when browsing files from a file/folder internal dialog). For some developers, it's a headache. However, I use CPU/GPU intensive use applications in AppImage format, like Olive, OpenShot, CinelerraGG, Ossia or ShutterEncoder and all them works like a charm like a lot more that need less resources, like  Frea:ac, Upscayl, Heroic and a lot more. User can get even an AppImage of LibreOffice!!!!

There have been cases where I myself have provided more information to developers in order to solve problems that have arisen with some applications that have started using the AppImage format. In order to reach more users in the simplest way, it is the best way to distribute applications. 

For example, I share my user folder between two different distributions and, by using AppImage, I don't have to install the application on both platforms, since an AppImage is like a "Portable" on Windows.

In any case, I will use your compressed version but hopefully you can solve the problem with the AppImage version and any information I can give you for testing, you can count on me.

Thanks

RafaelLinux, 2023-07-13:
Well, I tried compressed version and no luck:
```
./gyroflow 
./gyroflow: error while loading shared libraries: libc++.so.1: cannot open shared object file: No such file or directory

```

So I'll need to wait some time ...  :(

AdrianEddy, 2023-07-13:
libc++ is a needed dependency for Gyroflow, I think on openSUSE it's provided by ` libc++abi-devel` so install this package

I know that it should be all contained, but Gyroflow has a lot of pieces, it's not just a C/C++ app, it has many layers of different dependencies, built with different compilers and languages, we have Qt, OpenCV, ffmpeg, mdk-sdk, Rust, C++

sure I would love to have a simple binary that works everywhere but Linux is so diverse with so many different moving pieces and I alone can't simply prepare and test packages for all of them. The apps you mentioned probably have more community behind them, so it's easier there

I'll try to see if I can reproduce that issue with AppImage, but the reality is on Linux you'll have to do some tinkering yourself

RafaelLinux, 2023-07-14:
> libc++ is a needed dependency for Gyroflow, I think on openSUSE it's provided by ` libc++abi-devel` so install this package
Unfortunately, each distro uses his own suffix to indicate if a file is source code or compiled. In openSUSE (and maybe Fedora) the "-devel" indicates source code file.

After some search thru YaST, I found what is needed for openSUSE:  libc++1 and  libc++abi1 libraries:
`sudo zypper in libc++1 libc++abi1`

I'm now in a Wayland session, and despite no errors are showed, a black window is the only thing I can see. I'll try with X11 tomorrow.

Thank you

> I know that it should be all contained, but Gyroflow has a lot of pieces, it's not just a C/C++ app, it has many layers of different dependencies, built with different compilers and languages, we [...]

> 
> I'll try to see if I can reproduce that issue with AppImage, but the reality is on Linux you'll have to do some tinkering yourself

I'll comment here anything I found out about AppImage issue.

victorlysak, 2023-07-18:
I run on one machine Arch Linux, kde plasma (NV rtx 2060 super), on another Manjaoro, kde plasma (intel 920), and Yes, libc++ always pops up when you start gyroflow.AppImage, and then:
1. it runs very well on plasma X11.
2. it runs with fully transparent screen on plasma Wayland and the only thing you can see is the mouse trails and nothing else is happening.
3. it runs very well on gnome X11
4. it runs very well on gnome Wayland
the above applys only the my machine that has nvidia. the laptop with intel 920 works well in plasma wayland and x11.

So, maybe there is something that is needed for plasma Wayland. I must also say that I tried to run plasma by login from gdm instead of sddm (window manager) but it is the same problem with plasma Wayland.

AdrianEddy, 2023-07-18:
Try `export QT_QPA_PLATFORM=wayland`
You can also try `export QSG_RHI_BACKEND=vulkan`

In general Google for Qt + Wayland issues, there will probably be answers online

RafaelLinux, 2023-07-18:
As I wrote, I has pending trying in X11 and I can confirm that GyroFlow is visible and operative in  X11 session, .

nightflasher, 2023-08-14:
tried the export as well, but it din't work for me.
6e99a96f-dcf1-45b3-c9a0f98e-c8e059c7.dmp - tar.gz
8201afa1-46e9-42a7-b0271cbd-13f410ac.dmp - AppImage
don't know why it stopped working.

nightflasher, 2023-08-15:
tested many versions back, now i got 1.0.1 working again. no chance of running latest and greates.

RafaelLinux, 2023-09-28:
Same error "APPRUN_ERROR: Permission denied"

AdrianEddy, 2023-10-12:
@RafaelLinux can you try the latest one? https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-linux.zip
Try the `Gyroflow-linux64.AppImage` one

RafaelLinux, 2023-10-13:
Of course, I can. And, yes!!! Your changes made it work!!! (at least in X11). I will try tonight under Wayland and I'll update this info.

Good job!!! Thanks!!

nightflasher, 2023-10-13:
Forgot to answer, sorry!

I had to remove the oibaf-repo and all the drivers and everything is working since than.
Had to read through the #246 issue to get more fps for rendering. And now it's doing up to 6fps instead of up to 2fps, which is still very slow. All the export is running on cpu because GPU still doesn't work.

RafaelLinux, 2023-10-15:
> Forgot to answer, sorry!
>
> I had to remove the oibaf-repo and all the drivers and everything is working since than. Had to read through the #246 issue to get more fps for rendering. And now it's [...]

My apologies too, I needed to reinstall my os so it took me more time to report about results of launching in X11. It worked too!!!. However, I think maybe this output will useful to you (from console, in X11) cause shows some errors:

```
./Gyroflow-linux64.AppImage 
/tmp/.mount_GyroflMjWdEe//gyroflow: /tmp/.mount_GyroflMjWdEe//lib/x86_64-linux-gnu/libselinux.so.1: no version information available (required by /lib64/libkrb5support.so.0)
.22.1 (git 3d0bf01) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang17.0.3 (++20231010073202+37b79e779f44-1~exp1~20231010073304.52)GCC4.2.1; libc++170003; 01:40:09 Oct 12 2023
14:22:36 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
14:22:36 [DEBUG] (1) gyroflow: Gyroflow 1.5.3 (gh1593)
14:22:36 [DEBUG] (1) gyroflow: Executable path: Some("/tmp/.mount_GyroflMjWdEe/gyroflow")
14:22:36 [DEBUG] (1) gyroflow: SDK path: "/tmp/.mount_GyroflMjWdEe/lib/"
mdk plugin 'mdk-braw': 0x55d1bb24b6c0 from dir: /tmp/.mount_GyroflMjWdEe//lib/
mdk plugin 'mdk-r3d': 0x55d1bb24bbc0 from dir: /tmp/.mount_GyroflMjWdEe//lib/
14:22:37 [WARN] QStandardPaths: runtime directory '/run/user/1000' is not owned by UID 0, but a directory permissions 0700 owned by UID 1000 GID 1000
Gtk-Message: 16:22:37.335: Failed to load module "canberra-gtk-module"
Gtk-Message: 16:22:37.374: Failed to load module "colorreload-gtk-module"
Gtk-Message: 16:22:37.375: Failed to load module "window-decorations-gtk-module"
14:22:37 [DEBUG] (1) gyroflow_core::filesystem: path_to_url | path:  -> Ok("")
14:22:47 [ERROR] thread 'main' panicked at 'Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))': /root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ocl-0.19.5/src/standard/platform.rs:49
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: main
  13: __libc_start_call_main
  14: __libc_start_main@GLIBC_2.2.5
  15: <unknown>

14:22:47 [ERROR] Failed to initialize OpenCL Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))
14:22:47 [DEBUG] (1) gyroflow_core::gpu::wgpu: WGPU adapter: AdapterInfo { name: "AMD Radeon Graphics (RADV GFX1102)", vendor: 4098, device: 29824, device_type: DiscreteGpu, driver: "radv", driver_info: "Mesa 23.2.1", backend: Vulkan }
14:22:47 [DEBUG] (1) gyroflow::rendering: GPU type: Amd, from name: amd radeon graphics (radv gfx1102)
14:22:47 [DEBUG] (1) default: [in MDKPlayer::MDKPlayer()] X11 display: 0x55d1bb475510
14:22:49 [INFO] Lens profiles directory: "/tmp/.mount_GyroflMjWdEe/camera_presets"
14:22:49 [INFO] Latest version: v1.5.3, current version: 1.5.3 (gh1593)
14:22:49 [INFO] Loaded 10110 lens profiles in 554.515ms
14:22:58 [ERROR] thread '<unnamed>' panicked at 'Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))': /root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ocl-0.19.5/src/standard/platform.rs:49
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: start_thread
  14: __clone3

14:22:58 [ERROR] Failed to initialize OpenCL Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))
14:22:58 [INFO] GPU list: ["[wgpu] AMD Radeon Graphics (RADV GFX1102) (Vulkan)", "[wgpu] AMD Radeon Graphics (gfx1102, LLVM 17.0.2, DRM 3.54, 6.5.6-1-default) (Gl)", "[wgpu] AMD Radeon Graphics (RADV GFX1102) (Vulkan)"]

```

AdrianEddy, 2023-10-15:
Thanks for testing! the errors in this latest log are related to OpenCL. You can ignore them or you can try to install OpenCL package for your system and GPU, you should be able to find that info online

closed 2023-10-15 by AdrianEddy

RafaelLinux, 2023-10-15:
Ok. In my recent installed openSUSE (after updating, installing codecs, etc) I got similar errors under Wayland .... but finally crashed (dont know why):```

./Gyroflow-linux64.AppImage 
/tmp/.mount_GyroflopYO3O//gyroflow: /tmp/.mount_GyroflopYO3O//lib/x86_64-linux-gnu/libselinux.so.1: no version information available (required by /lib64/libkrb5support.so.0)
0.22.1 (git 3d0bf01) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang17.0.3 (++20231010073202+37b79e779f44-1~exp1~20231010073304.52)GCC4.2.1; libc++170003; 01:40:09 Oct 12 2023
15:32:23 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
15:32:23 [DEBUG] (1) gyroflow: Gyroflow 1.5.3 (gh1593)
15:32:23 [DEBUG] (1) gyroflow: Executable path: Some("/tmp/.mount_GyroflopYO3O/gyroflow")
15:32:23 [DEBUG] (1) gyroflow: SDK path: "/tmp/.mount_GyroflopYO3O/lib/"
mdk plugin 'mdk-braw': 0x556b31ae68f0 from dir: /tmp/.mount_GyroflopYO3O//lib/
mdk plugin 'mdk-r3d': 0x556b31ae7190 from dir: /tmp/.mount_GyroflopYO3O//lib/

(gyroflow:10756): dbind-WARNING **: 17:32:24.398: AT-SPI: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
15:32:24 [DEBUG] (1) gyroflow_core::filesystem: path_to_url | path:  -> Ok("")
15:32:34 [ERROR] thread 'main' panicked at 'Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))': /root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ocl-0.19.5/src/standard/platform.rs:49
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: main
  13: __libc_start_call_main
  14: __libc_start_main@GLIBC_2.2.5
  15: <unknown>

15:32:34 [ERROR] Failed to initialize OpenCL Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))
15:32:34 [DEBUG] (1) gyroflow_core::gpu::wgpu: WGPU adapter: AdapterInfo { name: "AMD Radeon Graphics (RADV GFX1102)", vendor: 4098, device: 29824, device_type: DiscreteGpu, driver: "radv", driver_info: "Mesa 23.2.1", backend: Vulkan }
15:32:34 [DEBUG] (1) gyroflow::rendering: GPU type: Amd, from name: amd radeon graphics (radv gfx1102)
Crashdump written to /mnt/Juegos_y_descargas/Descargas/Applications/9e2e210c-5a10-4eae-4222f584-08913e12.dmp
Violación de segmento (`core' generado)

```
And I attach the "dmp" file .... don't know if it is usefull ....
[9e2e210c-5a10-4eae-4222f584-08913e12.zip](https://github.com/gyroflow/gyroflow/files/12909963/9e2e210c-5a10-4eae-4222f584-08913e12.zip)