gyroflow/gyroflow #681: Gyroflow crashes when importing RunCam Thumb Pro videos and others

ShernCutta, 2023-07-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.2

### What operating system are you using?

Win 10

### What GPU are you using?

NVIDIA GeForce MX250

### What happened?

Anytime I try to open my RunCam Thumb Pro video, GyroFlow crashes.
It crashes when opening other, non runcam, videos as well (FIMI Xiomi V2).
But it does work for GoPro Hero 11 and iphone 10 videos.

Google Drive folder for video and gyro files, and log ==>  https://drive.google.com/drive/folders/15fMck9FVhChB7UyLLEBr1GrZiBDD6OCg?usp=sharing

AdrianEddy, 2023-07-15:
99% that it's the GPU driver. Download latest drivers for your GPU from the official website and make sure you install for both GPUs if you have integrated and dedicated

ShernCutta, 2023-07-15:
Thanks for the Reply. I just updated all of my drivers, but it's still
crashing. Does this mean my GPU is not powerful enough?

Thank You,

-- 
Best Regards,

*Shern Peters*
CEO, Founder, Developer, Inventor
email: ***@***.***
*https://www.linkedin.com/in/shernpeters/
<https://www.linkedin.com/in/shernpeters/>*
Schedule Appointment ==> calendly.com/nancoaero

<http://www.nanco.aero>
                * NANCo Aero*
*             www.NANCo.aero <http://www.NANCo.aero>*

AdrianEddy, 2023-07-15:
It's possible, it's a very old GPU. You could run Gyroflow without GPU support, by adding `set MDK_DECODERS=FFmpeg` to `Gyroflow_with_console.bat` as a second line and run this file.

You can also send me gyroflow.log file but I'm pretty sure it's the GPU fault

ShernCutta, 2023-07-15:
Understood.  I added that line to the .bat file and ran it. It was able to
load but the app froze afterwards. I ran it again and it fully loaded the
video and gyro data, and played the preview (albeit slowly). See .bat and
log file attached. I then tried to export and it crashed. The console shows
an error (see image below). My current Nvidia driver is 536.40.

[image: image.png]

-- 
Best Regards,

*Shern Peters*
CEO, Founder, Developer, Inventor
email: ***@***.***
*https://www.linkedin.com/in/shernpeters/
<https://www.linkedin.com/in/shernpeters/>*
Schedule Appointment ==> calendly.com/nancoaero

<http://www.nanco.aero>
                * NANCo Aero*
*             www.NANCo.aero <http://www.NANCo.aero>*

@echo off
set RUST_BACKTRACE=full
set MDK_DECODERS=FFmpeg
Gyroflow.exe
pause

ShernCutta, 2023-07-15:
Update:  I was able to render the video by unchecking the Use GPU encoding.

Thanks for the help.

-- 
Best Regards,

*Shern Peters*
CEO, Founder, Developer, Inventor
email: ***@***.***
*https://www.linkedin.com/in/shernpeters/
<https://www.linkedin.com/in/shernpeters/>*
Schedule Appointment ==> calendly.com/nancoaero

<http://www.nanco.aero>
                * NANCo Aero*
*             www.NANCo.aero <http://www.NANCo.aero>*

blogdemoi, 2023-08-17:
The same behaviour happened to me today with an mp4 video recorded with the Mcpro24fps app. However if I load a .mov video Gyroflow doesn't crash for me. I am running Windows 10 up to date on a laptop with two (old) NVIDIA GPUs GEFORCE GTX 880M.

AdrianEddy, 2023-08-21:
Please check latest dev build now and let me know if it works https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-windows.zip

AdrianEddy, 2023-08-22:
Should be fixed in latest dev build https://gyroflow.xyz/devbuild/