gyroflow/gyroflow #683: Building 1.5.2 with opencv 4.8 failed, opencv-rust 0.83 is needed.

AlynxZhou, 2023-07-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.2

### What operating system are you using?

Arch Linux, rolling

### What GPU are you using?

NVIDIA

### What happened?

I am trying to package gyroflow for AUR with the latest 1.5.2 version, and I got following error while building:

```
error[E0119]: conflicting implementations of trait `Debug` for type `RotatedRect`
     --> /home/alynx/aur/gyroflow/src/gyroflow-1.5.2/target/release/build/opencv-a28c77986db3e6fa/out/opencv/core.rs:17629:24
      |
17629 |     #[derive(Copy, Clone, Debug, PartialEq)]
      |                           ^^^^^ conflicting implementation for `RotatedRect`
      |
     ::: /home/alynx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/opencv-0.81.5/src/manual/core/rect.rs:277:1
      |
277   | impl fmt::Debug for RotatedRect {
      | ------------------------------- first implementation here
      |
      = note: this error originates in the derive macro `Debug` (in Nightly builds, run with -Z macro-backtrace for more info)

For more information about this error, try `rustc --explain E0119`.
error: could not compile `opencv` (lib) due to previous error
warning: build failed, waiting for other jobs to finish...
```

I searched about the error output, and it looks like <https://github.com/twistedfall/opencv-rust/issues/472#issuecomment-1634229266>, opencv-rust says version 0.83 is needed to support opencv 4.8, which is already in Arch Linux's official repo.

I am not so familiar with Rust, could we consider bumping the dependency version and make a new patch release?

AlynxZhou, 2023-07-17:
I manually created a patch to bump the opencv-rust version:

```
diff --unified --recursive --text --new-file --color gyroflow-1.5.2.old/src/core/Cargo.toml gyroflow-1.5.2.new/src/core/Cargo.toml
--- gyroflow-1.5.2.old/src/core/Cargo.toml	2023-07-13 03:55:46.000000000 +0800
+++ gyroflow-1.5.2.new/src/core/Cargo.toml	2023-07-17 15:13:40.211660037 +0800
@@ -61,7 +61,7 @@
 libloading = "0.8"
 libc = "0.2"
 
-opencv = { version = "0.81", features = ["clang-runtime"], optional = true }
+opencv = { version = "0.83", features = ["clang-runtime"], optional = true }
 
 wgpu       = { git = "https://github.com/gfx-rs/wgpu.git", rev = "89f721f", features = ["spirv"] }
 wgpu-hal   = { git = "https://github.com/gfx-rs/wgpu.git", rev = "89f721f" }
```

but then I got following error:

```
error[E0432]: unresolved import `opencv::prelude::DenseOpticalFlow`
  --> src/core/synchronization/optical_flow/opencv_dis.rs:12:78
   |
12 | use opencv::{ core::{ Mat, Size, CV_8UC1, Vec2f }, prelude::{ MatTraitConst, DenseOpticalFlow } };
   |                                                                              ^^^^^^^^^^^^^^^^
   |                                                                              |
   |                                                                              no `DenseOpticalFlow` in `prelude`
   |                                                                              help: a similar name exists in the module: `DenseOpticalFlowTrait`
   |
   = help: consider importing this struct instead:
           opencv::video::DenseOpticalFlow

error[E0404]: expected trait, found struct `opencv::video::DISOpticalFlow`
  --> src/core/synchronization/optical_flow/opencv_dis.rs:56:40
   |
56 |                 let mut optflow = <dyn opencv::video::DISOpticalFlow>::create(opencv::video::DISOpticalFlow_PRESET_FAST)?;
   |                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not a trait

Some errors have detailed explanations: E0404, E0432.
For more information about an error, try `rustc --explain E0404`.
error: could not compile `gyroflow-core` (lib) due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
```

AdrianEddy, 2023-07-17:
Should be fixed as of caf5089

closed 2023-07-17 by AdrianEddy