gyroflow/gyroflow #686: Gyroflow Crashed when open Movie

danilosousadba, 2023-07-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.2 (gh1321)

### What operating system are you using?

macOS Ventura 13.4.1 (c) (22F770820d)

### What GPU are you using?

Radeon Pro 560 4 GB Intel HD Graphics 630 1536 MB

### What happened?

Gyroflow opens normally but when selecting any video it crashes until the process is killed manually

### Relevant log output

```shell
Date/Time:        2023-07-19 22:32:40.143 -0300
End time:         2023-07-19 22:33:54.092 -0300
OS Version:       macOS 13.4.1 (Build 22F82)
Architecture:     x86_64h
Report Version:   40
Incident Identifier: E3AE74CD-2E7B-4624-9E3C-C413E9C39EF5

Data Source:      Microstackshots
Shared Cache:     E15DC6AC-9320-3B97-95DE-B6E917A43EC4 slid base address 0x7ff819179000, slide 0x19179000

Command:          gyroflow
Path:             /Volumes/*/Gyroflow.app/Contents/MacOS/gyroflow
Identifier:       xyz.gyroflow
Version:          1.5.2 (1.5.2)
Team ID:          RR3ZC2L4DF
Is First Party:   No
Resource Coalition ID: 9252
Architecture:     x86_64
PID:              8887

Event:            wakeups
Action taken:     none
Wakeups:          45002 wakeups over the last 74 seconds (609 wakeups per second average), exceeding limit of 150 wakeups per second over 300 seconds
Wakeups limit:    45000
Limit duration:   300s
Wakeups caused:   45002
Wakeups duration: 74s
Duration:         73.95s
Duration Sampled: 17.64s
Steps:            3

Hardware model:   MacBookPro14,3
Active cpus:      8
HW page size:     4096
VM page size:     4096

Fan speed:        2167 rpm
OS Cryptex File Extents: 8065

Heaviest stack for the target process:
  3  thread_start + 15 (libsystem_pthread.dylib + 7123) [0x7ff81956cbd3]
  3  _pthread_start + 125 (libsystem_pthread.dylib + 25043) [0x7ff8195711d3]
  3  ??? (QtCore + 2067843) [0x111a5dd83]
  3  ??? (QtQuick + 2818577) [0x1142b3211]
  3  ??? (QtQuick + 2814660) [0x1142b22c4]
  3  QQuickWindowPrivate::renderSceneGraph(QSize const&, QSize const&) + 291 (QtQuick + 1047363) [0x114102b43]
  3  ??? (QtCore + 774885) [0x1119222e5]
  3  ??? (gyroflow + 28572247) [0x10fb1fa57]
  3  ??? (gyroflow + 28637099) [0x10fb2f7ab]
  3  ??? (gyroflow + 28602470) [0x10fb27066]
  3  ??? (gyroflow + 6559182) [0x10e6215ce]
  3  ??? (gyroflow + 2535866) [0x10e24b1ba]
  3  ??? (gyroflow + 391643) [0x10e03f9db]
  3  ??? (gyroflow + 251714) [0x10e01d742]
  3  ??? (gyroflow + 5376593) [0x10e500a51]
  3  ??? (gyroflow + 9981010) [0x10e964c52]
  3  ??? (gyroflow + 9475601) [0x10e8e9611]
  3  ??? (gyroflow + 8951650) [0x10e869762]
  3  ??? (gyroflow + 10756368) [0x10ea22110]
  2  ??? (libsystem_c.dylib + 62748) [0x7ff81942751c]
  2  clock_get_time + 89 (libsystem_kernel.dylib + 23123) [0x7ff819536a53]
  2  mach_msg2_trap + 10 (libsystem_kernel.dylib + 5554) [0x7ff8195325b2]

Powerstats for:   gyroflow (Gyroflow) [8887]
UUID:             0A66C8A4-5D1A-31DF-96FB-890D42C239FF
Path:             /Volumes/*/Gyroflow.app/Contents/MacOS/gyroflow
Identifier:       xyz.gyroflow
Version:          1.5.2 (1.5.2)
Team ID:          RR3ZC2L4DF
Is First Party:   No
Resource Coalition ID: 9252
Architecture:     x86_64
Footprint:        373.20 MB
Pageins:          1 pages
Start time:       2023-07-19 22:33:35.928 -0300
End time:         2023-07-19 22:33:53.567 -0300
Num samples:      3 (100%)
Primary state:    3 samples Non-Frontmost App, Non-Suppressed, Kernel mode, Effective Thread QoS Default, Requested Thread QoS Default, Override Thread QoS Unspecified
User Activity:    0 samples Idle, 3 samples Active
Power Source:     3 samples on Battery, 0 samples on AC
  3  thread_start + 15 (libsystem_pthread.dylib + 7123) [0x7ff81956cbd3]
    3  _pthread_start + 125 (libsystem_pthread.dylib + 25043) [0x7ff8195711d3]
      3  ??? (QtCore + 2067843) [0x111a5dd83]
        3  ??? (QtQuick + 2818577) [0x1142b3211]
          3  ??? (QtQuick + 2814660) [0x1142b22c4]
            3  QQuickWindowPrivate::renderSceneGraph(QSize const&, QSize const&) + 291 (QtQuick + 1047363) [0x114102b43]
              3  ??? (QtCore + 774885) [0x1119222e5]
                3  ??? (gyroflow + 28572247) [0x10fb1fa57]
                  3  ??? (gyroflow + 28637099) [0x10fb2f7ab]
                    3  ??? (gyroflow + 28602470) [0x10fb27066]
                      3  ??? (gyroflow + 6559182) [0x10e6215ce]
                        3  ??? (gyroflow + 2535866) [0x10e24b1ba]
                          3  ??? (gyroflow + 391643) [0x10e03f9db]
                            3  ??? (gyroflow + 251714) [0x10e01d742]
                              3  ??? (gyroflow + 5376593) [0x10e500a51]
                                3  ??? (gyroflow + 9981010) [0x10e964c52]
                                  3  ??? (gyroflow + 9475601) [0x10e8e9611]
                                    3  ??? (gyroflow + 8951650) [0x10e869762]
                                      3  ??? (gyroflow + 10756368) [0x10ea22110]
                                        2  ??? (libsystem_c.dylib + 62748) [0x7ff81942751c]
                                          2  clock_get_time + 89 (libsystem_kernel.dylib + 23123) [0x7ff819536a53]
                                            2  mach_msg2_trap + 10 (libsystem_kernel.dylib + 5554) [0x7ff8195325b2]
                                        1  __semwait_signal + 10 (libsystem_kernel.dylib + 16318) [0x7ff819534fbe]

  Binary Images:
           0x10dfe0000 -        0x1116b7fff  xyz.gyroflow 1.5.2 (1.5.2)          <0A66C8A4-5D1A-31DF-96FB-890D42C239FF>  /Volumes/*/Gyroflow.app/Contents/MacOS/gyroflow
           0x111865000 -        0x111d74fff  org.qt-project.QtCore 6.4 (6.4.3)   <6238782E-D1AA-3245-81EF-2925BF1A4C96>  /Applications/Gyroflow.app/Contents/Frameworks/QtCore.framework/Versions/A/QtCore
           0x114003000 -        0x114452fff  org.qt-project.QtQuick 6.4 (6.4.3)  <2320A4FF-4480-3FAB-B65E-D0BFA30E7175>  /Applications/Gyroflow.app/Contents/Frameworks/QtQuick.framework/Versions/A/QtQuick
        0x7ff819418000 -     0x7ff81949fff7  libsystem_c.dylib (1534.100.14)     <5CA0D0CF-FCDD-3180-AF7B-F137FE1E7ACA>  /usr/lib/system/libsystem_c.dylib
        0x7ff819531000 -     0x7ff81956aff7  libsystem_kernel.dylib (8796.121.3) <EB4E80A0-99DA-32DC-B9AD-394FBB50A0AC>  /usr/lib/system/libsystem_kernel.dylib
        0x7ff81956b000 -     0x7ff819576fff  libsystem_pthread.dylib (514.100.2) <8CB0B396-011F-31F2-8915-94CDE1ABAE84>  /usr/lib/system/libsystem_pthread.dylib

--------

Console Logs:
Prompting policy for hardened runtime; service: kTCCServiceAppleEvents requires entitlement com.apple.security.automation.apple-events but it is missing for responsible={TCCDProcess: identifier=xyz.gyroflow, pid=565, auid=501, euid=501, responsible_path=/Applications/Gyroflow.app/Contents/MacOS/gyroflow, binary_path=/Applications/Gyroflow.app/Contents/MacOS/gyroflow}, accessing={TCCDProcess: identifier=com.apple.quicklook.QuickLookUIService, pid=1091, auid=501, euid=501, binary_path=/System/Library/Frameworks/QuickLookUI.framework/Versions/A/XPCServices/QuickLookUIService.xpc/Contents/MacOS/QuickLookUIService}, requesting={TCCDProcess: identifier=com.apple.appleeventsd, pid=480, auid=55, euid=55, binary_path=/System/Library/CoreServices/appleeventsd},
---
AddInstanceForFactory: No factory registered for id <CFUUID 0x6000019a8b60> F8BB1C28-BAE8-11D6-9C31-00039315CD46
---
Execution of the command buffer was aborted due to an error during execution. <private>
---
RBSStateCapture remove item called for untracked item 231-166-1464 (target:[app<application.xyz.gyroflow.137489265.137492873(501)>:565])
----
RBSStateCapture remove item called for untracked item 231-138-771 (target:[xpcservice<com.apple.appkit.xpc.openAndSavePanelService([app<application.xyz.gyroflow.137489265.137492873(501)>:565])(501)>:1090:1090])
```

AdrianEddy, 2023-08-17:
Can you try if it works in latest dev build? https://gyroflow.xyz/devbuild/

AdrianEddy, 2023-08-21:
Please check latest dev build now and let me know if it works https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-macos.zip

AdrianEddy, 2023-08-22:
Should be fixed in latest dev build https://gyroflow.xyz/devbuild/

closed 2023-08-22 by AdrianEddy