gyroflow/gyroflow #688: Would like to specify the output container in CLI mode

cprima, 2023-07-24:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

As a user of the batch-features in CLI mode, I would like to specify the output container or have the output path exposed.

Usecase: I have ~1250 GoPro files in mp4 container with H265 codec with 10bit (Hero11) containing the gyro data.

Problem:
- To stabilize while keeping the H265 I need to manually enter the mov extension in the GUI. Tedious for the amount of files. I CANNOT keep the mp4 container with H265 because mov is required as per screenshot below:
[image]
- I cannot preserve other tracks with H265 into mp4. The CLI says: "Could not find tag for codec none in stream #2, codec not currently supported in container" (I think the GoPro vendorspecific stream with the metadata is the culprit)
- I can hardly stabilize into mp4 with DNxHR HQ (for 10 bit) because that is 4x the file size.
- I cannot specify the extension or the full path of the output file in the CLI mode.
- I can also not re-containerize in ffmpeg before using Gyroflow as that would remove the gyrodata.

Seems that at the moment all I can do is via the GUI changing the output path and filling the render queue.

AdrianEddy, 2023-07-24:
You can do anything in CLI you can do in GUI by using the preset, which can either be a file, or preset contents directly from the command line.
Example:
```
--preset "{ 'version': 2, 'output': { 'output_path': 'path_to_output.mov' } }"
```

but for output path specifically, you can use `--out_params "{ 'output_path': 'path_to_output.mov' }"`

Let me know if that works for you

cprima, 2023-07-24:
Hi @AdrianEddy 

I had tried that syntax before, but it wasn't working. Based on your suggestion I tried harder ;)

The only way I can get it to work with my input files (https://drive.google.com/file/d/1OTw3Lc_WkdNFimFATewjVbbtYKIS2HPX/view?usp=sharing is one of them) is

1. I create a preset file via the GUI
1. I remove the line with the output_path
1. I place that (for the duration of this project) in the camera_presets folder as default.gyroflow
1. feed with a PowerShell script the input filename to Gyroflow.exe

Read careful below: The output seems hardcoded T:/myfile.mov
But actually Gyroflow does only honor the folder, and creates filenames as per input file, and with the mov extension. So there is no file named myfile.mov

I had also tried various combinations on the commandline with --preset and --out-params (hyphen, not underscore) and did not get the desired result. The "merging" of --out-params with a --preset file I did not get to work

So the workaround for me is that the following script does use a default.gyroflow file with the output_folder setting removed:

```$basePath = "A:\cpm_travel_2023.ingest\"
$inputFiles = get-childitem -Include @("*.mp4") -Path $basePath -Recurse
foreach ($singleFile in $inputFiles) {
    Write-Output "$singleFile"
    Start-Process -FilePath "O:\Gyroflow-1.5.2\Gyroflow.exe" -ArgumentList "$singleFile --export-project 4 --out-params `"{ 'output_path': 'T:/myfile.mov' }`" -f "  -Wait -NoNewWindow
}```

AdrianEddy, 2023-07-24:
Ok, it may be a bug then, I'll try to check it out in a few days

cprima, 2023-07-24:
As so often in IT, your quick reply helped me to narrow the problem down. I would not have found the workaround without.

I also had another problem that I only realized when I troubleshooted the above: "Something" is wrong with "preserve_other_tracks": true and my specific input files. Seems that my GoPro Hero11 writes a codec to one of the tracks that Gyroflow does not understand.

But for the moment my PC is busy with 4 jobs of that PowerShell script, I cannot properly reproduce it.

AdrianEddy, 2023-09-20:
Should be done in 462ac7a06f2a978e504efb512845e91fd1c018b1
In latest dev build, I added `output_extension` field to the `output` object in `preset` or `out_params`, which you can set to `mov`
Also, `output_path` was replaced by `output_folder` and `output_filename`

closed 2023-09-20 by AdrianEddy