gyroflow/gyroflow #689: color layer not stabilized when lense profile is not applied

cqw0630, 2023-07-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.2

### What operating system are you using?

Windows and Mac

### What happened?

Camera: sony zve10
Video: 4k 25fps
Encoding: H264 53.37Mbps
Pixel format: YUV420P 8 bit
Version: 1.5.2 and nightly gh1323

In v1.5.2, when no lense profile is selected, exported stabilized video would look like b&w and color layer separated. b&w layer is stabilized while color layer is not. I've tried the same media on v1.5.1, it does not have this issue. please check below screenshot.

And if I applied any lense profile in v1.5.2, the issue is gone.

[image]

Please let me know if I need to provide any other info, thanks!

AdrianEddy, 2023-07-26:
Hmm, can you share the sample file? I know why it may be happening but I can't reproduce with my files from a7sIII

cqw0630, 2023-07-26:
A very short clip, but should be able to be reproduced with.

https://github.com/gyroflow/gyroflow/assets/15188098/bfb00ab0-e30f-43ef-8649-43a712541321

AdrianEddy, 2023-07-30:
Fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload

llWHiTE, 2024-02-29:
Hello.
Similar error on sony a7r5 when shooting at 8k 25fps
Cant upload file here cuz of file size:(
https://drive.google.com/file/d/1TFkWfsb8h4ghh9rx6UahEabRFjSFGThn/view?usp=sharing

AdrianEddy, 2024-02-29:
@llWHiTE Did you test in the latest dev build? https://gyroflow.xyz/devbuild/

AdrianEddy, 2024-03-12:
@llWHiTE fixed in latest dev build https://gyroflow.xyz/devbuild/

closed 2024-03-12 by AdrianEddy