gyroflow/gyroflow #690: Version 1.5.2 gyroflow.exe on win64 build is possibly infected by a trojan

empperi, 2023-07-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.5.2

### What operating system are you using?

Windows 11

### What GPU are you using?

RTX 3080

### What happened?

Latest version of `gyroflow.exe` is flagged to have `Bearfoos.B!ml` trojan by Windows Defender. This might be a false positive or it might be accurate but in any case, I suggest devs check this out as soon as possible.

[image]

AdrianEddy, 2023-07-27:
Hmm, VirusTotal doeyst detect anything, it might either be a false positive or something infected it on your system 
https://www.virustotal.com/gui/file/ecd8636a6e4d92215f16e7931e7a8dadc7adc277a19d840406fbc4796e564c39

empperi, 2023-07-27:
> Hmm, VirusTotal doesn't detect anything, it might either be a false positive or something infected it on your system 
> https://www.virustotal.com/gui/file/ecd8636a6e4d92215f16e7931e7a8dadc7adc277a19d840406fbc4796e564c39

Very possible especially on the false positive side. I'm highly doubtful on it having been infected by something on my system considering Defender flagged it immediately and it hasn't done that on anything else on this computer. And I don't install or download random stuff to that computer :)

It is definitely not unheard of that virus scanners do false positives. I'll skip this version for now to be on the safe side and wait for the next version.

AdrianEddy, 2023-07-27:
You can check the Dev build to see if it also detects there https://gyroflow.xyz/devbuild?autodownload

empperi, 2023-07-27:
Dev build seems to be clean. This would most likely point towards false positive.

closed 2023-07-28 by AdrianEddy