gyroflow/gyroflow #692: Implement CineForm (cfhd) codec

victorlysak, 2023-07-30:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Gyroflow works with ffmpeg which supports cfhd encoding.

Please, would it be possible to implement this coded as well?

Also:
When encoding to hevc/x265, using nvidia, Gyroflow uses only 9% of my CPU.
I wander, if is there any option to utilise the whole or max CPU usage.

When I do the encoding with ffmpeg using:
ffmpeg -i input.mp4 -c:v hevc_nvenc -qp 12 output.mp4
than I get 100% CPU usage and the speed is ~30 fps
But with Gyroflow the encoding to the same x265 codec
uses only 9% of CPU and goes with the max speed of 17 fps.

Thank you very much.

AdrianEddy, 2023-07-30:
Were using ffmpeg lite from avbuild which doesn't have cineform encoder included. Does it have an advantage over ProRes? What would be your use case?

As for the rendering speed, the CPU shouldn't really be used, in Gyroflow there's GPU decoding and GPU encoding, both have checkboxes you can switch in the UI. Your command line doesn't use GPU decoding, that's why you see 100% CPU usage. Add `-hwaccel cuda` to your command line to use GPU decoding. Or disable GPU decoding in Gyroflow to use CPU decoding there.
Another factor is that Gyroflow doesn't use zero-copy rendering yet because the graphics API interoperability is very hard. This is tracked in #555 .
On windows and Nvidia, switching Device for video processing to wgpu + Vulkan is the fastest so it should give you a few more rendering fps

victorlysak, 2023-07-30:
OK. First I'll replay to your comment and then give my case.

system Arch Linux (kernel 6.4.7)

In Gyroflow I did the video 5312x2988 50 fps, with stab. setting of 0.25, and the only custom option `-force_key_frames 10`
The video is 32 seconds long.
Gyroflow did the video in 1m32 seconds with the speed of 17 fps.

After that I took the same stabilised file and did it with ffmpeg,
right after ffmpeg I added
-hwaccel cuda 
and the full command line was:
`time ffmpeg -hwaccel cuda -i 06-24--18-18--GX010189_stab.mp4 -c:v hevc_nvenc -profile:v 1 -qp 10 -force_key_frames 10 -color_range pc 06-24--18-18--GX010189_stab-nvenc.mp4`

the time taken to process was 32 seconds with the speed of 50 fps
the CPU load was 93%, the GPU load (accordin to nvtop) was 42%.

AdrianEddy, 2023-07-30:
What device for video rendering you have set in Gyroflow? Try others.
Gyroflow will render faster once zero-copy is implemented but that's a lot of work, so it will be at least few months

victorlysak, 2023-07-30:
my case is:

videos from GoPro 11 Black
all the videos are 5.3k (5312x2988) 50 fps, 10 bit

I can do them in Vegas, or in Resolve Studio.

my dilemma is that I need the stabil. videos in the same format and in full color range...
to work well with the files in Vegas the best ever and of all codecs is Lagarith,
because it plays absolutely smoothly, has rgb format, no more that 2 gbit/sec, and has full color range.
Gyroflow uses ffmpeg and therefore doesn't and will not do Lagarith.
Resolve will not even think of accepting Lagarith files :), it won't even load them.

So when I don't need to stabilise the videos, I convert them into Lagarith in VirtualDub2 in Win10, and work in Vegas Pro 18.
Or I convert them also in GoPro player into CineForm which will give me the full color range, and they will be accepted and well played in Resolve, but not in Vegas
VirtualDub2 can also convert them to CineForm and if saved in .avi format, will play well in Vegas. but this means that after converting with GP Player, I have to run the file again in VirtualDub2 :) :) :) 

If I need to stabilise, I use Gyroflow.
ProRes will not give full color range, unless the encoding is with 4444 pix format. 
The same applies to DNxHD codec.
In both cases the files come as 5 gbit/sec, which makes Vegas and Resolve struggle badly and I basically get the same struggle as working with x265 files. 

So, as you see, I am constantly looking for something best that would be smooth in either Vegas or Resolve and let me do the editing without frustration.
I need the files in full color range.

So far I came out with exporting videos from Gyroflow in x265 format and I am tryring to make them as editor freandly as possible, and therefore I try options like  -force_key_frames to get better seeking in the editors,
and they are more or less OK in Resolve or Vegas,
but if I start adding colors and effects, Resolve starts struggling and Vegas will start crashing during rendering.

Thank you very much.

victorlysak, 2023-07-30:
> What device for video processing you have set in Gyroflow? Try others. Gyroflow will render faster once zero-copy is implemented but that's a lot of work, so it will be at least few months

In the above cases I used the device chosen by Gyroflow by default, which is
NVIDIA CUDA NVIDIA GeForce RTX 2060 SUPER 3.0

now I changed to 
NVIDIA GeForce RTX 2060 SUPER 3.0
the same result - same render time and speed,

than I tried the 3rd and the last option:
NVIDIA GeForce RTX 2060 SUPER/PCIe/SSE2
the same result - same render time and speed.

thank you very much

victorlysak, 2023-07-30:
I also have to say that when I convert these files with GP Player into CineForm format (in Windows of course), it does it with fully utilising my GPU and CPU.

I choose only profile high and the files come out  in 10 bit full color range with the size of about 1.2 gbit/sec max only,
which is very good for the editors and they play smoothly in Resolve even when I add more effects and do color grading.

I cannot yet achieve this in cfhd with ffmpeg, because ffmpeg with option 8 (high) can give me only limited color range.
I can only get full color range in ffmeg cfhd when using higher quality profile and that makes file go over 2.5 gbit/sec.
... but I keep trying different options :)

thank you very much

AdrianEddy, 2023-07-30:
I was talking about "Device for video processing" which is in Advanced and should have either OpenCL or wgpu options. The fastest one should be wgpu + Vulkan

victorlysak, 2023-07-30:
> I was talking about "Device for video processing" which is in Advanced and should have either OpenCL or wgpu options. The fastest one should be wgpu + Vulkan

Yes. This is better.

So, the default was set to 
[OpenCL] NVIDIA CUDA NVIDIA GeForce RTX 2060 SUPER 3.0
and Preview Pipeline is
Zero-copy Qt RHI

now trying
[wgpu] NVIDIA CUDA NVIDIA GeForce RTX 2060 SUPER (Vulkan)
speed: 23 fps, done in 1m 05 seconds
and Preview Pipeline is
Zero-copy Qt RHI

now trying
[wgpu] NVIDIA CUDA NVIDIA GeForce RTX 2060 SUPER/PCIe/SSE2 (Gl)
speed: 1.2 fps, done in 20 m 10 seconds :)
and Preview Pipeline is
Zero-copy Qt RHI

now switched back to the fastest
[wgpu] NVIDIA CUDA NVIDIA GeForce RTX 2060 SUPER (Vulkan)
and Preview Pipeline is
Zero-copy Qt RHI
speed: 23 fps, done in 1m 05 seconds

now
[wgpu] NVIDIA CUDA NVIDIA GeForce RTX 2060 SUPER (Vulkan)
and Preview Pipeline is
OpenCL/wgpu/CPU
speed: 23 fps, done in 1m 05 seconds

in both cases CPU goes up to 20%

thank you very much

victorlysak, 2023-07-30:
And for your interest,

my command:
`ffmpeg -h encoder=hevc_nvenc`

outputs the following:

`ffmpeg version n6.0 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 13.1.1 (GCC) 20230429
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libdav1d --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libjack --enable-libjxl --enable-libmfx --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librav1e --enable-librsvg --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nvdec --enable-nvenc --enable-opencl --enable-opengl --enable-shared --enable-version3 --enable-vulkan
  libavutil      58.  2.100 / 58.  2.100
  libavcodec     60.  3.100 / 60.  3.100
  libavformat    60.  3.100 / 60.  3.100
  libavdevice    60.  1.100 / 60.  1.100
  libavfilter     9.  3.100 /  9.  3.100
  libswscale      7.  1.100 /  7.  1.100
  libswresample   4. 10.100 /  4. 10.100
  libpostproc    57.  1.100 / 57.  1.100
Encoder hevc_nvenc [NVIDIA NVENC hevc encoder]:
    General capabilities: dr1 delay hardware 
    Threading capabilities: none
    Supported hardware devices: cuda cuda 
    Supported pixel formats: yuv420p nv12 p010le yuv444p p016le yuv444p16le bgr0 bgra rgb0 rgba x2rgb10le x2bgr10le gbrp gbrp16le cuda
hevc_nvenc AVOptions:
  -preset            <int>        E..V....... Set the encoding preset (from 0 to 18) (default p4)
     default         0            E..V....... 
     slow            1            E..V....... hq 2 passes
     medium          2            E..V....... hq 1 pass
     fast            3            E..V....... hp 1 pass
     hp              4            E..V....... 
     hq              5            E..V....... 
     bd              6            E..V....... 
     ll              7            E..V....... low latency
     llhq            8            E..V....... low latency hq
     llhp            9            E..V....... low latency hp
     lossless        10           E..V....... lossless
     losslesshp      11           E..V....... lossless hp
     p1              12           E..V....... fastest (lowest quality)
     p2              13           E..V....... faster (lower quality)
     p3              14           E..V....... fast (low quality)
     p4              15           E..V....... medium (default)
     p5              16           E..V....... slow (good quality)
     p6              17           E..V....... slower (better quality)
     p7              18           E..V....... slowest (best quality)
  -tune              <int>        E..V....... Set the encoding tuning info (from 1 to 4) (default hq)
     hq              1            E..V....... High quality
     ll              2            E..V....... Low latency
     ull             3            E..V....... Ultra low latency
     lossless        4            E..V....... Lossless
  -profile           <int>        E..V....... Set the encoding profile (from 0 to 4) (default main)
     main            0            E..V....... 
     main10          1            E..V....... 
     rext            2            E..V....... 
  -level             <int>        E..V....... Set the encoding level restriction (from 0 to 186) (default auto)
     auto            0            E..V....... 
     1               30           E..V....... 
     1.0             30           E..V....... 
     2               60           E..V....... 
     2.0             60           E..V....... 
     2.1             63           E..V....... 
     3               90           E..V....... 
     3.0             90           E..V....... 
     3.1             93           E..V....... 
     4               120          E..V....... 
     4.0             120          E..V....... 
     4.1             123          E..V....... 
     5               150          E..V....... 
     5.0             150          E..V....... 
     5.1             153          E..V....... 
     5.2             156          E..V....... 
[... 39 lines cut ...]
  -init_qpP          <int>        E..V....... Initial QP value for P frame (from -1 to 51) (default -1)
  -init_qpB          <int>        E..V....... Initial QP value for B frame (from -1 to 51) (default -1)
  -init_qpI          <int>        E..V....... Initial QP value for I frame (from -1 to 51) (default -1)
  -qp                <int>        E..V....... Constant quantization parameter rate control method (from -1 to 51) (default -1)
  -qp_cb_offset      <int>        E..V....... Quantization parameter offset for cb channel (from -12 to 12) (default 0)
  -qp_cr_offset      <int>        E..V....... Quantization parameter offset for cr channel (from -12 to 12) (default 0)
  -weighted_pred     <int>        E..V....... Set 1 to enable weighted prediction (from 0 to 1) (default 0)
[... cut ...]
  -extra_sei         <boolean>    E..V....... Pass on extra SEI data (e.g. a53 cc) to be included in the bitstream (default true)
  -udu_sei           <boolean>    E..V....... Pass on user data unregistered SEI if available (default false)
  -intra-refresh     <boolean>    E..V....... Use Periodic Intra Refresh instead of IDR frames (default false)
  -single-slice-intra-refresh <boolean>    E..V....... Use single slice intra refresh (default false)
  -constrained-encoding <boolean>    E..V....... Enable constrainedFrame encoding where each slice in the constrained picture is independent of other slices (default false)
`

latenitefilms, 2023-07-30:
Why not just use the OpenFX Plugin in DaVinci Resolve or Vegas?

victorlysak, 2023-07-31:
> Why not just use the OpenFX Plugin in DaVinci Resolve or Vegas?

Hi.
Thank you for the comment.

Trying...
put Gyroflow.ofx.bundle into /usr/OFX/Plugins

opened Resolve... no Gyroflow plugin... in the Prefs., Video plugins, it says Gyroflow | Failed to load

victorlysak, 2023-07-31:
OK, there is some success with it in Windows, but only partial.

In Vegas and Resolve the plugin loads and works well but only on the original .mp4 files, not on the transcoded (cfhd, dnxhd, prores ..).

Tried different options, including saving the project in Gyroflow and then loading the file in Resolve.
Tried to copy plugin data from orig. file and applied to the cfhd, but no success. The whole video shakes like mad.

This is just to report for your info.

Also, maybe someone knows better about Gyroflow issues in kde plasma on wayland.
A couple of weeks ago I didn't have Gyroflow working, but after setting QT_QPA_PLATFORM=wayland, it began to work.
However, after updating kde plasma, the issue re-appeared, and now when starting Gyroflow, I see only a lot of mouse trails all over the screen.
Setting the above variable to xcb works, but then my whole plasma becomes very slow.
I think there are some issues with QT in wayland.
I have qt6 and qt5 installed.
Does anything else has to be done.

thank you all very much

AdrianEddy, 2023-08-20:
CineForm is added in the latest dev build https://gyroflow.xyz/devbuild/

closed 2023-08-20 by AdrianEddy

victorlysak, 2023-08-20:
thank you very much indeed

victorlysak, 2023-08-21:
Tried both, .AppImage and the executable.

In both cases the prog. loads ok, but it halts when opening the file.
It basically says "loading the file ..." and nothing happens even after half an hour waiting.

Thank you.

AdrianEddy, 2023-08-21:
you mean when loading CineForm video in Gyroflow? What's in gyroflow.log file?

victorlysak, 2023-08-21:
No. It happens when loading the .mp4 (x265) video - original file from the cam.

thank you

AdrianEddy, 2023-08-21:
What about now? I just updated latest dev build a few minutes ago https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-linux.zip

AdrianEddy, 2023-08-21:
please upload the gyroflow.log file

victorlysak, 2023-08-21:
Here:

17:48:10 [INFO] [MDK] 0.21.1 (git ac889b7) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang17.0.0 (++20230818083155+242fe2d8d86e-1~exp1~20230818083303.26)GCC4.2.1; libc++170000; 15:37:09 Aug 21 2023
17:48:10 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
17:48:10 [DEBUG] (1) gyroflow: Gyroflow 1.5.2 (gh1420)
17:48:10 [DEBUG] (1) gyroflow: Executable path: Some("/home/vic/Downloads/Gyroflow-linux (1)/Gyroflow/gyroflow")
17:48:10 [DEBUG] (1) gyroflow: SDK path: "/home/vic/Downloads/Gyroflow-linux (1)/Gyroflow/lib/"
17:48:10 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
17:48:10 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
17:48:10 [DEBUG] (1) mdk: [MDK] key time: -1
17:48:10 [DEBUG] (1) mdk: [MDK] check version
17:48:10 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
17:48:10 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.2.135") ["NVIDIA GeForce RTX 2060 SUPER"]
17:48:10 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("rusticl") Ok("OpenCL 3.0") []
17:48:10 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Clover") Ok("OpenCL 1.1 Mesa 23.1.5") []
17:48:10 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_opaque_fd cl_khr_external_memory_opaque_fd" } Device: NVIDIA Corporation NVIDIA GeForce RTX 2060 SUPER
17:48:11 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
17:48:11 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
17:48:11 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
17:48:11 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 2060 super
17:48:11 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA NVIDIA GeForce RTX 2060 SUPER: 3.0", "[wgpu] NVIDIA GeForce RTX 2060 SUPER (Vulkan)", "[wgpu] NVIDIA GeForce RTX 2060 SUPER/PCIe/SSE2 (Gl)"]
17:48:12 [INFO] Lens profiles directory: "/home/vic/Downloads/Gyroflow-linux (1)/Gyroflow/camera_presets"
17:48:12 [INFO] Loaded 7460 lens profiles in 165.814ms
17:48:17 [DEBUG] (1) mdk: [MDK] default0x559462b48e60 new FrameReader...
17:48:17 [DEBUG] (1) mdk: [MDK] Registered audio backends: ALSA OpenAL PulseAudio null
17:48:17 [DEBUG] (1) mdk: [MDK] ALSA device name: alsa_playback.gyroflow
17:48:17 [DEBUG] (1) mdk: [MDK] ALSA device id: alsa_playback.gyroflow
17:48:17 [DEBUG] (1) mdk: [MDK] ALSA subdevice name: alsa_playback.gyroflow
17:48:17 [DEBUG] (1) mdk: [MDK] default0x559462b46aa0 new FrameReader...
17:48:17 [DEBUG] (1) mdk: [MDK] Registered audio backends: ALSA OpenAL PulseAudio null
17:48:17 [DEBUG] (1) mdk: [MDK] ALSA device name: alsa_playback.gyroflow
17:48:17 [DEBUG] (1) mdk: [MDK] ALSA device id: alsa_playback.gyroflow
17:48:17 [DEBUG] (1) mdk: [MDK] ALSA subdevice name: alsa_playback.gyroflow
17:48:17 [DEBUG] (1) mdk: [MDK] 0x55945ebc8980mdk::Player::Player()
17:48:17 [DEBUG] (1) MDKPlayer: [in setUrl] Final path: "/home/vic/BACKUP/eurodrive/144-galway/06-19--15-32--End-Galway.MP4"
17:48:17 [DEBUG] (1) mdk: [MDK] default FrameReader0x559462b48e60 state: 0=>0=>0, 0
17:48:17 [INFO] [MDK] default0x559462b48e60 virtual void mdk::FrameReader::stop(), current state: 0
17:48:17 [DEBUG] (1) mdk: [MDK] default FrameReader0x559462b48e60 state: 0=>0=>0, 0
17:48:17 [INFO] [MDK] default0x559462b48e60 virtual void mdk::FrameReader::stop(), current state: 0
17:48:17 [DEBUG] (1) mdk: [MDK] AudioBackendALSA ERROR@285>>> snd_pcm_pause(pcm_, value) (ffffffb3): File descriptor in bad state
17:48:17 [DEBUG] (1) mdk: [MDK] 
17:48:17 [INFO] [MDK] 0x559462556ca0 MediaControl.prepare(0, ...) /home/vic/BACKUP/eurodrive/144-galway/06-19--15-32--End-Galway.MP4
17:48:17 [DEBUG] (1) mdk: [MDK] default 0x559462b48e60 FrameReader.start(0, ...)
17:48:17 [DEBUG] (1) mdk: [MDK] default 0x559462b48e60 FrameReader::update MediaStatus 0=>0X2
17:48:17 [DEBUG] (1) mdk: [MDK] default FrameReader0x559462b48e60 state: 0=>0=>1, 0
17:48:17 [DEBUG] (1) mdk: [MDK] default FrameReader0x559462b48e60 state: 0=>1=>1, 0
17:48:17 [DEBUG] (14) mdk: [MDK] start frame reader thread: 140631473116864
17:48:17 [DEBUG] (14) mdk: [MDK] try to load av module: /home/vic/Downloads/Gyroflow-linux (1)/Gyroflow/lib/libffmpeg.so.6
17:48:17 [DEBUG] (14) mdk: [MDK] dlopen error: /home/vic/Downloads/Gyroflow-linux (1)/Gyroflow/lib/libffmpeg.so.6: cannot open shared object file: No such file or directory
17:48:17 [DEBUG] (14) mdk: [MDK] try to load av module: /home/vic/Downloads/Gyroflow-linux (1)/Gyroflow/lib/libavformat.so.60
17:48:17 [DEBUG] (14) mdk: [MDK] ffmpeg loaded: /home/vic/Downloads/Gyroflow-linux (1)/Gyroflow/lib/libavformat.so.60
17:48:17 [DEBUG] (14) mdk: [MDK] Selected avformat runtime version: 60.10.100 (build: 60.10.100), license: GPL version 2 or later
[... cut ...]
17:48:21 [DEBUG] (1) mdk: [MDK] default 0x559462b48e60-FrameReader::seek(1, 0X44)
17:48:21 [DEBUG] (1) mdk: [MDK] default FrameReader 0x559462b48e60 seek target skipped: 1 frequent seek requested
17:48:21 [INFO] [MDK] bad delay. written: 80 - delay = -5
17:48:21 [DEBUG] (1) mdk: [MDK] 0x559462556ca0-MediaControl::seek(pos: 0, flag:0X44 +FromNow). now: 80
17:48:21 [DEBUG] (1) mdk: [MDK] default 0x559462b48e60-FrameReader::seek(0, 0X44)
17:48:21 [DEBUG] (1) mdk: [MDK] default FrameReader 0x559462b48e60 seek target skipped: 1 frequent seek requested

AdrianEddy, 2023-08-21:
@wang-bin looks like some seeking loop?

victorlysak, 2023-08-21:
I'll keep trying tomorrow. and will let you know

thank you very much.

victorlysak, 2023-08-21:
also, here is the log from 1.5.2 (works ok)
17:52:01 [INFO] [MDK] 0.21.1 (git eee6bb2) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang16.0.6 (++20230704084245+7cbf1a259152-1~exp1~20230704204333.104)GCC4.2.1; libc++160006; 16:58:17 Jul 12 2023
17:52:01 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
17:52:01 [DEBUG] (1) gyroflow: Gyroflow 1.5.2 (gh1313)
17:52:01 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
17:52:01 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
17:52:01 [DEBUG] (1) mdk: [MDK] key time: -1
17:52:01 [DEBUG] (1) mdk: [MDK] check version
17:52:01 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
17:52:02 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.2.135") ["NVIDIA GeForce RTX 2060 SUPER"]
17:52:02 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("rusticl") Ok("OpenCL 3.0") []
17:52:02 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Clover") Ok("OpenCL 1.1 Mesa 23.1.5") []
17:52:02 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_opaque_fd cl_khr_external_memory_opaque_fd" } Device: NVIDIA Corporation NVIDIA GeForce RTX 2060 SUPER
17:52:02 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
17:52:02 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
17:52:02 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
17:52:02 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 2060 super
17:52:03 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA NVIDIA GeForce RTX 2060 SUPER: 3.0", "[wgpu] NVIDIA GeForce RTX 2060 SUPER (Vulkan)", "[wgpu] NVIDIA GeForce RTX 2060 SUPER/PCIe/SSE2 (Gl)"]
17:52:03 [INFO] Lens profiles directory: "/home/vic/Downloads/Gyroflow/camera_presets"
17:52:03 [INFO] Loaded 7460 lens profiles in 160.996ms
17:52:03 [INFO] Latest version: v1.5.2, current version: 1.5.2 (gh1313)
17:52:08 [DEBUG] (1) mdk: [MDK] default0x5579360b99c0 new FrameReader...
17:52:08 [DEBUG] (1) mdk: [MDK] Registered audio backends: ALSA OpenAL PulseAudio null
17:52:08 [DEBUG] (1) mdk: [MDK] ALSA device name: alsa_playback.gyroflow
17:52:08 [DEBUG] (1) mdk: [MDK] ALSA device id: alsa_playback.gyroflow
17:52:08 [DEBUG] (1) mdk: [MDK] ALSA subdevice name: alsa_playback.gyroflow
17:52:08 [DEBUG] (1) mdk: [MDK] default0x557931723d70 new FrameReader...
17:52:08 [DEBUG] (1) mdk: [MDK] Registered audio backends: ALSA OpenAL PulseAudio null
17:52:08 [DEBUG] (1) mdk: [MDK] ALSA device name: alsa_playback.gyroflow
17:52:08 [DEBUG] (1) mdk: [MDK] ALSA device id: alsa_playback.gyroflow
17:52:08 [DEBUG] (1) mdk: [MDK] ALSA subdevice name: alsa_playback.gyroflow
17:52:08 [DEBUG] (1) mdk: [MDK] 0x557931f6f7a0mdk::Player::Player()
17:52:08 [DEBUG] (1) MDKPlayer: [in setUrl] Final path: "/home/vic/BACKUP/eurodrive/144-galway/06-19--15-32--End-Galway.MP4"
17:52:08 [DEBUG] (1) mdk: [MDK] default FrameReader0x5579360b99c0 state: 0=>0=>0, 0
17:52:08 [INFO] [MDK] default0x5579360b99c0 virtual void mdk::FrameReader::stop(), current state: 0
17:52:08 [DEBUG] (1) mdk: [MDK] default FrameReader0x5579360b99c0 state: 0=>0=>0, 0
17:52:08 [INFO] [MDK] default0x5579360b99c0 virtual void mdk::FrameReader::stop(), current state: 0
17:52:08 [DEBUG] (1) mdk: [MDK] AudioBackendALSA ERROR@285>>> snd_pcm_pause(pcm_, value) (ffffffb3): File descriptor in bad state
17:52:08 [DEBUG] (1) mdk: [MDK] 
17:52:08 [INFO] [MDK] 0x55793610ecd0 MediaControl.prepare(0, ...) /home/vic/BACKUP/eurodrive/144-galway/06-19--15-32--End-Galway.MP4
17:52:08 [DEBUG] (1) mdk: [MDK] default 0x5579360b99c0 FrameReader.start(0, ...)
17:52:08 [DEBUG] (1) mdk: [MDK] default 0x5579360b99c0 FrameReader::update MediaStatus 0=>0X2
17:52:08 [DEBUG] (1) mdk: [MDK] default FrameReader0x5579360b99c0 state: 0=>0=>1, 0
17:52:09 [DEBUG] (1) mdk: [MDK] default FrameReader0x5579360b99c0 state: 0=>1=>1, 0
17:52:09 [DEBUG] (14) mdk: [MDK] start frame reader thread: 139743706724032
17:52:09 [DEBUG] (14) mdk: [MDK] try to load av module: /home/vic/Downloads/Gyroflow/lib/libffmpeg.so.6
17:52:09 [DEBUG] (14) mdk: [MDK] dlopen error: /home/vic/Downloads/Gyroflow/lib/libffmpeg.so.6: cannot open shared object file: No such file or directory
17:52:09 [DEBUG] (14) mdk: [MDK] try to load av module: /home/vic/Downloads/Gyroflow/lib/libavformat.so.60
17:52:09 [DEBUG] (14) mdk: [MDK] ffmpeg loaded: /home/vic/Downloads/Gyroflow/lib/libavformat.so.60
17:52:09 [DEBUG] (14) mdk: [MDK] Selected avformat runtime version: 60.4.100 (build: 60.9.100), license: GPL version 2 or later
17:52:09 [DEBUG] (14) mdk: [MDK] Selected avformat runtime configuration: --extra-version=avbuild --disable-doc --disable-debug --enable-shared --enable-runtime-cpudetect --enable-libvpl --enable-libdrm --disable-vulkan --disable-postproc --ranlib=llvm-ranlib-16 --nm=llvm-nm-16 --ar=llvm-ar-16 -- [... cut ...]
17:52:09 [DEBUG] (1) qml: [in onTelemetry_loaded] Telemetry additional data: {"camera_identifier":{"additional":"NO-EIS","brand":"GoPro","camera_setting":"","fps":50000,"identifier":"gopro-hero11black-wide-5312x2988@50000-no-eis","lens_info":"Wide","lens_model":"","model":"HERO11 Black","video_height":2988,"video_width":5312},"contains_motion":true,"contains_quats":true,"contains_raw_gyro":true,"frame_readout_time":0,"has_accurate_timestamps":true,"imu_orientation":"XYZ","sample_rate":197.23337673657073}
17:52:09 [INFO] No gravity vectors - using accelerometer
17:52:13 [DEBUG]

victorlysak, 2023-08-22:
Hi.
Just to update.

The problem seems to be only when loading the x265 file.

I tried to convert the same video with ffmpeg to dnxhr, to x264 and to cfhd.
They all, except x265, load without a problem.

thank you

AdrianEddy, 2023-08-22:
What about in latest one? (gh1422) https://gyroflow.xyz/devbuild/

victorlysak, 2023-08-22:
the same problem

I tried to run gyroflow in konsole and here is the output of the differ. stages 
look for my comments when I am loading differ. files

[vic@v1 Gyroflow]$ ./gyroflow
0.21.1 (git ac889b7) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang17.0.0 (++20230818083155+242fe2d8d86e-1~exp1~20230818083303.26)GCC4.2.1; libc++170000; 15:37:09 Aug 21 2023
13:17:42 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
13:17:42 [DEBUG] (1) gyroflow: Gyroflow 1.5.2 (gh1422)
13:17:42 [DEBUG] (1) gyroflow: Executable path: Some("/home/vic/Downloads/Gyroflow/gyroflow")
13:17:42 [DEBUG] (1) gyroflow: SDK path: "/home/vic/Downloads/Gyroflow/lib/"
mdk plugin 'mdk-braw': 0x559aab161190 from dir: /home/vic/Downloads/Gyroflow/lib/
mdk plugin 'mdk-r3d': 0x559aab161f00 from dir: /home/vic/Downloads/Gyroflow/lib/

(gyroflow:5287): Gtk-WARNING **: 14:17:42.254: Failed to parse /home/vic/.config/gtk-3.0/settings.ini: Key file does not have group “Settings”
13:17:42 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.2.135") ["NVIDIA GeForce RTX 2060 SUPER"]
13:17:42 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("rusticl") Ok("OpenCL 3.0") []
13:17:42 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Clover") Ok("OpenCL 1.1 Mesa 23.1.5") []
13:17:42 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_opaque_fd cl_khr_external_memory_opaque_fd" } Device: NVIDIA Corporation NVIDIA GeForce RTX 2060 SUPER
13:17:42 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
13:17:42 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
13:17:42 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
13:17:42 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 2060 super
13:17:43 [DEBUG] (8) gyroflow::util: Minidump uploaded: OK
13:17:43 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA NVIDIA GeForce RTX 2060 SUPER: 3.0", "[wgpu] NVIDIA GeForce RTX 2060 SUPER (Vulkan)", "[wgpu] NVIDIA GeForce RTX 2060 SUPER/PCIe/SSE2 (Gl)"]
13:17:43 [INFO] Lens profiles directory: "/home/vic/Downloads/Gyroflow/camera_presets"
13:17:44 [INFO] Loaded 7460 lens profiles in 132.836ms
13:17:44 [INFO] Latest version: v1.5.2, current version: 1.5.2 (gh1422)

=================
program loaded
=================

=========================
=========================
now opening x264 file
#########################

13:18:33 [DEBUG] (1) MDKPlayer: [in setUrl] Final path: "/home/vic/Videos/color-card-test2.mp4"
13:18:33 [DEBUG] (1) qml: Setting param smoothness 0.015
13:18:33 [DEBUG] (1) qml: Setting param smoothness_pitch 0.5
13:18:33 [DEBUG] (1) qml: Setting param smoothness_yaw 0.5
13:18:33 [DEBUG] (1) qml: Setting param smoothness_roll 0.5
13:18:33 [DEBUG] (1) qml: Setting param per_axis 0
13:18:33 [DEBUG] (1) qml: Setting param trim_range_only 0
13:18:33 [DEBUG] (1) qml: Setting param max_smoothness 1
13:18:33 [DEBUG] (1) qml: Setting param alpha_0_1s 0.1
13:18:33 [DEBUG] (14) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
13:18:33 [DEBUG] (15) MDKPlayer: [in m_player->onEvent] "thread.video" ""
13:18:33 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
13:18:33 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
13:18:33 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
13:18:33 [DEBUG] (17) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
13:18:33 [DEBUG] (17) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7fb0cc7cc930) QSize(1389, 781)
13:18:33 [DEBUG] (15) MDKPlayer: [in m_player->onEvent] "decoder.video" "FFmpeg"
13:18:33 [DEBUG] (15) MDKPlayer: [in m_player->onEvent] "decoder.video" "VDPAU"
13:18:33 [DEBUG] (15) MDKPlayer: [in m_player->onEvent] "decoder.video" "FFmpeg"
13:18:33 [DEBUG] (15) MDKPlayer: [in m_player->onEvent] "decoder.video" "VDPAU"
13:18:33 [DEBUG] (15) MDKPlayer: [in m_player->onEvent] "decoder.video" "FFmpeg"
13:18:33 [DEBUG] (18) MDKPlayer: [in m_player->onEvent] "decoder.video" "VAAPI"
13:18:33 [DEBUG] (18) MDKPlayer: [in m_player->onEvent] "decoder.video" "FFmpeg"
13:18:35 [DEBUG] (17) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
13:18:35 [INFO] surface size: 1920x1080
13:18:35 [INFO] surface size: 1920x1080
13:18:35 [DEBUG] (17) gyroflow_core::stabilization: Setting device 0
13:18:35 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce RTX 2060 SUPER
13:18:35 [INFO] Initialized OpenCL for (1389, 781, 5556) -> (1389, 781, 5556)
13:18:35 [DEBUG] (1) qml: [in onTelemetry_loaded] Telemetry additional data: {"contains_motion":false,"contains_quats":false,"contains_raw_gyro":false,"frame_readout_time":0,"has_accurate_timestamps":false,"imu_orientation":"XYZ","sample_rate":0}
[... cut ...]
13:20:21 [DEBUG] (1) qml: Setting param smoothness_yaw 0.5
13:20:21 [DEBUG] (1) qml: Setting param smoothness_roll 0.5
13:20:21 [DEBUG] (1) qml: Setting param per_axis 0
13:20:21 [DEBUG] (1) qml: Setting param trim_range_only 0
13:20:21 [DEBUG] (1) qml: Setting param max_smoothness 1
13:20:21 [DEBUG] (1) qml: Setting param alpha_0_1s 0.1
13:20:21 [DEBUG] (38) MDKPlayer: [in m_player->onEvent] "decoder.video" "CUDA"
13:20:22 [DEBUG] (34) MDKPlayer: [in m_player->onEvent] "thread.audio" ""

and this is where it halts

AdrianEddy, 2023-08-22:
Can you please upload gyroflow.log files instead of logs from console? the log file has more information

victorlysak, 2023-08-22:
Yes. Here is the latest log (it gets bigger till I close the program):

`14:00:34 [INFO] [MDK] 0.21.1 (git ac889b7) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang17.0.0 (++20230818083155+242fe2d8d86e-1~exp1~20230818083303.26)GCC4.2.1; libc++170000; 15:37:09 Aug 21 2023
14:00:34 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
14:00:34 [DEBUG] (1) gyroflow: Gyroflow 1.5.2 (gh1422)
14:00:34 [DEBUG] (1) gyroflow: Executable path: Some("/home/vic/Downloads/Gyroflow/gyroflow")
14:00:34 [DEBUG] (1) gyroflow: SDK path: "/home/vic/Downloads/Gyroflow/lib/"
14:00:34 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
14:00:34 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
14:00:34 [DEBUG] (1) mdk: [MDK] key time: -1
14:00:34 [DEBUG] (1) mdk: [MDK] check version
14:00:34 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
14:00:35 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.2.135") ["NVIDIA GeForce RTX 2060 SUPER"]
14:00:35 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("rusticl") Ok("OpenCL 3.0") []
14:00:35 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Clover") Ok("OpenCL 1.1 Mesa 23.1.5") []
14:00:35 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_opaque_fd cl_khr_external_memory_opaque_fd" } Device: NVIDIA Corporation NVIDIA GeForce RTX 2060 SUPER
14:00:35 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
14:00:35 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
14:00:35 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
14:00:35 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 2060 super
14:00:36 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA NVIDIA GeForce RTX 2060 SUPER: 3.0", "[wgpu] NVIDIA GeForce RTX 2060 SUPER (Vulkan)", "[wgpu] NVIDIA GeForce RTX 2060 SUPER/PCIe/SSE2 (Gl)"]
14:00:36 [INFO] Lens profiles directory: "/home/vic/Downloads/Gyroflow/camera_presets"
14:00:37 [INFO] Latest version: v1.5.2, current version: 1.5.2 (gh1422)
14:00:37 [INFO] Loaded 7460 lens profiles in 772.803ms
14:00:42 [DEBUG] (1) mdk: [MDK] default0x561a75eb6190 new FrameReader...
14:00:42 [DEBUG] (1) mdk: [MDK] Registered audio backends: ALSA OpenAL PulseAudio null
14:00:42 [DEBUG] (1) mdk: [MDK] ALSA device name: alsa_playback.gyroflow
14:00:42 [DEBUG] (1) mdk: [MDK] ALSA device id: alsa_playback.gyroflow
14:00:42 [DEBUG] (1) mdk: [MDK] ALSA subdevice name: alsa_playback.gyroflow
14:00:42 [DEBUG] (1) mdk: [MDK] default0x561a7626d7e0 new FrameReader...
14:00:42 [DEBUG] (1) mdk: [MDK] Registered audio backends: ALSA OpenAL PulseAudio null
14:00:42 [DEBUG] (1) mdk: [MDK] ALSA device name: alsa_playback.gyroflow
14:00:42 [DEBUG] (1) mdk: [MDK] ALSA device id: alsa_playback.gyroflow
14:00:42 [DEBUG] (1) mdk: [MDK] ALSA subdevice name: alsa_playback.gyroflow
14:00:42 [DEBUG] (1) mdk: [MDK] 0x561a762195b0mdk::Player::Player()
14:00:42 [DEBUG] (1) MDKPlayer: [in setUrl] Final path: "/home/vic/Videos/GX010100.MP4"
14:00:42 [DEBUG] (1) mdk: [MDK] default FrameReader0x561a75eb6190 state: 0=>0=>0, 0
14:00:42 [INFO] [MDK] default0x561a75eb6190 virtual void mdk::FrameReader::stop(), current state: 0
14:00:42 [DEBUG] (1) mdk: [MDK] default FrameReader0x561a75eb6190 state: 0=>0=>0, 0
14:00:42 [INFO] [MDK] default0x561a75eb6190 virtual void mdk::FrameReader::stop(), current state: 0
14:00:42 [DEBUG] (1) mdk: [MDK] AudioBackendALSA ERROR@285>>> snd_pcm_pause(pcm_, value) (ffffffb3): File descriptor in bad state
14:00:42 [DEBUG] (1) mdk: [MDK] 
14:00:42 [INFO] [MDK] 0x561a7a485860 MediaControl.prepare(0, ...) /home/vic/Videos/GX010100.MP4
14:00:42 [DEBUG] (1) mdk: [MDK] default 0x561a75eb6190 FrameReader.start(0, ...)
14:00:42 [DEBUG] (1) mdk: [MDK] default 0x561a75eb6190 FrameReader::update MediaStatus 0=>0X2
14:00:42 [DEBUG] (1) mdk: [MDK] default FrameReader0x561a75eb6190 state: 0=>0=>1, 0
14:00:42 [DEBUG] (1) mdk: [MDK] default FrameReader0x561a75eb6190 state: 0=>1=>1, 0
14:00:42 [DEBUG] (14) mdk: [MDK] start frame reader thread: 140457117988544
14:00:42 [DEBUG] (14) mdk: [MDK] try to load av module: /home/vic/Downloads/Gyroflow/lib/libffmpeg.so.6
14:00:42 [DEBUG] (14) mdk: [MDK] dlopen error: /home/vic/Downloads/Gyroflow/lib/libffmpeg.so.6: cannot open shared object file: No such file or directory
14:00:42 [DEBUG] (14) mdk: [MDK] try to load av module: /home/vic/Downloads/Gyroflow/lib/libavformat.so.60
14:00:42 [DEBUG] (14) mdk: [MDK] ffmpeg loaded: /home/vic/Downloads/Gyroflow/lib/libavformat.so.60
14:00:42 [DEBUG] (14) mdk: [MDK] Selected avformat runtime version: 60.10.100 (build: 60.10.100), license: GPL version 2 or later
[... cut ...]
14:01:07 [DEBUG] (1) mdk: [MDK] default 0x561a75eb6190-FrameReader::seek(1, 0X44)
14:01:07 [DEBUG] (1) mdk: [MDK] default FrameReader 0x561a75eb6190 seek target skipped: 1 frequent seek requested
14:01:07 [INFO] [MDK] bad delay. written: 80 - delay = -5
14:01:07 [DEBUG] (1) mdk: [MDK] 0x561a7a485860-MediaControl::seek(pos: 0, flag:0X44 +FromNow). now: 80
14:01:07 [DEBUG] (1) mdk: [MDK] default 0x561a75eb6190-FrameReader::seek(0, 0X44)
14:01:07 [DEBUG] (1) mdk: [MDK] default FrameReader 0x561a75eb6190 seek target skipped: 1 frequent seek requested
`

victorlysak, 2023-08-22:
also,
here is a link to a short orginal (but stabilized in camera) gopro x265 file,
if you want to experiment yourself
[https://drive.google.com/file/d/1ZPjcVMN-vSjyYjSLj3eWcHH_SX2pwFS2/view?usp=sharing](url)

thank you

victorlysak, 2023-08-22:
> How about now? https://gyroflow.xyz/devbuild/

Yes. This works perfectly on all files.
Thank you very much indeed.

victorlysak, 2023-08-22:
just for your records:
conversion to cfhd with 
-hwaccel_device cuda -profile 7 -color_range 1
goes at 11 fps with cpu @ 38%
 
so it's all good

thank you

victorlysak, 2023-08-22:
there is a difference in the param:
-color_range pc
between Gyroflow and ffmpeg

when I set -color_range pc, than ffmpeg gives me full range,
but the same in Gyroflow gives me always limited (tv) range.

Is there any other option that I can add to get the full range?

thank you

victorlysak, 2023-08-22:
thank you very much