gyroflow/gyroflow #693: Crash when attempting an export with GPU rendering

matejkarasek, 2023-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.5.2 gh1324

### What operating system are you using?

Win11 Home, 22H2, build 22621.1992

### What GPU are you using?

nVidia GeForce MX350

### What happened?

When I hit the Export button with 'Use GPU encoding' enabled, Gyroflow crashes.
Pre v 1.5 versions (e.g. 1.4.2) would show an error message about outdated/unsupported driver.

Reproducible with the GoPro 10 example video.

AdrianEddy, 2023-08-19:
This should be fixed in latest dev build, please test and let me know https://gyroflow.xyz/devbuild/

AdrianEddy, 2023-08-22:
Should be fixed in latest dev build https://gyroflow.xyz/devbuild/

closed 2023-08-22 by AdrianEddy

matejkarasek, 2023-09-01:
I confirm the error message now shows up correctly again, thanks!