gyroflow/gyroflow #694: Export outputs green frames

matejkarasek, 2023-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.5.2 gh1324

### What operating system are you using?

Win11 Home, 22H2, build 22621.1992

### What GPU are you using?

nVidia GeForce MX350

### What happened?

No matter what settings or file format I use, when I export I only get green frames in the output video/images (png).
The video is correctly stabilized and displayed in the preview window, just the export is not working.

Using default settings, GPU encoding unticked (so CPU encoding used0.

Same result observed in older Gyroflow versions and with various input files (e.g. GoPro Hero 10 example video), so seems to be related to my software and hardware configuration.

Examples of the output are below

https://github.com/gyroflow/gyroflow/assets/10943061/4650e5bb-de56-4990-a29d-5afaefef3c44

[image]
[image]
[image]

AdrianEddy, 2023-07-31:
Your GPU is too old to properly decode this codec. Uncheck "Use GPU decoding" in Advanced. Note that this is different than GPU encoding which is in Export settings.

matejkarasek, 2023-08-01:
I see, I assumed unchecking "Use GPU encoding" in the export settings will force the use of CPU everywhere...

In any case, this got fixed with the latest drivers. 
Prior to updating the drivers, selecting either the [wgpu] (Dx12) or [OpenCL] device options also fixed the export (so only [wgpu] (Vulcan) gave the green frames). Might be useful for others facing the same issue...

Thanks!

closed 2023-08-01 by matejkarasek