gyroflow/gyroflow #695: Opening Insta 360 GO2 video's fails since 1.5.2

matejkarasek, 2023-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.5.2 gh1324

### What operating system are you using?

Win11 Home, 22H2, build 22621.1992

### What GPU are you using?

nVidia GeForce MX350

### What happened?

Gyroflow crashes when attempting to open Pro videos from Insta360 GO2.
The same file opens correctly in 1.5.1 and earlier, the issue must have gotten introduced in 1.5.2

Reproducible with any Insta360 GO2 video

### Relevant log output

```shell
12:49:32 [INFO] [MDK] 0.21.1 (git eb669c4) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Windows0x0600/10.0.22621 x86_64; MSVC1935; MSVCRT14.35.32215.0/msvcp140.dll; 12:26:23 Jul 30 2023
12:49:32 [DEBUG] (1) gyroflow: [in main] Qt version: 6.5.1
12:49:32 [DEBUG] (1) gyroflow: Gyroflow 1.5.2 (gh1324)
12:49:32 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
12:49:32 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
12:49:32 [DEBUG] (1) mdk: [MDK] key time: -1
12:49:32 [DEBUG] (1) mdk: [MDK] check version
12:49:32 [DEBUG] (1) mdk: [MDK] user module: C:\Users\Matej\Downloads\gyroflow\Gyroflow-windows64\Gyroflow.exe
12:49:32 [DEBUG] (1) mdk: [MDK] user module: C:\Users\Matej\Downloads\gyroflow\Gyroflow-windows64\Gyroflow.exe
12:49:32 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
12:49:32 [WARN] qrc:/src/ui/main_window.qml:28:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
12:49:33 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 11.6.106") ["NVIDIA GeForce MX350"]
12:49:33 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") Ok("OpenCL 3.0") ["Intel(R) Iris(R) Plus Graphics"]
12:49:33 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation NVIDIA GeForce MX350
12:49:33 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
12:49:33 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
12:49:33 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
12:49:33 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce mx350
12:49:33 [WARN] QWindowsWindow::setGeometry: Unable to set geometry 1925x1009+0+23 (frame: 1941x1048-8-8) on QQuickWindowQmlImpl_QML_218/"" on "\\.\DISPLAY1". Resulting geometry: 1924x1009+0+23 (frame: 1940x1048-8-8) margins: 8, 31, 8, 8)
12:49:33 [WARN] qrc:/src/ui/menu/Synchronization.qml:17:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
12:49:34 [WARN] qrc:/src/ui/menu/Stabilization.qml:25:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
12:49:34 [WARN] qrc:/src/ui/menu/Export.qml:76:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
12:49:34 [WARN] qrc:/src/ui/menu/Advanced.qml:15:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
12:49:34 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA NVIDIA GeForce MX350: 3.0", "[OpenCL] Intel(R) OpenCL HD Graphics Intel(R) Iris(R) Plus Graphics: 3.0", "[wgpu] NVIDIA GeForce MX350 (Vulkan)", "[wgpu] Intel(R) Iris(R) Plus Graphics (Vulkan)", "[wgpu] NVIDIA GeForce MX350 (Dx12)", "[wgpu] Intel(R) Iris(R) Plus Graphics (Dx12)"]
12:49:34 [WARN] qrc:/src/ui/components/Timeline.qml:161:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
12:49:35 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\Matej\\Downloads\\gyroflow\\Gyroflow-windows64\\camera_presets"
12:49:35 [WARN] qrc:/src/ui/App.qml:64:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
12:49:35 [INFO] Loaded 7460 lens profiles in 570.450ms
12:50:08 [DEBUG] (1) mdk: [MDK] default0000026E4F94DDB0 new FrameReader...
12:50:08 [DEBUG] (1) mdk: [MDK] Registered audio backends: XAudio2 DSound null
12:50:08 [DEBUG] (1) mdk: [MDK] XAudio2_9.dll
12:50:08 [DEBUG] (1) mdk: [MDK] Build with XAudio2 from Win8+ SDK
12:50:08 [DEBUG] (1) mdk: [MDK] Try symbol 'XAudio2Create' from Windows SDK dll
12:50:08 [DEBUG] (1) mdk: [MDK] default0000026E4F95B3D0 new FrameReader...
12:50:08 [DEBUG] (1) mdk: [MDK] Registered audio backends: XAudio2 DSound null
12:50:08 [DEBUG] (1) mdk: [MDK] 0000026E4F955190__cdecl mdk::abi::Player::Player(void)
12:50:08 [DEBUG] (1) MDKPlayer: [in setUrl] Final path: "C:/Users/Matej/Downloads/PRO_VID_test.mp4"
12:50:08 [DEBUG] (1) mdk: [MDK] default FrameReader0000026E4F94DDB0 state: 0=>0=>0, 0
12:50:08 [INFO] [MDK] default0000026E4F94DDB0 void __cdecl mdk::abi::FrameReader::stop(void), current state: 0
[... 273 lines cut ...]
12:50:08 [DEBUG] (21) mdk: [MDK] d3d11va_vld
12:50:08 [DEBUG] (21) mdk: [MDK] d3d11 (selected)
12:50:08 [DEBUG] (21) mdk: [MDK] yuvj420p
12:50:08 [DEBUG] (21) mdk: [MDK] preparing AVHWFramesContext...
12:50:08 [DEBUG] (21) mdk: [MDK] hwdec hw output pixel format: d3d11
12:50:08 [DEBUG] (21) mdk: [MDK] [FFmpeg:h264] Failed setup for format d3d11: hwaccel initialisation returned error.
12:50:08 [DEBUG] (21) mdk: [MDK] h264 codec pixel format list: (expected d3d11)
12:50:08 [DEBUG] (21) mdk: [MDK] cuda
12:50:08 [DEBUG] (21) mdk: [MDK] dxva2_vld
12:50:08 [DEBUG] (21) mdk: [MDK] d3d11va_vld
12:50:08 [DEBUG] (21) mdk: [MDK] yuvj420p
12:50:08 [DEBUG] (21) mdk: [MDK] yuvj420p
12:50:08 [DEBUG] (21) mdk: [MDK] hwdec not available, sw fallback is disabled
12:50:08 [DEBUG] (21) mdk: [MDK] [FFmpeg:h264] decode_slice_header error
12:50:08 [DEBUG] (21) mdk: [MDK] [FFmpeg:h264] no frame!
12:50:08 [DEBUG] (11) mdk: [MDK] Error ret @113 D:\a\mdk-sdk\mdk-sdk\mdk\ffmpeg\plugin\AVCoderBase.cpp: (0xbebbb1b7) Invalid data found when processing input
12:50:08 [ERROR] [MDK] video stream#0 decode error
12:50:08 [DEBUG] (22) mdk: [MDK] h264 codec pixel format list: (expected d3d11)
12:50:08 [DEBUG] (22) mdk: [MDK] cuda
12:50:08 [DEBUG] (22) mdk: [MDK] dxva2_vld
12:50:08 [DEBUG] (22) mdk: [MDK] d3d11va_vld
12:50:08 [DEBUG] (22) mdk: [MDK] d3d11 (selected)
12:50:08 [DEBUG] (22) mdk: [MDK] yuvj420p
12:50:08 [DEBUG] (22) mdk: [MDK] preparing AVHWFramesContext...
```

AdrianEddy, 2023-07-31:
Both your issues are related to your old GPU. Do you have latest drivers from nvidia.com installed? 
Please check if it's working when you add `set MDK_DECODERS=ffmpeg` to Gyroflow_with_console.bat as a second line and then run this bat file

matejkarasek, 2023-08-01:
Updating the drivers to the latest available 536.67 didn't solve the issue.

However, with the FFmpeg in the bat file, the video opens correctly!

AdrianEddy, 2023-08-17:
Can you please try if it works in latest dev build? https://gyroflow.xyz/devbuild/

If not, can you replace these files in latest dev build and try again?
[mdk.zip](https://github.com/gyroflow/gyroflow/files/12364497/mdk.zip)

AdrianEddy, 2023-08-21:
Please check latest dev build now and let me know if it works https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-windows.zip

cooljam, 2023-08-22:
uncheck “Use GPU encoding”  could solve the crash  problem.

AdrianEddy, 2023-08-22:
Should be fixed in latest dev build https://gyroflow.xyz/devbuild/

closed 2023-08-22 by AdrianEddy

matejkarasek, 2023-09-01:
Sorry for the delay, I confirm this is fixed in the latest dev build