gyroflow/gyroflow #696: Add option to "Auto Sync" automatically when importing Media File

latenitefilms, 2023-08-01:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

It would be great if "Auto Sync" was triggered automatically when importing in a Media File. This means I could just drag and drop a R3D file on the Gyroflow Dock Icon (on Mac), and it'll automatically sync, then I can just press `COMMAND+S`

AdrianEddy, 2023-08-01:
Autosync can't be done without lens profile. If the lens profile can be loaded automatically, the autosync will also be done, as long as it's specified in the lens profile. Try the Hero6 example from test files, it does autosync automatically

I don't think we should always do autosync with any lens, because this will lead user to believe that it's all automatic and good, while bad lens profile can severely affect sync accuracy, and users need to learn that it needs care to get right.

AdrianEddy, 2023-08-01:
There's also an option to do autosync when creating preset so users have an option to enforce that

closed 2023-08-01 by latenitefilms