gyroflow/gyroflow #698: Terrible fonts on Linux

boris-gu, 2023-08-05:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

This is more of a "problem" rather than a "feature request". The fact is that on Linux Gyroflow has disgusting fonts.

The font of the interface (the one that is bold) is stretched too much, and the font for the data, on the contrary, is too narrow and difficult to read.

Please fix it or tell me how to fix it yourself on my computer
[image]

AdrianEddy, 2023-08-05:
Hmm, the font of the UI is "Arial" that should be available everywhere by default, but that font on your screenshot doesn't look like it? Maybe you don't have it so it uses some fallback font? please check if you can install Arial on your system

boris-gu, 2023-08-05:
There is an Arial font in my system

[image]

AdrianEddy, 2023-08-05:
Try to set some of these environment variables before starting Gyroflow:
`export QML_USE_GLYPHCACHE_WORKAROUND=1`
`export QSG_PREFER_FULLSIZE_GLYPHCACHE_TEXTURES=1`
`export QML_DISABLE_DISTANCEFIELD=1`

I'm not sure why it looks like that on your system, it looks ok on the Linuxes i tested and nobody else reported such issue

boris-gu, 2023-08-05:
It didn't change anything

[image]

AdrianEddy, 2023-08-05:
What if you run the .tar.gz package instead of AppImage?

boris-gu, 2023-08-05:
Without changes

AdrianEddy, 2023-08-05:
How does English language look like?

boris-gu, 2023-08-05:
I tried to change the language, first by changing the language of the entire system, then by changing the environment variables. But for some reason Gyroflow did not change the language to English.

If it helps a lot, then I can install a virtual machine with English localization, but it takes time

[image]

AdrianEddy, 2023-08-05:
The language is in Advanced panel on bottom right

AdrianEddy, 2023-08-05:
the screenshot you posted now looks better, what did you change? It's not bold and stretched anymore

boris-gu, 2023-08-05:
Yes, you're right, I just noticed it now  
If I run `LC_ALL=C ./Gyroflow-linux64.AppImage`, then the fonts come back to normal

boris-gu, 2023-08-05:
So, this could be called a hack rather than a fix, but I'm quite happy with it. I'll just create a shortcut to run Gyroflov properly.

Thank you very much for the reply

closed 2023-08-05 by boris-gu