gyroflow/gyroflow #700: Add support for HDR

latenitefilms, 2023-08-07:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Currently if you import a HDR clip, you can't retain the HDR metadata on export.

**Reference:**
https://codecalamity.com/encoding-uhd-4k-hdr10-videos-with-ffmpeg/

Shadetail, 2024-07-13:
One way to work around this limitation currently is to render the video using the Davinci Resolve plugin. All the stabilization work can be done with Gyroflow, then the .gyroflow file can be saved and the plugin can implement the smoothed out version, and Resolve can render it with correct HDR metadata.

hsb79, 2024-11-16:
Is there a way to preserve HLG also for Final Cut Pro? For reference, I use DJI Osmo Action 5 Pro in 10 bit HLG color mode, and the new Final Cut Pro 11 for editing.

AdrianEddy, 2024-11-16:
this issue is only in the standalone app, use the plugin directly in FCPX to avoid it https://gyroflowtoolbox.io/

NeariX67, 2025-06-25:
Can this please be looked into? I am getting worse results using Davinci Resolve Studio 20 with the OpenFX plugin (2.0.2) than the native gyroflow app. I'm using a low pass filter on the gyro signal and i don't think that's working with the plugin (i was using the gyroflow project file including gyro data)

eathtespagheti, 2025-07-07:
Hi, I've noticed that the Android version seems to be able to render the video keeping the HLG metadata