gyroflow/gyroflow #701: About Default Video Output Path

hzjianglf, 2023-08-08:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

After opening Gyroflow, when the first video specifies a custom output path, can the second video file default to the output path of the previous video? This will be much more convenient.

AdrianEddy, 2023-08-18:
Done in latest dev build, you can active it under `Export settings -> Advanced -> Preserve export parameters` 
https://gyroflow.xyz/devbuild/

closed 2023-08-18 by AdrianEddy

hzjianglf, 2023-08-22:
Thank you, the development version has been downloaded, but the output video has no content ，all green screen。

AdrianEddy, 2023-08-22:
Please upload log file

AdrianEddy, 2023-08-22:
That should be fixed now, try downloading again from the same link

hzjianglf, 2023-08-23:
Often, we default to the same bitrate as the source material when gyroflow outputting。

AdrianEddy, 2023-09-20:
In latest dev build, these two parameters are separated