gyroflow/gyroflow #702: gyroflow crashes when attempting to export - MAC OS

Ionzhou, 2023-08-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.52

### What operating system are you using?

macOS monterey 12.6.7

### What GPU are you using?

Intel Iris Pro 1536 MB

### What happened?

every time i attempt to export my video, gyroflow crashes. 
even i try to setting different device for processing under Advanced.

### Relevant log output

```shell
02:45:36 [INFO] [MDK] 0.21.1 (git eee6bb2) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: macOS1090/12.6.7 x86_64; AppleClang14.0.0 (clang-1400.0.29.202)GCC4.2.1; libc++13000; 17:00:50 Jul 12 2023
02:45:36 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
02:45:36 [DEBUG] (1) gyroflow: Gyroflow 1.5.2
02:45:36 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
02:45:36 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
02:45:36 [DEBUG] (1) mdk: [MDK] key time: -1
02:45:36 [DEBUG] (1) mdk: [MDK] check version
02:45:36 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
02:45:36 [WARN] Detected system locale encoding (US-ASCII, locale "C") is not UTF-8.
Qt shall use a UTF-8 locale ("UTF-8") instead. If this causes problems,
reconfigure your locale. See the locale(1) manual for more information.
02:45:37 [WARN] Populating font family aliases took 408 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. 
02:45:37 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Apple") Ok("OpenCL 1.2 (Aug  8 2022 21:29:33)") ["Iris Pro"]
02:45:37 [INFO] OpenCL Platform: Apple, ext: Extensions { inner: "cl_APPLE_SetMemObjectDestructor cl_APPLE_ContextLoggingFunctions cl_APPLE_clut cl_APPLE_query_kernel_names cl_APPLE_gl_sharing cl_khr_gl_event" } Device: Intel Iris Pro
02:45:37 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_VIDEOTOOLBOX
02:45:37 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_VIDEOTOOLBOX, device: None
02:45:37 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_VIDEOTOOLBOX
02:45:37 [DEBUG] (1) gyroflow::rendering: GPU type: Intel, from name: intel iris pro
02:45:40 [INFO] GPU list: ["[OpenCL] Apple Iris Pro: 1.2", "[wgpu] Intel Iris Pro Graphics (Metal)"]
02:45:40 [WARN] Unknown GPU apple iris pro: 1.2
02:45:40 [DEBUG] (1) gyroflow::rendering: GPU type: Intel, from name: apple iris pro: 1.2
02:45:40 [WARN] Failed to load gyro data from "/Users/anna/Documents/周转箱/20210101004903_202111.MP4": Custom { kind: Other, error: "Unsupported file format" }
02:45:41 [INFO] Lens profiles directory: "/Applications/Gyroflow.app/Contents/Resources/camera_presets"
02:45:41 [INFO] Latest version: v1.5.2, current version: 1.5.2
02:45:42 [INFO] Loaded 7460 lens profiles in 1109.143ms
02:45:51 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["videotoolbox"]
02:45:51 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: None
02:45:51 [DEBUG] (2) gyroflow::rendering: Possible encoders with Intel: [("prores_videotoolbox", true), ("prores_ks", false)]
02:45:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: HW formats: [VIDEOTOOLBOX]
02:45:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: SW formats: [NV12, AYUV64LE, YUV420P, UYVY422, P010LE, NV16, P210LE, P216LE, NV24, P410LE, P416LE, BGRA]
02:45:51 [DEBUG] (2) gyroflow::rendering: video_codec: Some("prores_videotoolbox"), processing_order: PreConversion
02:45:51 [ERROR] [h264_videotoolbox @ 0x7f989abc8940] Error: -constant_bit_rate true is not supported by the encoder.
02:45:51 [DEBUG] (2) gyroflow::rendering: start_us: 128620000, render_duration: 12699.999999999996, render_frame_count: 635
02:45:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Codec formats: [VIDEOTOOLBOX, YUV420P, NV12, AYUV64LE, UYVY422, P010LE, NV16, P210LE, P216LE, NV24, P410LE, P416LE, BGRA]
02:45:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Output metadata: {"major_brand": "mp42", "minor_version": "0", "compatible_brands": "isomavc1mp42", "creation_time": "2021-01-01T00:52:31.000000Z"}
02:45:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Converting from YUV420P to NV12
02:45:51 [DEBUG] (2) gyroflow_core::stabilization: Setting device 0
02:45:51 [INFO] OpenCL Platform: Apple, Device: Intel Iris Pro
02:45:51 [INFO] Initialized OpenCL for (2592, 1944, 2592) -> (1920, 1080, 1920)
02:46:00 [DEBUG] (2) gyroflow_core::stabilization: Setting device 0
02:46:00 [INFO] OpenCL Platform: Apple, Device: Intel Iris Pro
02:46:00 [INFO] Initialized OpenCL for (1296, 972, 2592) -> (960, 540, 1920)
02:46:08 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_VIDEOTOOLBOX), encoder_pixel_format: Some(NV12)
02:46:08 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: NV12, color range: MPEG
02:46:08 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type Some(AV_HWDEVICE_TYPE_VIDEOTOOLBOX)
02:46:08 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: Setting hw_device_ctx 0x7f98a9612f40
```

Ionzhou, 2023-08-09:
I'm testing exporting video, so I didn't do stabilization

AdrianEddy, 2023-08-12:
Change "Device for video processing" to "wgpu" 
https://docs.gyroflow.xyz/app/getting-started/faq#im-getting-a-crash-when-trying-to-export-and-im-using-macos-with-intel-processor
If that doesn't help, disable GPU encoding

closed 2023-08-12 by AdrianEddy