gyroflow/gyroflow #707: No video preview on upgrade from 1.5.0 to 1.5.2

bmsleight, 2023-08-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.2

### What operating system are you using?

[X]ubuntu 22.04.1 LTS _Jammy Jellyfish_

### What GPU are you using?

 GP107GLM [Quadro P620]

### What happened?

On upgrade from 1.5.0 to 1.5.2 error display video without any processing. 

[image]
[image]

### Relevant log output

```shell
bms@dustin:~/Old_Gyroflow$ cd ../Gyroflow/
bms@dustin:~/Gyroflow$ set MDK_DECODERS=FFmpeg ; ./gyroflow 
0.21.1 (git eee6bb2) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang16.0.6 (++20230704084245+7cbf1a259152-1~exp1~20230704204333.104)GCC4.2.1; libc++160006; 16:58:17 Jul 12 2023
20:11:58 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
20:11:58 [DEBUG] (1) gyroflow: Gyroflow 1.5.2 (gh1313)
mdk plugin 'mdk-braw': 0x55cbfbb9fbd0 from dir: /home/bms/Gyroflow/lib/
mdk plugin 'mdk-r3d': 0x55cbfbba0960 from dir: /home/bms/Gyroflow/lib/
20:11:59 [DEBUG] (1) gyroflow_core::gpu::wgpu: WGPU adapter: AdapterInfo { name: "Intel(R) UHD Graphics 630 (CFL GT2)", vendor: 32902, device: 16027, device_type: IntegratedGpu, driver: "Intel open-source Mesa driver", driver_info: "Mesa 22.2.5", backend: Vulkan }
20:11:59 [DEBUG] (1) gyroflow::rendering: GPU type: Intel, from name: intel(r) uhd graphics 630 (cfl gt2)
20:12:00 [INFO] GPU list: ["[wgpu] Intel(R) UHD Graphics 630 (CFL GT2) (Vulkan)", "[wgpu] llvmpipe (LLVM 15.0.6, 256 bits) (Vulkan)", "[wgpu] Mesa Intel(R) UHD Graphics 630 (CFL GT2) (Gl)", "[wgpu] Intel(R) UHD Graphics 630 (CFL GT2) (Vulkan)"]
20:12:00 [INFO] Lens profiles directory: "/home/bms/Gyroflow/camera_presets"
20:12:00 [INFO] Loaded 7460 lens profiles in 161.117ms
20:12:01 [DEBUG] (2) gyroflow::util: Minidump uploaded: OK
20:12:05 [DEBUG] (1) MDKPlayer: [in setUrl] Final path: "/home/bms/fpv/2023-08-14/Thumb0001.MP4"
20:12:05 [DEBUG] (1) qml: Setting param smoothness 0.5
20:12:05 [DEBUG] (1) qml: Setting param smoothness_pitch 0.5
20:12:05 [DEBUG] (1) qml: Setting param smoothness_yaw 0.5
20:12:05 [DEBUG] (1) qml: Setting param smoothness_roll 0.5
20:12:05 [DEBUG] (1) qml: Setting param per_axis 0
20:12:05 [DEBUG] (1) qml: Setting param trim_range_only 0
20:12:05 [DEBUG] (1) qml: Setting param max_smoothness 1
20:12:05 [DEBUG] (1) qml: Setting param alpha_0_1s 0.1
20:12:05 [DEBUG] (14) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
20:12:05 [DEBUG] (15) MDKPlayer: [in m_player->onEvent] "thread.video" ""
20:12:05 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
20:12:05 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
20:12:05 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
20:12:05 [DEBUG] (17) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
20:12:05 [DEBUG] (17) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7f12391e1aa0) QSize(1220, 686)
Failed to open VDPAU backend libvdpau_va_gl.so: cannot open shared object file: No such file or directory
Failed to open VDPAU backend libvdpau_va_gl.so: cannot open shared object file: No such file or directory
[... 102 lines cut ...]
    lens_model: "Runcam_Thumb_1080p_16by9",
    lens_info: "wide",
    focal_length: None,
    camera_setting: "",
    fps: 50000,
    video_width: 1920,
    video_height: 1080,
    additional: "",
    identifier: "runcam-thumb-runcam_thumb_1080p_16by9-wide-1920x1080@50000",
}
20:14:40 [DEBUG] (2) telemetry_parser::util: Reading duration: (None, None, None)
20:14:40 [DEBUG] (17) gyroflow_core::gpu::wgpu: WGPU initializing adapter #3
20:14:40 [ERROR] Failed to initialize wgpu RequestDevice(RequestDeviceError)
20:14:40 [WARN] IMU duration 209999.9999952381 is different than video duration (213000)
20:14:40 [DEBUG] (1) qml: [in onTelemetry_loaded] Telemetry additional data: {"camera_identifier":{"additional":"","brand":"runcam","camera_setting":"","fps":50000,"identifier":"runcam-thumb-runcam_thumb_1080p_16by9-wide-1920x1080@50000","lens_info":"wide","lens_model":"Runcam_Thumb_1080p_16by9","model":"thumb","video_height":1080,"video_width":1920},"contains_motion":true,"contains_quats":false,"contains_raw_gyro":true,"frame_readout_time":0,"fwversion":"Thumb V2.1.0","id":"runcam_thumb","imu_orientation":"XYZ","note":"development_tanky","sample_rate":1000.0047619274378,"vendor":"runcam","version":"1.1","videofilename":"Thumb0001.MP4"}
20:14:41 [DEBUG] (1) gyroflow::controller: Chart mutex locked, retrying
20:14:41 [DEBUG] (1) gyroflow::controller: Chart mutex locked, retrying
20:14:41 [DEBUG] (1) gyroflow::controller: Chart mutex locked, retrying
20:14:41 [DEBUG] (1) gyroflow::controller: Chart mutex locked, retrying
20:14:41 [DEBUG] (1) gyroflow::controller: Chart mutex locked, retrying
thread panicked while processing panic. aborting.
Crashdump written to /home/bms/Old_Gyroflow/3c7f27ee-c3fd-4a30-f0c076b0-b6bfa3e9.dmp
Aborted (core dumped)
bms@dustin:~/Old_Gyroflow$
```

bmsleight, 2023-08-14:
I did see logs line 2

AdrianEddy, 2023-08-14:
you called `set MDK_DECODERS=FFmpeg ; ./gyroflow ` which is different

bmsleight, 2023-08-14:
Apologies on mobile
Will try correct command tomorrow

bmsleight, 2023-08-15:
Thanks - with the command: MDK_DECODERS=FFmpeg ./gyroflow video is displayed. 
Apologies for the noise.

AdrianEddy, 2023-08-16:
Does it work in latest dev build? https://gyroflow.xyz/devbuild/

AdrianEddy, 2023-08-17:
Please also attach gyroflow.log file after this happens, it has more information

AdrianEddy, 2023-08-21:
Please check latest dev build now and let me know if it works https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-linux.zip

bmsleight, 2023-08-21:
Logs .

Current version using MDK_DECODERS=FFmpeg ./gyroflow opens video file successfully

[gyroflow_MDK_DECODERS=FFmpeg_.log](https://github.com/gyroflow/gyroflow/files/12402092/gyroflow_MDK_DECODERS.FFmpeg_.log)

Dev build hangs whilst try to open files. 
[gyroflow_dev_hangs_whilst_openning.log](https://github.com/gyroflow/gyroflow/files/12402093/gyroflow_dev_hangs_whilst_openning.log)

Dev version using MDK_DECODERS=FFmpeg ./gyroflow opens video file successfully

[gyroflow_MDK_DECODERS=FFmpeg_opens_video_fine.log](https://github.com/gyroflow/gyroflow/files/12402096/gyroflow_MDK_DECODERS.FFmpeg_opens_video_fine.log)

Hope this helps

AdrianEddy, 2023-08-22:
How about now? https://gyroflow.xyz/devbuild/

bmsleight, 2023-08-23:
Late devbuild still needs flags

[image]

[image]

[gyroflow_no_flags.log](https://github.com/gyroflow/gyroflow/files/12423303/gyroflow_no_flags.log)
[gyroflow_MDK_DECODERS=FFmpeg_opens_video_ok.log](https://github.com/gyroflow/gyroflow/files/12423304/gyroflow_MDK_DECODERS.FFmpeg_opens_video_ok.log)

AdrianEddy, 2023-08-23:
can you also include gyroflow.log file from v1.5.0 which worked without flags?

bmsleight, 2023-08-23:
[gyroflow-1.50.0_plain.log](https://github.com/gyroflow/gyroflow/files/12423698/gyroflow-1.50.0_plain.log)

AdrianEddy, 2023-08-25:
Maybe now? https://gyroflow.xyz/devbuild/

bmsleight, 2023-08-28:
Still requires  MDK_DECODERS=FFmpeg ./gyroflow
[gyroflow_MDK_DECODERS=FFmpeg_opens_video_fine.log](https://github.com/gyroflow/gyroflow/files/12457357/gyroflow_MDK_DECODERS.FFmpeg_opens_video_fine.log)
[gyroflow_simple_launch.log](https://github.com/gyroflow/gyroflow/files/12457358/gyroflow_simple_launch.log)

bmsleight, 2023-08-29:
Nope
still needs  MDK_DECODERS=FFmpeg ./gyroflow

(But the smoothing is so much better in latest version)

[gyroflow_MDK_DECODERS=FFmpeg_opens_video_fine.log](https://github.com/gyroflow/gyroflow/files/12469881/gyroflow_MDK_DECODERS.FFmpeg_opens_video_fine.log)
[gyroflow_simple_launch.log](https://github.com/gyroflow/gyroflow/files/12469882/gyroflow_simple_launch.log)

AdrianEddy, 2023-08-31:
what if you call it like this: `QT_XCB_GL_INTEGRATION=xcb_egl ./gyroflow` ?

bmsleight, 2023-08-31:
>what if you call it like this: QT_XCB_GL_INTEGRATION=xcb_egl ./gyroflow ?

When call QT_XCB_GL_INTEGRATION=xcb_egl ./gyroflow No errors - video can be seen fine. 

Need logs ?

wang-bin, 2023-09-10:
egl is required for iHD driver, see https://github.com/wang-bin/mdk-sdk/issues/114#issuecomment-1712705880

AdrianEddy, 2023-09-20:
Please check latest dev build: https://gyroflow.xyz/devbuild/

bmsleight, 2023-09-24:
Fixed :) 
Thanks - not longer requires  MDK_DECODERS=FFmpeg

closed 2023-09-24 by bmsleight