gyroflow/gyroflow #708: BRAW issues

timoschwarzer, 2023-08-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.2 (gh1344)

### What operating system are you using?

Windows 11 22H2 22621.2134

### What GPU are you using?

Intel Arc A770, Driver 31.0.101.4578

### What happened?

- [ ] Trying to load any `braw` file results in a black preview when GPU decoding is enabled. GPU decoding the same file in Davinci Resolve works without problems in the latest version. Maybe the SDK needs to be updated?
- [ ] Seeking through `braw` files results in the timeline cursor jumping around:

https://github.com/gyroflow/gyroflow/assets/8849554/c27bfbe9-134f-4732-a9a8-9457c400306b

This happens with all `braw` files for me.

As you can see in the video, I re-enabled GPU decoding after disabling it. Gyroflow kept playing the video afterwards, but I'm not sure if it really did GPU decoding after re-enabling it.

I've uploaded the `braw` here for testing: https://drive.google.com/file/d/10w_KEeHSxL0tEl8Kd66a7vHcuQWdd1As/view?usp=sharing

Logfile from loading the video, disabling GPU decoding and playing around with decoding settings until I got a preview, seeking through the timeline:
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/12339756/gyroflow.log)

AdrianEddy, 2023-08-14:
@wang-bin

wang-bin, 2023-08-21:
It's werid, i don't see any log from BRawReader.cpp

AdrianEddy, 2023-09-11:
@timoschwarzer  Please check in latest dev build https://gyroflow.xyz/devbuild/

timoschwarzer, 2023-09-13:
@AdrianEddy I tried the gh1430 build and unfortunately the issues don't seem to be fixed. Additionally, I cannot get the video preview to work anymore at all. The timeline cursor still jumps around when seeking through it but there is no video preview, just a black rectangle.
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/12599748/gyroflow.log)

timoschwarzer, 2023-10-02:
@AdrianEddy I figured out how to get the preview working again: I had to disable GPU decoding and *then* load the braw file. Disabling GPU decoding after loading the file doesn't seem to take effect. The issue with the timeline still occurrs in the latest build.

AdrianEddy, 2023-10-25:
Both issues should be fixed in latest dev build: https://gyroflow.xyz/devbuild/
Please test and confirm

closed 2023-10-25 by AdrianEddy

timoschwarzer, 2023-10-25:
@AdrianEddy will do so next week with new footage! Than you very much!

AdrianEddy, 2023-10-25:
any chance to test earlier? I wanted to release a new version tomorrow

timoschwarzer, 2023-10-26:
@AdrianEddy if you can wait until evening tomorrow (CEST UTC+2) I can test it then. I won't have any computer until then but I'll test as soon as possible when I'm back tomorrow.