gyroflow/gyroflow #712: only SUDO starts gyroflow - crashes on file open

nightflasher, 2023-08-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.5.2 (gh1313) - AppImage

### What operating system are you using?

ubuntu 22.04.3 LTS

### What GPU are you using?

AMD ATI Radeon R7 530 | INTEL UHD 620

### What happened?

Starting gyroflow without superuser rights and open files to edit
[e33dd536-f7d2-4795-c277bca7-3f9f8d7f.zip](https://github.com/gyroflow/gyroflow/files/12369471/e33dd536-f7d2-4795-c277bca7-3f9f8d7f.zip)

### Relevant log output

```shell
kevin@ideapad330:~/Gyroflow$ sudo ./Gyroflow-linux64.AppImage 
0.21.1 (git eee6bb2) - Multimedia Development Kit. Copyright (c) 2016-2023 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang16.0.6 (++20230704084245+7cbf1a259152-1~exp1~20230704204333.104)GCC4.2.1; libc++160006; 16:58:17 Jul 12 2023
12:42:14 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
12:42:14 [DEBUG] (1) gyroflow: Gyroflow 1.5.2 (gh1313)
mdk plugin 'mdk-braw': 0x560eca1cbcf0 from dir: /tmp/.mount_Gyroflfu9QaV//lib/
mdk plugin 'mdk-r3d': 0x560eca1ccbf0 from dir: /tmp/.mount_Gyroflfu9QaV//lib/
error: XDG_RUNTIME_DIR not set in the environment.
12:42:14 [DEBUG] (3) gyroflow::util: Minidump uploaded: OK
12:42:14 [WARN] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Gtk-Message: 14:42:14.645: Failed to load module "canberra-gtk-module"
Gtk-Message: 14:42:14.646: Failed to load module "canberra-gtk-module"
12:42:15 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Clover") Ok("OpenCL 1.1 Mesa 23.3~git2308170600.9a2a0c~oibaf~j (git-9a2a0c6 2023-08-17 jammy-oibaf-ppa)") ["AMD Radeon R5 M445 Series (iceland, LLVM 15.0.7, DRM 3.42, 5.15.0-41-generic)"]
12:42:15 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL Graphics") Ok("OpenCL 3.0") ["Intel(R) UHD Graphics 620"]
12:42:15 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("rusticl") Ok("OpenCL 3.0") []
12:42:15 [INFO] OpenCL Platform: Clover, ext: Extensions { inner: "cl_khr_icd" } Device: AMD AMD Radeon R5 M445 Series (iceland, LLVM 15.0.7, DRM 3.42, 5.15.0-41-generic)
12:42:15 [DEBUG] (1) gyroflow::rendering: GPU type: Amd, from name: amd amd radeon r5 m445 series (iceland, llvm 15.0.7, drm 3.42, 5.15.0-41-generic)
12:42:17 [INFO] GPU list: ["[OpenCL] Clover AMD Radeon R5 M445 Series (iceland, LLVM 15.0.7, DRM 3.42, 5.15.0-41-generic): 1.1", "[OpenCL] Intel(R) OpenCL Graphics Intel(R) UHD Graphics 620: 3.0", "[wgpu] Mesa Intel(R) UHD Graphics 620 (KBL GT2) (Gl)"]
12:42:18 [INFO] Lens profiles directory: "/tmp/.mount_Gyroflfu9QaV/camera_presets"
12:42:19 [INFO] Loaded 7460 lens profiles in 817.043ms

(gyroflow:22520): Gtk-WARNING **: 14:42:25.674: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /tmp/.mount_Gyroflfu9QaV//usr/share/icons/Adwaita/16x16/status/image-missing.png: Format der Bilddatei unbekannt (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /tmp/.mount_Gyroflfu9QaV//usr/share/icons/Adwaita/16x16/status/image-missing.png: Format der Bilddatei unbekannt (gdk-pixbuf-error-quark, 3)
Crashdump written to /home/kevin/Gyroflow/e33dd536-f7d2-4795-c277bca7-3f9f8d7f.dmp
Abgebrochen
```

AdrianEddy, 2023-08-17:
Does .tar.gz work?

nightflasher, 2023-08-18:
After trying to compile gyroflow (successfully!) I couldn't manage to run out anyways.
Than I've tried ffmpeg6 and all the other beta stuff around, it doesn't work either.
So every savoury1 ppa was purged and than I've tried something "stupid": I have the latest (oibaf) amd graphic drivers installed and purged this ppa - gyroflow is running again, but it's so slow.

AdrianEddy, 2023-08-25:
You can check out latest build - https://gyroflow.xyz/devbuild
No idea if this fixes your issue, but there's some change in the Linux/X11 handling stuff, so it might be related

AdrianEddy, 2023-09-20:
Check out latest dev build, AppImage should run now https://gyroflow.xyz/devbuild