gyroflow/gyroflow #714: Add information about original video file parts before joining into .gyroflow file

czukowski, 2023-08-23:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Please add a list of original video file parts to `.gyroflow` file, containing at least file names and the respective numbers of frames.

I use GyroflowToCSV to extract the gyro data to use it in Resolve Fusion and I have a video consisting of multiple parts. Fusion comps use the original footage (split parts) but Gyroflow has to join them before processing, otherwise the resulting data are not continuous (it would be really cool if Gyroflow could handle it seamlessly, ie the combined data from individually processed parts would be the same as the data from the joined video file, but I realize it's probably not very easy to do).

As for the gyro data, I know I can set an offset for the CSV row, so it's not such a big deal to use the combined CSV for that, although the offsets have to be calculated manually, so there is a room for error. Using the data I'm requesting, I'll be able to split the CSV and not have to care about offsets.

The problem I'm facing is that it turned out, the exported gyro data are not sufficient since I've added some keyframes for Roll angle correction and I have to extract that too and load into Fusion comps. They are not at each frame, so it wouldn't work to export it as CSV and load in the same fashion as gyro data, at least I haven't figured out how to do it. I can extract them as (frame number, value) pairs, that can be loaded into Fusion as keyframes, but these have to be split according to original parts. Using the data I'm requesting, I'll be able to achieve it.

AdrianEddy, 2023-08-23:
hmm, why you can't just use the joined file in Fusion?

czukowski, 2023-08-23:
The parts are already quite big as it is. Joining them in the first place would solve this issue, but add other ones. Also I'd prefer to work with camera originals from start to finish. Gyroflow is a data source for me at this point, so I'm looking for a way to integrate it into my current workflow, if possible, without having to redesign it.

AdrianEddy, 2023-08-23:
what are the issues with joined files? They should be exactly equivalent to the original files.
These files are not transcoded, and no metadata is lost. There is no video processing, it's pure binary joining
Take a look at the description here https://github.com/gyroflow/mp4-merge

czukowski, 2023-08-23:
It's not related to quality or metadata loss, I know it's joined without transcoding, it's more about handling and sending over the project files if needed. It's relatively easy for me to do with camera originals and not so much with intermediate media files, I guess this part can vary a lot depending on how people set up their workflows, so please let me spare you the details 😁

In fact I was contemplating about asking for a split-file processing feature like I mentioned briefly in the description but somehow I had a feeling it's not something easy enough to do for an odd use case like this, so I came up with this idea that should be much simpler and I'm willing to go through all the trouble of extracting all the data I need from `.gyroflow` files myself. I would even go as far as to attempt to whip up a pull request for this but I've never programmed anything in Rust yet so I don't think I could do it within a reasonable time without some kind of guidance...

AdrianEddy, 2023-08-24:
The problem with your request is that it seems simple, but it's actually not easy. There's no metadata saved when joining files, so where would I get the info about source files from? Gyroflow simply opens the joined file as a separate file. Do you want me to store that information in some intermediate text file? Keep it in memory? If memory, what if you open that joined file later?

It requires to write a lot of new code, for a feature that's very niche and I feel like you're trying to solve a different issue. That's why I wanted to know why not just use the joined files directly, as it seems to solve everything except your workflow

czukowski, 2023-08-24:
I see, well, I must have assumed the joining algorithm had this info and could just return it back to Gyroflow that would in turn store it in the memory representation of project file data and then save alongside with everything else. If the joiner doesn't have it, then it makes it more complicated indeed.

Anyway I've just spent half of my afternoon to come up with way to get what I need in a different way. A bit too hacky but it gets the job done. I still wish there was a proper way to retrieve that data from Gyrowlof but beggars can't be choosers 😃

closed 2023-08-25 by AdrianEddy