gyroflow/gyroflow #717: The reverse pass of smoothing algorithm

ysyscool, 2023-08-28:
What is it? Is there any reference for this algorithm?

AdrianEddy, 2023-08-28:
What do you mean? it's here https://github.com/gyroflow/gyroflow/blob/master/src/core/smoothing/plain.rs#L125-L136

ysyscool, 2023-08-28:
> What do you mean? it's here https://github.com/gyroflow/gyroflow/blob/master/src/core/smoothing/plain.rs#L125-L136
 Yes. I want to know why we need this reverse pass?

AdrianEddy, 2023-08-28:
Without it, the smoothing will be skewed towards the front, because we would only multiply current with next sample.
Doing reverse pass after that makes the smoothing symmetrical because we then multiply current with previous as well

This is a standard technique in signal filtering https://www.dsprelated.com/freebooks/filters/Forward_Backward_Filtering.html

closed 2023-08-28 by AdrianEddy

ClarkWang1214, 2023-10-20:
> Without it, the smoothing will be skewed towards the front, because we would only multiply current with next sample. Doing reverse pass after that makes the smoothing symmetrical because we then [...]

Hi, I want to know why Gyroflow just smooth the rotational motion without smoothing the translational motion? Thx