gyroflow/gyroflow #718: Way to export camera orientation and movement per frame

michaelochs, 2023-08-28:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I would like to be able to export what seems to be an intermediate state of the processing that Gyrobox does. What I am interested in is actually not image stabilization but instead I want a data stream that gives me orientation and location of the camera for each frame in the video.

It seems that this data should be available in Gyrobox as my understanding is it uses this to reverse the camera movement to some degree to stabilize the image.

It would be amazing if there was an option to get the data for this, ideally per frame. This allows other tools to then create virtual cameras that should, in theory, follow the exact movement the real camera did, which allows for a lot of options in video editing and compositing.

AdrianEddy, 2023-08-28:
See https://github.com/EmberLightVFX/Gyroflow-to-CSV and https://github.com/EmberLightVFX/Gyroflow-to-CSV/issues/6

closed 2023-08-28 by AdrianEddy

michaelochs, 2023-08-29:
Hey. That's interesting. For me the rotation doesn't match the camera movement 100%. It feels a bit as if the GoPro's hypersmooth settings aren't accounted for? Also in this one the x/z/y position offset is missing, which imho would be required to drive a virtual camera from it? I was under the assumption Gyroflow collects that already to stabilize positional movement.

AdrianEddy, 2023-08-29:
Everything is included in the .gyroflow project file, it has original rotations, smoothed rotations, all timestamps and offsets
If you had hypersmooth enabled then you have to use Quaternions, and that script uses the raw gyro data, unless you pass `--quaternions` option.

AdrianEddy, 2023-08-29:
`x/z/y position offset` you mean the translational motion from the acceleromater?
Gyroflow only uses rotations for stabilization. Translational motion is not tracked, and it's not really possible to do it with just the accelerometer data (it's not accurate enough)

michaelochs, 2023-08-29:
Okay, thanks for the info! I guess I need to figure out another way to determine x/y/z position movement.