gyroflow/gyroflow #726: Is it possible to support hyperlapse for DJI drones?

mordom0404, 2023-09-19:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

The following text uses translation software, so if there is any ambiguity in the expression please let me know :-)

The hyperlapse sequence shot using a DJI drone will obtain a series of dng files. Through exiftool, I found the following data in the exif information.
```
"XMP-drone-dji:GimbalRollDegree": "+0.00",
"XMP-drone-dji:GimbalYawDegree": "+0.00",
"XMP-drone-dji:GimbalPitchDegree": "+0.00",
"XMP-drone-dji:FlightRollDegree": -5.2,
"XMP-drone-dji:FlightYawDegree": -60.4,
"XMP-drone-dji:FlightPitchDegree": "+2.30",
"XMP-drone-dji:FlightXSpeed": 0.6,
"XMP-drone-dji:FlightYSpeed": -0.1,
"XMP-drone-dji:FlightZSpeed": 0,
```
It seems that DJI saves some angle information in dng files.
I usually use these dng photos to create hyperlapse videos (each photo is a frame in the final video), but when trying to use AE or DaVinci Resolve to stabilize the picture, sometimes the effect is not very satisfactory.
So is it possible for gyroflow to support the situation described above?

I know gyroflow supports custom gcsv files, and I have some programming skills (but not very good) and I tried the following works:
1. I created a video file using the hyperlapse dng photos shot by DJI Mavic 3 Drone. the video framerate is 50fps, resolution is 5272*2948 and without any post-processing.
2. I created a gcsv file, with tscale=0.02(because the framerate is 50fps), and I try to simulate the angular velocity of the gyroscope using the difference in angle between the current frame and the previous frame divided by 0.02 seconds.
the file looks like below:
```
GYROFLOW IMU LOG
version,1.3
id,custom_logger_name
orientation,Zyx
tscale,0.02
gscale,1
t,gx,gy,gz
0,0,0,0
1,1.7453292519943295,-1.4835298641951797,0.3490658503988647
2,-0.6108652381980151,-0.3490658503988659,0
3,0.08726646259971656,1.48352986419518,-0.08726646259971771
4,0.4363323129985824,-0.2617993877991493,0
5,-0.2617993877991497,-0.4363323129985824,0.08726646259971771
...and so on
```
But when I imported the video into gyroflow, I found that I did not have the corresponding lens configuration file.
I selected a Mavic3 profile existing (even though its aspect ratio didn't match my video), and as expected there seemed to be no stabilization effect.
I don't know if I should go any further because I don't know if this data is really useful right now, in principle.
If anyone could read the above and give some comments, I would be very grateful.

I have uploaded my video files as well as exif files and gcsv files to google drive as below:
https://drive.google.com/drive/folders/16sfTBE2mImqT9GcdQ-mM3YimZD5YAClL?usp=sharing

AdrianEddy, 2023-10-23:
Can you upload original .DNGs?

mordom0404, 2023-10-24:
> Can you upload original .DNGs?

Thank you for reply. I have uploaded the sequence to the google drive link above.

AdrianEddy, 2023-10-24:
Hmm, were these dngs modified? I can't see the DJI metadata in them, and exiftool says they were edited in Lightroom?
```
History Action                  : saved, saved
History Instance ID             : xmp.iid:7b62ec64-cd58-8942-a1d5-9095d8bc5fd0, xmp.iid:77e1773b-6c35-4d44-ac8d-0b706e3c5e3a
History When                    : 2023:05:07 02:22:48+08:00, 2023:05:07 02:25:32+08:00
History Software Agent          : Adobe Photoshop Lightroom Classic 12.2 (Windows), Adobe Photoshop Camera Raw 15.2 (Windows)
History Changed                 : /metadata, /metadata
```

mordom0404, 2023-10-25:
> Hmm, were these dngs modified? I can't see the DJI metadata in them, and exiftool says they were edited in Lightroom?
>
> ```
> History Action                  : saved, saved
> History Instance [...]

Yeah, I fogot to mention that I have adjust something in lightroom. But in my mind I only adjusted options like color. （If you need a completely pure dng sequence, I can shoot another immediately）

By the way, In the current dng sequence, I'm sure exiftool can read the information I mentioned.
Here is the exiftool command I use:
```
.\exiftool.exe -j -u -g1 HYPERLAPSE_0001.DNG
```
and then I get:
```
[{
  "SourceFile": "HYPERLAPSE_0001.DNG",
  "ExifTool": {
    "ExifToolVersion": 12.63
  },
  "System": {
    "FileName": "HYPERLAPSE_0001.DNG",
    "Directory": ".",
    "FileSize": "43 MB",
    "FileModifyDate": "2023:05:07 02:25:32+08:00",
    "FileAccessDate": "2023:10:25 10:54:11+08:00",
    "FileCreateDate": "2023:10:25 10:50:19+08:00",
    "FilePermissions": "-rw-rw-rw-"
  },
  "File": {
    "FileType": "DNG",
    "FileTypeExtension": "dng",
    "MIMEType": "image/x-adobe-dng",
    "ExifByteOrder": "Little-endian (Intel, II)"
  },
  "IFD0": {
    "SubfileType": "Reduced-resolution image",
    "ImageWidth": 160,
    "ImageHeight": 120,
    "BitsPerSample": "8 8 8",
    "Compression": "JPEG",
    "PhotometricInterpretation": "YCbCr",
    "Make": "Hasselblad",
    "Model": "L2D-20c",
    "PreviewImageStart": 79360,
    "Orientation": "Horizontal (normal)",
    "SamplesPerPixel": 3,
    "PreviewImageLength": 19125,
    "Software": "10.00.91.97",
    "ModifyDate": "2023:04:29 18:26:54",
    "XPComment": "0.9.142",
    "XPKeywords": "N",
    "DNGVersion": "1.4.0.0",
    "DNGBackwardVersion": "1.3.0.0",
    "UniqueCameraModel": "Hasselblad L2D-20c",
    "ColorMatrix1": "1.3364 -0.8291 0.123 -0.3009 1.1905 0.3937 0.0446 -0.0215 0.8437",
    "ColorMatrix2": "0.8575 -0.3219 -0.0868 -0.3351 1.1451 0.1593 0.0207 0.0468 0.4876",
    "AnalogBalance": "1 1 1",
    "AsShotNeutral": "0.472379195 1 0.5042471993",
    "BaselineExposure": 0,
    "BaselineNoise": 1,
    "BaselineSharpness": 1,
    "LinearResponseLimit": 1,
    "CameraSerialNumber": "493OJB58AA00P8",
    "ShadowScale": 1,
    "CalibrationIlluminant1": "Standard Light A",
    "CalibrationIlluminant2": "D65",
    "RawDataUniqueID": "282D1D7606A04E2312665F11AA423D02",
    "ProfileName": "Embedded",
    "ProfileHueSatMapDims": "18 6 1",
[... 179 lines cut ...]
    "Scap_Info": "(Binary data 8192 bytes, use -b option to extract)",
    "Sisr_Info": "(Binary data 4096 bytes, use -b option to extract)"
  },
  "Composite": {
    "Aperture": 2.8,
    "CFAPattern": "[Red,Green][Green,Blue]",
    "ImageSize": "5376x3956",
    "Megapixels": 21.3,
    "ScaleFactor35efl": 2.0,
    "ShutterSpeed": "1/160",
    "GPSAltitude": "91.4 m Above Sea Level",
    "GPSLatitude": "31 deg 15' 2.47\" N",
    "GPSLongitude": "120 deg 42' 51.57\" E",
    "CircleOfConfusion": "0.015 mm",
    "DOF": "inf (0.00 m - inf)",
    "FOV": "73.7 deg",
    "FocalLength35efl": "12.3 mm (35 mm equivalent: 24.0 mm)",
    "GPSPosition": "31 deg 15' 2.47\" N, 120 deg 42' 51.57\" E",
    "HyperfocalDistance": "3.51 m",
    "LightValue": 10.3,
    "LensID": "24.0mm f/2.8"
  }
}]
``` 
and you can find the data in the key named as "XMP-drone-dji"

AdrianEddy, 2023-10-25:
ok you're right, the data is there. I didn't see it before for some reason. 
I'll take a look if it can be used in Gyroflow

AdrianEddy, 2023-10-25:
Unfortunately I wasn't able to get it working. Looks like that data is not suitable for stabilization.

If you want to play with it, clone gyroflow repo and [telemetry-parser](https://github.com/AdrianEddy/telemetry-parser) repo, then extract files from [dji-dng.zip](https://github.com/gyroflow/gyroflow/files/13170000/dji-dng.zip) to overwrite these files. Then build gyroflow and load the mp4 video from sequences.

I have the dngs in D:/dng/HYPERLAPSE_0001.DNG and also axis.txt there
axis.txt contains mapping of axis to play with it, to determine which is pitch, which is roll and which is yaw and if it needs to be inverted. Axes seems to be correct but doesn't look like I can get any useful stabilization.

After editing axis.txt, you can use F8 key to dynamically reload the telemetry data from .dngs with axes remapped

I assumed 30fps, and the video from the sequence is here: https://drive.google.com/file/d/1H4-YX0pe9GlbeGsWeokVGtTpeE393pFM/view?usp=sharing

I recommend to check the git diff view in VS Code to see all the changes vs stock repo

I'm closing this because I don't think it may work, but if you play with it and figure out a way to get it working, feel free to let me know and I'll reopen

closed 2023-10-25 by AdrianEddy