gyroflow/gyroflow #728: Add tests on project core components

ologbonowiwi, 2023-09-24:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Since you've suggested on Discord that I'm open to this issue for this purpose!

What are the most critical parts of the project to have tests on?

I know that this issue is probably something that'll grow and grow (because, indeed, there are a lot of places to test), but I'm bringing this up to discuss stuff like tests cases that should be implemented, what are the kind of tests you think are most important to the project now (unit, integration, or e2e), and tooling that we could use to accomplish so.

I have plenty of experience with testing, so this is something that I could sum up for the project.

AdrianEddy, 2023-09-27:
Sorry for late response, I've been super busy with getting the mobile apps to usable state.
To be honest, I think what we need are benchmarks rather than tests. There weren't many regressions in the past and writing tests for core functionality will be a lot of work with little benefit. I think writing benchmarks instead and trying to optimize performance and/or memory usage will be much more beneficial

Places to benchmark:
1. [Gyro data parsing](https://github.com/AdrianEddy/telemetry-parser)
2. [Gyro integration](https://github.com/gyroflow/gyroflow/blob/master/src/core/imu_integration/mod.rs)
3. [Calculating frame transform](https://github.com/gyroflow/gyroflow/blob/master/src/core/stabilization/frame_transform.rs)
4. [Calculating FOVs for zooming](https://github.com/gyroflow/gyroflow/blob/master/src/core/zooming/mod.rs)
5. [Smoothing algorithm](https://github.com/gyroflow/gyroflow/blob/master/src/core/smoothing/default_algo.rs)
6. [Definitely GPU kernels](https://github.com/gyroflow/gyroflow/tree/master/src/core/gpu/stabilize_spirv) - we can focus on the SPIR-V implementation, because other ones will probably be replaced by this one soon

There's also the UI (in QML), which can also be profiled and optimized, but not sure how to write automated benchmarks for that

AdrianEddy, 2023-09-30:
Also, rendering time! There was just a rendering performance regressions after ffmpeg update, and it was significant (40% slower!).
I didn't notice it for a few weeks until just now when I compared previous version for some other test.
Having rendering time benchmark will definitely be needed

closed 2026-08-31 by AdrianEddy