gyroflow/gyroflow #732: GyroFlow crashes on startup

frankie111, 2023-09-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.5.2 (gh1493)

### What operating system are you using?

Windows 10.0.19045

### What GPU are you using?

NVIDIA GeForce RTX 3050

### What happened?

App loads forever then it crashes. 
Tried 3 of the latest versions - same problem.  
Tried running the app as Admin and deleting Regedit key.

### Relevant log output

```shell
18:36:26 [DEBUG] (1) gyroflow: [in main] Qt version: 6.5.2
18:36:26 [DEBUG] (1) gyroflow: Gyroflow 1.5.2 (gh1493)
18:36:26 [DEBUG] (1) gyroflow: Executable path: Some("D:\\Gyroflow-windows\\Gyroflow-windows64\\Gyroflow.exe")
18:36:26 [DEBUG] (1) gyroflow: SDK path: "D:\\Gyroflow-windows\\Gyroflow-windows64"
18:36:30 [WARN] qrc:/src/ui/main_window.qml:31:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
18:36:30 [DEBUG] (1) gyroflow_core::filesystem: path_to_url | path:  -> Ok("")
18:36:30 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.2.138") ["NVIDIA GeForce RTX 3050 Laptop GPU"]
18:36:30 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("AMD Accelerated Parallel Processing") Ok("OpenCL 2.1 AMD-APP (3180.7)") ["gfx902"]
18:36:30 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation NVIDIA GeForce RTX 3050 Laptop GPU
18:36:30 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
18:36:30 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
18:36:30 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
18:36:30 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation nvidia geforce rtx 3050 laptop gpu
18:36:30 [WARN] QWindowsWindow::setGeometry: Unable to set geometry 1925x1055+9+38 (frame: 1943x1102+0+0) on QQuickWindowQmlImpl_QML_222/"" on "\\.\DISPLAY1". Resulting geometry: 1924x1055+9+38 (frame: 1942x1102+0+0) margins: 9, 38, 9, 9 minimum size: 900x400 MINMAXINFO(maxSize=POINT(x=0, y=0), maxpos=POINT(x=0, y=0), maxtrack=POINT(x=0, y=0), mintrack=POINT(x=1143, y=547)))
18:36:31 [WARN] qrc:/src/ui/menu/Synchronization.qml:17:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
18:36:31 [WARN] qrc:/src/ui/menu/Stabilization.qml:25:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
18:36:31 [WARN] qrc:/src/ui/menu/Export.qml:80:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
18:36:31 [WARN] qrc:/src/ui/menu/Advanced.qml:15:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
```

AdrianEddy, 2023-09-25:
Do you have latest GPU drivers from nvidia.com? the ones installed by Windows are not enough

frankie111, 2023-09-25:
I do have the latest GPU drivers.
I have an integrated GPU from the AMD Ryzen 7 5800H.

AdrianEddy, 2023-09-25:
Then install drivers for the AMD one, it's important

frankie111, 2023-09-25:
The AMD drivers were indeed the problem. Thank you!
The latest OEM drivers are probably outdated, hope I didn't break something else now

closed 2023-09-25 by frankie111