gyroflow/gyroflow #733: timecode stripped

someTV, 2023-09-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.2

### What operating system are you using?

mac ventura 13.5.2

### What GPU are you using?

apple m1 max

### What happened?

stabilized file timecode stripped 
[gx010563.txt](https://github.com/gyroflow/gyroflow/files/12732483/gx010563.txt)
[gx010563_stabilized.txt](https://github.com/gyroflow/gyroflow/files/12732484/gx010563_stabilized.txt)

AdrianEddy, 2023-09-26:
Check "Preserve other tracks" in Export settings -> Advanced

someTV, 2023-09-26:
Tried that, will try again

AdrianEddy, 2023-09-26:
I don't see timecode in the `gx010563.txt` file though?
Do you mean these?
```
Encoded date                             : 2023-07-08 12:10:21 UTC
Tagged date                              : 2023-07-08 12:10:21 UTC
```

someTV, 2023-09-26:
Not the utc creation date, actual tc. Must have not ticked that box! Thank you very much! Should I assume it preserves all metadata?

someTV, 2023-09-26:
worked...

someTV, 2023-09-26:
Weird, preserved it but gave wrong timecode, original starts at 12:34:46:28 stabilized starts at 12:09:37:12
[gx010563_2stabilized.txt](https://github.com/gyroflow/gyroflow/files/12732534/gx010563_2stabilized.txt)
[gx010563.txt](https://github.com/gyroflow/gyroflow/files/12732535/gx010563.txt)

AdrianEddy, 2023-09-26:
can you upload the original video file? I'll check

someTV, 2023-09-26:
weird thing is that my original file displays the correct tc but using mediainfo it gives an incorrect number. will do some more investigating when i get a chance.

someTV, 2023-09-27:
Next question, trying to batch stabilize in render cue preserving other tracks and keep getting error that starts with: Make sure your output extension suppports ther selected codec. ".mov" should work in most cases. It works fine directly from the settings window. what the gyro terminal said:
[gyroterminal.txt](https://github.com/gyroflow/gyroflow/files/12734578/gyroterminal.txt)
Thanks,
Barnaby

AdrianEddy, 2023-09-27:
So is the timecode good after all?

someTV, 2023-09-27:
Yes, the timecode is perfect, just now my issue with the render queue which I opened with another issue as I was not sure if you would see what I had posted above in this issue. thank you for your timely response!

AdrianEddy, 2023-09-27:
Ok, the other one I also just fixed, will be available in ~30 min

closed 2023-09-27 by AdrianEddy

someTV, 2023-09-27:
thank you!  just download again?