gyroflow/gyroflow #736: render cue will not render when preserve other tracks selected

someTV, 2023-09-27:
Have tried rendering on both pc and mac after adding files directly to the render queue with preserve tracks selected and get the error seen in attached file
[image]
Help please for this newbie.
Thanks,
Barnaby

AdrianEddy, 2023-09-27:
Fixed in e2bed40ed0b96466d44524f07224509c7a0e5b03, will be available in https://gyroflow.xyz/devbuild/ once [this](https://github.com/gyroflow/gyroflow/actions/runs/6330600516) finishes building

closed 2023-09-27 by AdrianEddy

AdrianEddy, 2023-09-27:
Just download from here https://gyroflow.xyz/devbuild/

someTV, 2023-09-27:
great. any chance for windows? Running on two machines...

AdrianEddy, 2023-09-27:
it's all there

someTV, 2023-09-27:
just seeing mac, linux iOS and android, no windows build unless I am blind...

someTV, 2023-09-27:
Must add as well that this is an awesome tool. I accidentally had gopro cameras remove all processing beyond capturing and did not realize that included all the stabilization info.

someTV, 2023-09-27:
linking to ping...

someTV, 2023-09-27:
Ah, I can see it momentarily when I am scrolling through the window but it hides behind the iOS link before I can get to it.

someTV, 2023-09-27:
can get to it by viewing page source...

AdrianEddy, 2023-09-27:
ok works ok on Windows, but I'll fix the page when viewed from other system

someTV, 2023-09-27:
Getting error on windows machine please see attached
Barnaby
323.401.8642

AdrianEddy, 2023-09-27:
Replying to email with attachments doesn't work, you have to attach in github comment

someTV, 2023-09-27:
Same error message as before...
[image]

AdrianEddy, 2023-09-27:
I just tested and it works as expected for me. Did you re-add these files to the queue? Does it say `(gh1508)` in the window title?

someTV, 2023-09-27:
tried it again, same problem, this is where I downloaded gyroflow from: "https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-windows.zip">Gyroflow dev build

AdrianEddy, 2023-09-27:
but the screenshot is from mac?

AdrianEddy, 2023-09-27:
How did you add the file to the queue? Can you write exact steps you're taking?

someTV, 2023-09-27:
from pc, window title says Gyroflow 1.5.2 if that is what you are asking regards (gh1508)
dragged the file from folder onto queue, dropped gyroflow preset onto queue, checked settings by clicking edit, then the render button followed by that error message

AdrianEddy, 2023-09-27:
Window title should say `Gyroflow v1.5.2 (gh1508)`
[image]

AdrianEddy, 2023-09-27:
Can you attach that preset?

someTV, 2023-09-27:
nope.
[CLEORGEhorlock.txt](https://github.com/gyroflow/gyroflow/files/12743798/CLEORGEhorlock.txt)

someTV, 2023-09-27:
i must not have 1.5.2 (gh1508). Could you send me a direct link to download it?

AdrianEddy, 2023-09-27:
https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-windows.zip

that's the one

someTV, 2023-09-27:
Got from the dev page! Will let you know how it works in a moment

someTV, 2023-09-27:
hit render button, vaporized, installing again.

someTV, 2023-09-27:
ddo I need to do anything with the "Gyroflow.msix" file?

AdrianEddy, 2023-09-27:
no, just extract the Gyroflow-windows64.zip file

someTV, 2023-09-27:
Worked on one! Will try a batch. Thank you for your help.

someTV, 2023-09-28:
Weirdness, on mac batch will render 1 file but when I added four files i got this error message:
[image]
PC seems to functiong fine.

AdrianEddy, 2023-09-28:
maybe you don't have write permissions to that directory? are you using the same preset? preset contains output folder, which will be different on mac and windows, so you have to have 2 presets

someTV, 2023-09-28:
Will play around with the mac more, it just started when I added two to the batch. It still asks me for a directory even when I drop the preset on it.

AdrianEddy, 2023-09-28:
It should always ask for directory when dropping video files
It shouldn't ask for directory when dropping preset.
Preset applies only to current files in the queue, it doesn't apply to future files

someTV, 2023-09-28:
is there a way to make a preset that allows me to choose the output directory after the preset is applied to all files in the queue? Not seeing output directory in the preset menu.

AdrianEddy, 2023-09-28:
`Output path`
[image]

someTV, 2023-09-28:
I'm blind. Need to go to Hollylodz and have my eyes checked.

someTV, 2023-09-28:
On pc windows 10 the rendered ".mov" file will not play in quicktime (will play in others such as potplayer or windows media player). This is the error message: "Error -2002: a bad public movie atom was found in the movie 0" 
[image]

AdrianEddy, 2023-09-28:
I'm afraid it's outside of my control, ffmpeg renders the file

someTV, 2023-09-28:
Interesting. Why does the file need to be rendered as a .mov and not as its original container when preserving other tracks? In my case mp4s. Must reiterate how much I like the design and function of the ap.

AdrianEddy, 2023-09-28:
ffmpeg doesn't know how to include custom binary tracks in the mp4 container, it's simply not implemented there and only implemented for mov. This is also outside of my control

someTV, 2023-09-28:
could the binary tracks including audio just pass through and not be custom?

AdrianEddy, 2023-09-28:
they are pass through, but that doesn't change anything, ffmpeg mp4 muxer just can't handle them

someTV, 2023-09-28:
ok, thanks.