gyroflow/gyroflow #738: ERROR 22: Invalid argument

StefanMeGit, 2023-10-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.2 gh1537

### What operating system are you using?

Win10

### What GPU are you using?

NVIDIA Quadro RTX 4000

### What happened?

Cant render the video. Did try to render with H264 and 265.. same message. Not sure what to do.

[image]

### Relevant log output

```shell
09:07:52 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.1.98") ["Quadro RTX 4000"]
09:07:52 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") Ok("OpenCL 2.1") ["Intel(R) UHD Graphics"]
09:07:52 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_khr_gl_event cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_nv_kernel_attribute cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation Quadro RTX 4000
09:07:53 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
09:07:53 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: HWDevice::from_type AV_HWDEVICE_TYPE_CUDA, device: None
09:07:53 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
09:07:53 [DEBUG] (1) gyroflow::rendering: GPU type: Nvidia, from name: nvidia corporation quadro rtx 4000
09:07:54 [WARN] qrc:/src/ui/menu/Synchronization.qml:17:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
09:07:54 [WARN] qrc:/src/ui/menu/Stabilization.qml:25:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
09:07:54 [WARN] qrc:/src/ui/menu/Export.qml:80:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
09:07:54 [WARN] qrc:/src/ui/menu/Advanced.qml:15:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
09:07:55 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA Quadro RTX 4000: 3.0", "[OpenCL] Intel(R) OpenCL HD Graphics Intel(R) UHD Graphics: 2.1", "[wgpu] Quadro RTX 4000 (Vulkan)", "[wgpu] Intel(R) UHD Graphics (Vulkan)", "[wgpu] NVIDIA Quadro RTX 4000 (Dx12)", "[wgpu] Intel(R) UHD Graphics (Dx12)"]
09:07:55 [WARN] qrc:/src/ui/components/Timeline.qml:165:5: QML Settings: The Settings type from Qt.labs.settings is deprecated and will be removed in a future release. Please use the one from QtCore instead.
09:07:55 [DEBUG] (2) gyroflow_core::filesystem: path_to_url | path: K:/DCIM/100GOPRO/GX011477.MP4 -> Ok("file:///K:/DCIM/100GOPRO/GX011477.MP4")
09:07:55 [DEBUG] (2) gyroflow_core::filesystem: path_to_url | path: K:/DCIM/100GOPRO/GX011477.MP4 -> Ok("file:///K:/DCIM/100GOPRO/GX011477.MP4")
09:07:55 [DEBUG] (2) gyroflow_core::filesystem: can_open_file | url: file:///K:/DCIM/100GOPRO/GX011477.MP4
09:07:55 [DEBUG] (3) gyroflow_core::filesystem: path_to_url | path: K:/DCIM/100GOPRO/GX011477.MP4 -> Ok("file:///K:/DCIM/100GOPRO/GX011477.MP4")
09:07:55 [DEBUG] (2) gyroflow_core::filesystem: exists | url: file:///K:/DCIM/100GOPRO/GX011477.MP4 -> Ok(true)
09:07:55 [DEBUG] (2) gyroflow_core::filesystem: open_file | url: file:///K:/DCIM/100GOPRO/GX011477.MP4 | writing: false
09:07:55 [DEBUG] (2) gyroflow_core::filesystem: url_to_path | url: file:///K:/DCIM/100GOPRO/GX011477.MP4 -> Ok("K:\\DCIM\\100GOPRO\\GX011477.MP4")
09:07:55 [DEBUG] (3) gyroflow_core::filesystem: path_to_url | path: K:/DCIM/100GOPRO/GX011477.MP4 -> Ok("file:///K:/DCIM/100GOPRO/GX011477.MP4")
09:07:55 [DEBUG] (3) gyroflow_core::filesystem: can_open_file | url: file:///K:/DCIM/100GOPRO/GX011477.MP4
09:07:55 [DEBUG] (2) gyroflow_core::filesystem: FileWrapper::drop file:///K:/DCIM/100GOPRO/GX011477.MP4
09:07:55 [DEBUG] (3) gyroflow_core::filesystem: exists | url: file:///K:/DCIM/100GOPRO/GX011477.MP4 -> Ok(true)
09:07:55 [DEBUG] (2) gyroflow_core::filesystem: exists | url: file:///K:/DCIM/100GOPRO/GX011477.MP4 -> Ok(true)
09:07:55 [DEBUG] (2) gyroflow_core::filesystem: open_file | url: file:///K:/DCIM/100GOPRO/GX011477.MP4 | writing: false
09:07:55 [DEBUG] (2) gyroflow_core::filesystem: url_to_path | url: file:///K:/DCIM/100GOPRO/GX011477.MP4 -> Ok("K:\\DCIM\\100GOPRO\\GX011477.MP4")
09:07:55 [DEBUG] (3) gyroflow_core::filesystem: open_file | url: file:///K:/DCIM/100GOPRO/GX011477.MP4 | writing: false
09:07:55 [DEBUG] (2) gyroflow_core::filesystem: url_to_path | url: file:///K:/DCIM/100GOPRO/GX011477.MP4 -> Ok("K:\\DCIM\\100GOPRO\\GX011477.MP4")
09:07:55 [DEBUG] (3) gyroflow_core::filesystem: url_to_path | url: file:///K:/DCIM/100GOPRO/GX011477.MP4 -> Ok("K:\\DCIM\\100GOPRO\\GX011477.MP4")
09:07:55 [DEBUG] (3) gyroflow_core::filesystem: FileWrapper::drop file:///K:/DCIM/100GOPRO/GX011477.MP4
09:07:55 [DEBUG] (3) gyroflow_core::filesystem: exists | url: file:///K:/DCIM/100GOPRO/GX011477.MP4 -> Ok(true)
09:07:55 [DEBUG] (3) gyroflow_core::filesystem: open_file | url: file:///K:/DCIM/100GOPRO/GX011477.MP4 | writing: false
[... 465 lines cut ...]
09:08:14 [DEBUG] (2) gyroflow::rendering: video_codec: Some("hevc_nvenc"), processing_order: PreConversion
09:08:14 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 284334, render_frame_count: 17043
09:08:14 [DEBUG] (2) gyroflow_core::filesystem: exists | url: file:///K:/DCIM/100GOPRO/ -> Ok(true)
09:08:14 [DEBUG] (2) gyroflow_core::filesystem: get_file_url | folder_url: file:///K:/DCIM/100GOPRO/ | filename: GX011477_stabilized.mp4 | can_create: true -> Ok("file:///K:/DCIM/100GOPRO/GX011477_stabilized.mp4")
09:08:14 [DEBUG] (2) gyroflow_core::filesystem: FfmpegPathWrapper::new file:///K:/DCIM/100GOPRO/GX011477_stabilized.mp4, write: true
09:08:14 [DEBUG] (2) gyroflow_core::filesystem: url_to_path | url: file:///K:/DCIM/100GOPRO/GX011477_stabilized.mp4 -> Ok("K:\\DCIM\\100GOPRO\\GX011477_stabilized.mp4")
09:08:14 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Codec formats: [YUV420P, NV12, P010LE, YUV444P, P016LE, YUV444P16LE, BGRZ, BGRA, RGBZ, RGBA, X2RGB10LE, X2BGR10LE, GBRP, GBRP16LE, CUDA, D3D11]
09:08:14 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Output metadata: {"major_brand": "mp41", "minor_version": "538120216", "compatible_brands": "mp41", "creation_time": "2023-10-01T12:53:01.000000Z", "firmware": "H22.03.02.20.70"}
09:08:14 [DEBUG] (2) gyroflow_core::stabilization: Setting device 0
09:08:14 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation Quadro RTX 4000
09:08:15 [INFO] Initialized OpenCL for (4000, 3000, 4096) -> (3840, 2160, 3840)
09:08:15 [DEBUG] (2) gyroflow_core::stabilization: Setting device 0
09:08:15 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation Quadro RTX 4000
09:08:16 [INFO] Initialized OpenCL for (2000, 1500, 4096) -> (1920, 1080, 3840)
09:08:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: Some([NV12])
09:08:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_CUDA), encoder_pixel_format: Some(NV12)
09:08:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: NV12, color range: JPEG
09:08:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type Some(AV_HWDEVICE_TYPE_CUDA)
09:08:16 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: Setting hw_device_ctx 0x1af93e71c80
09:08:16 [DEBUG] (2) gyroflow::rendering: [hevc_nvenc @ 000001AFC77A1680] Loaded Nvenc version 12.1
09:08:28 [DEBUG] (2) gyroflow::rendering: [aac @ 000001AFC77A2280] Qavg: 10708.611
09:08:28 [WARN] [aac @ 000001AFC77A2280] 2 frames left in the queue on closing
09:08:28 [DEBUG] (2) gyroflow_core::filesystem: FfmpegPathWrapper::drop file:///K:/DCIM/100GOPRO/GX011477_stabilized.mp4
09:08:29 [DEBUG] (2) gyroflow_core::filesystem: FfmpegPathWrapper::drop file:///K:/DCIM/100GOPRO/GX011477.MP4
```

AdrianEddy, 2023-10-06:
Looks like you're trying to export to the SD card. Did you try to change the output folder to some local folder on disk? Did you try other codec like ProRes?

AdrianEddy, 2023-10-11:
Any update? Also check if you maybe have any advanced codec parameters in Export settings -> Advanced -> Custom encoder options

In general "Invalid argument" refers to some unexpected flags for the codec or pixel format
Did you also try to reset all settings to defaults? in Advanced on bottom right

AdrianEddy, 2023-10-18:
Closing due to inactivity and I'm unable to reproduce. If you have any new information, feel free to comment

closed 2023-10-18 by AdrianEddy