gyroflow/gyroflow #739: Segfaults ~25% of the time when opening video files

andpol, 2023-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.2 and also today's devbuild

### What operating system are you using?

up-to-date Arch Linux

### What GPU are you using?

NVidia RTX3080

### What happened?

On both the 1.5.2 release and today's nightly build from https://gyroflow.xyz/devbuild I often get segfaults when opening video files. I've tried running both the AppImage binary as well as the one inside the tar.gz archive from the nightly.

I have a few files already in the render queue when this happens, in case it makes a difference.

I've attached a crash dump.
[crashdump.tar.gz](https://github.com/gyroflow/gyroflow/files/12849449/crashdump.tar.gz)

### Relevant log output

```shell
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: path_to_url | path: /run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1 -> Ok("file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: get_folder | url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230.MP4 -> Ok("file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: get_filename | url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230.MP4 -> Ok("GX010230.MP4")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: path_to_url | path: /run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1 -> Ok("file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: get_folder | url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230.MP4 -> Ok("file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: get_file_url | folder_url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/ | filename: GX010230.MP4 | can_create: false -> Ok("file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230.MP4")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: exists | url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230.MP4 -> Ok(true)
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: exists_in_folder | folder_url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/ | filename: GX010230.MP4 -> true
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: get_file_url | folder_url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/ | filename: GX020230.MP4 | can_create: false -> Ok("file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX020230.MP4")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: exists | url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX020230.MP4 -> Ok(false)
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: exists_in_folder | folder_url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/ | filename: GX020230.MP4 -> false
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: get_filename | url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230.MP4 -> Ok("GX010230.MP4")
ALSA lib pcm_dmix.c:1000:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:1000:(snd_pcm_dmix_open) unable to open slave
18:38:14 [DEBUG] (1) MDKPlayer: [in setUrl] Final url: "/run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230.MP4"
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: filename_with_suffix | filename: GX010230.MP4 | suffix: _stabilized -> "GX010230_stabilized.MP4"
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: get_file_url | folder_url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/ | filename: GX010230_stabilized.MP4 | can_create: false -> Ok("file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.MP4")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: display_url | url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.MP4
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: url_to_path | url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.MP4 -> Ok("/run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.MP4")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: display_folder_filename | folder: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/ | filename: GX010230_stabilized.MP4 -> "/run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.MP4"
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: get_file_url | folder_url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/ | filename: GX010230_stabilized.MP4 | can_create: false -> Ok("file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.MP4")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: display_url | url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.MP4
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: url_to_path | url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.MP4 -> Ok("/run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.MP4")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: display_folder_filename | folder: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/ | filename: GX010230_stabilized.MP4 -> "/run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.MP4"
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: get_file_url | folder_url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/ | filename: GX010230_stabilized.mp4 | can_create: false -> Ok("file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.mp4")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: display_url | url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.mp4
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: url_to_path | url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.mp4 -> Ok("/run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.mp4")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: display_folder_filename | folder: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/ | filename: GX010230_stabilized.mp4 -> "/run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230_stabilized.mp4"
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: filename_with_extension | filename: GX010230.MP4 | ext: gyroflow -> "GX010230.gyroflow"
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: get_file_url | folder_url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/ | filename: GX010230.gyroflow | can_create: false -> Ok("file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230.gyroflow")
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: exists | url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/GX010230.gyroflow -> Ok(false)
18:38:14 [DEBUG] (1) gyroflow_core::filesystem: exists_in_folder | folder_url: file:///run/media/apolyakov/FF7F-A338/2023/elk_mountain_oct_1/ | filename: GX010230.gyroflow -> false
18:38:14 [DEBUG] (72) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
18:38:14 [DEBUG] (73) MDKPlayer: [in m_player->onEvent] "thread.video" ""
18:38:14 [DEBUG] (74) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
18:38:14 [DEBUG] (74) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
18:38:14 [DEBUG] (74) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
OpenAL 1.1 ALSOFT 1.23.1 vendor: OpenAL Community; renderer: OpenAL Soft
18:38:14 [DEBUG] (73) MDKPlayer: [in m_player->onEvent] "decoder.video" "CUDA"
18:38:15 [DEBUG] (29) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
18:38:15 [DEBUG] (29) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7f230cd6cfe0) QSize(5312, 4648)
Crashdump written to /home/apolyakov/Downloads/d09638da-cec5-4ff7-e1c0e1bc-8082bfff.dmp
[1]    16632 segmentation fault (core dumped)  ./Gyroflow-linux64.AppImage
```

AdrianEddy, 2023-10-09:
The crash happens in libcuda.so. Do you have latest GPU drivers installed?
Try setting `export MDK_DECODERS="VAAPI:sw_fallback=1"` and let me know if that helps
or `export MDK_DECODERS="FFmpeg"`

in any case, it's outside of Gyroflow's code

andpol, 2023-10-09:
Looks like on Arch libcuda comes from the nvidia-utils package, which I have 535.113.01-2, which is indeed the latest version: https://www.nvidia.com/Download/driverResults.aspx/211711/en-us/

`MDK_DECODERS="VAAPI:sw_fallback=1"` results in the video not loading in the few minutes I waited (just spinning a spinning icon)

`MDK_DECODERS="FFmpeg"` seems to have done the trick though! At least I was able to open >20 video files with no crashes so far.

With both the CUDA and FFmpeg decoders, I noticed that audio playback within Gyroflow is crackly/distorted, but the export itself is fine, so not a problem for me personally :shrug: 

I really appropriate the support and awesome project!

closed 2023-10-09 by andpol