gyroflow/gyroflow #741: Small simple feature: ignore trim markers

eried, 2023-10-11:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hi! Fantastic project! could you add a small checkbox in advanced to ignore the trim markers? 

I find the add to render queue feature awesome but working with the markers it is a bit cumbersome. So the easiest solution would be an option to ignore them so I can add new ones easily :)

AdrianEddy, 2023-10-11:
I don't understand the purpose, could you explain it in more detail? 
If you don't want the trim markers, simply don't set them

eried, 2023-10-12:
hi, thanks! example:
-------------- video ---------------

and I want to queue 2 pieces
-[-------]---- video ---------------

and
--------------video -[-------]------

yes? so if I set the first segment, I cannot scroll to the start of the second and set the beginning of the second easily. Maybe I am missing something? maybe just a button to remove the trim markers is enough

AdrianEddy, 2023-10-12:
Done in latest dev build https://gyroflow.xyz/devbuild/
"C" keyboard shortcut or "Clear trim range" in context menu

closed 2023-10-12 by AdrianEddy

eried, 2023-10-12:
oh awesome!!!!! thanks 👍