gyroflow/gyroflow #743: Gyroflow crashes on file export | Intel UHD Graphics 630

snekras0v, 2023-10-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.3

### What operating system are you using?

Windows 10

### What GPU are you using?

Intel(R) UHD Graphics 630

### What happened?

Gyroflow crashes after pressing export button. 
Intel latest driver 31.0.101.2115 is installed.
Quick Sync is supported on Intel(R) UHD Graphics 630.

Tried to switch to [wgpu] - didn't help.
[image]
CPU only options doesnt work either.

Export works only with "Use GPU encoding" unchecked.

logs are attached:
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/12908709/gyroflow.log)

AdrianEddy, 2023-10-15:
Please try latest dev build https://gyroflow.xyz/devbuild/

snekras0v, 2023-10-18:
[image]
it works now with a good fps rate

Adrian thanks for such a quick answer!

AdrianEddy, 2023-10-18:
What if you change to [wgpu] (Vulkan) in `Device for video processing`?

AdrianEddy, 2023-10-18:
Please attach gyroflow.log file

uavquadfpv, 2023-10-18:
> Please attach gyroflow.log file

Back to v1.5.2, when exporting files, the blue screen does not appear. Only version 1.5.3 shows all files with errors displayed in blue on the output video. The same thing happened to my computer when I updated to v1.5.3

AdrianEddy, 2023-10-18:
Please send me gyroflow.log file from latest dev build, I'll need to see why it happens

uavquadfpv, 2023-10-18:
> What if you change to [wgpu] (Vulkan) in `Device for video processing`?

Screenshot I just took.

[image]

AdrianEddy, 2023-10-18:
ok but I need to see the log file after rendering

snekras0v, 2023-10-18:
please find the log attached
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/12983241/gyroflow.log)

if i change to anything but [wgpu] (Vulkan) the FPS decreases to 2

AdrianEddy, 2023-10-18:
Should be fixed in latest dev build https://gyroflow.xyz/devbuild/

snekras0v, 2023-10-18:
it's exporting correct image, but FPS is only 4
is it supposed to be such low?

[image]

AdrianEddy, 2023-10-18:
Is there a speed difference between v1.5.2 and this build?
Try also OpenCL and wgpu options. Please also attach the log file after rendering I'll check if everything is as expected

snekras0v, 2023-10-18:
actually it is the first build i've tried. also didn't find a link to download 1.5.2 as only 1.5.3 is now available.

but still log is attached
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/13025121/gyroflow.log)

AdrianEddy, 2023-10-18:
v1.5.2 you can download here https://github.com/gyroflow/gyroflow/releases/tag/v1.5.2

Everything appears fine in the log. Intel HD 630 is not the newest so it's possible it takes this slow. You can use H264/AVC instead of HEVC, should be faster

snekras0v, 2023-10-19:
it is 0.7 FPS in 1.5.2 )

Thanks for your help!

AdrianEddy, 2023-10-19:
Thanks for testing!

closed 2023-10-19 by AdrianEddy