gyroflow/gyroflow #744: Gyroflow ios blue screen error on iPhone 6s - iPhone X

uavquadfpv, 2023-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

V1.5.3

### What operating system are you using?

ios 15.7.9

### What GPU are you using?

[wgpu] Apple A10 GPU (Metal)

### What happened?

Gyroflow ios blue screen error on iPhone 6s - iPhone X
It first appeared when I installed it on all my existing devices, which are those running iOS 15 or later. I don't know if everyone has the same blue screen phenomenon as me. Hopefully the next version will fix errors on older Apple devices using the iOS 15 operating system.

https://github.com/gyroflow/gyroflow/assets/148299671/cf1c8556-334f-45ad-b304-4278f4583b0d

### Relevant log output

```shell
20210101000723_000003_stabilized_2.mp4
```

AdrianEddy, 2023-10-18:
This is only in rendered file? Does the preview in app look good?

uavquadfpv, 2023-10-18:
> This is only in rendered file? Does the preview in app look good?

The preview in gyroflow ios is beautiful, very smooth.

uavquadfpv, 2023-10-18:
> This is only in rendered file? Does the preview in app look good?

[image]

Screenshot I just took.

AdrianEddy, 2023-10-19:
Please check in TestFlight version https://testflight.apple.com/join/e0RI1n5i

uavquadfpv, 2023-10-20:
> Please check in TestFlight version https://testflight.apple.com/join/e0RI1n5i

Very good! It works very smoothly and quickly on my iphone 7 device. 7.5 fps, awesome. This version will officially go into operation soon.

Thanks you so much!

[image]
[image]

closed 2023-10-20 by AdrianEddy