gyroflow/gyroflow #746: Auto sync not saving when creating settings preset

bradMatias, 2023-10-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.3

### What operating system are you using?

Windows 10

### What GPU are you using?

Nvidia RTX 2070

### What happened?

I'm trying to create a settings preset with Auto sync enabled, but after I create the preset and try to drag it back in it disables the setting.
[image]
[image]

Video of issue occurring:
https://github.com/gyroflow/gyroflow/assets/3682725/cc745bf3-eb79-44b6-a569-c2403e788b5e

AdrianEddy, 2023-10-23:
Autosync is not done if you already have any sync points. The checkbox you're pointing to is not autosync, but a different autosync mode.

Delete your sync points and drag the preset again

bradMatias, 2023-10-23:
In this example I was editing a single clip, but I have a thousand or so videos I wanted to add to a render queue and apply auto sync to.  When I load all of the videos into the render queue then drag in my settings preset then right click edit any video in the list it also doesn't have the auto sync tickbox checked.  Is this expected?  Sorry I'm also confused what is the tickbox for?

AdrianEddy, 2023-10-23:
Autosync in render queue is ran just before the rendering of the particular file. Add the video file to render queue, drag & drop the preset and click render - it should do autosync and render. Let me know if that works

bradMatias, 2023-10-23:
Thanks for the quick reply!

it appears to have done some kind of autosync because when I right click edit any video in the render queue it has sync points on the timeline, but the tickbox is still disabled.  The tooltip says: "Experimental automatic sync point selection." and since I didn't manually select any sync points I was confused as I'm not sure how the sync points appeared if the box that says it will automatically find sync points is unticked?

sorry I might be missing something obvious, thanks for the help.

AdrianEddy, 2023-10-23:
that checkbox doesn't matter much in your case. If it's checked then a different method of sync point selection will be used. If it's unchecked, a default method will be used. Autosync is done by the "Autosync", regarless of that checkbox.

If you have "Do autosync" in the preset, the autosync will be done in the render queue. Drop all your video files to the render queue and then drop the preset. It will correctly sync and render all your videos

bradMatias, 2023-10-23:
Ah okay I see: `If it's checked then a different method of sync point selection will be used. If it's unchecked, a default method will be used.` 

this wasn't clear to me from the tooltip as I thought it meant the entire auto selection of sync points was experimental.  Cool looks good and my videos seem to have all been stabilized correctly, should we keep an issue for the experimental setting not being saved in the preset? or is this not important

Thanks again for your help here!

AdrianEddy, 2023-10-24:
indeed there was a typo related to that checkbox, fixed now in 653e68a466b23a985f21b25941d5f6ed75261262

closed 2023-10-24 by AdrianEddy