gyroflow/gyroflow #748: pixel format yuv420p10le is not supported by the hevc_vaapi encoder Supported pixel formats: vaapi

sanj3k, 2023-10-25:
Hi, quick question somehow it was impossible for me to use the qsv device there was only wgpu and opencl intel hd 630 available. Was it some permission issues or what does the qsv device look like? Ubuntu 22.04 and 1.5.3 rel. with newest nightly the app can be executed but will never start somehow it hung up at I dont know why.

Path to Log: https://pastebin.com/ZPKM2w3g

AdrianEddy, 2023-10-25:
The render device selection is under Export settings -> Advanced -> `Device for rendering`
The `Device for video processing` is for processing (stablizing), not rendering

sanj3k, 2023-10-25:
But whats the name of the qsv device or how do I select to use that one? because in the logs I can see " Failed setup for format cuda: hwaccel initialisation returned error" When i never wanted to even use cuda. Unfortunately its not inside this log but i also tried opencl and wgpu in the rendering box and i had the same errors.

AdrianEddy, 2023-10-25:
There should be your intel device to select in `Device for rendering`. Is it not there? How does the list look like?

sanj3k, 2023-10-26:
`10:13:51 [INFO] GPU list: ["[OpenCL] NVIDIA CUDA NVIDIA GeForce 940MX: 3.0", "[OpenCL] Intel(R) OpenCL Graphics Intel(R) HD Graphics 630: 3.0", "[wgpu] Intel(R) HD Graphics 630 (KBL GT2) (Vulkan)", "[wgpu] llvmpipe (LLVM 15.0.7, 256 bits) (Vulkan)", "[wgpu] NVIDIA GeForce 940MX (Vulkan)", "[wgpu] NVIDIA GeForce 940MX/PCIe/SSE2 (Gl)"] `

It was either opencl Intel hd 630 or without any prefix nor suffix  in the render option but i assume this was the Vulkan

AdrianEddy, 2023-10-26:
No I mean in the UI in Export settings sectionw

sanj3k, 2023-10-26:
That's what it said too i had two Intel options, two Nvidia and one llvmpipe option. 

I have added output of vainfo to the log file in the end. Does this already look bad? Because there it says VAProfileHEVCMain10 is supported. By the way at the moment I don't have access to the notebook because i don't want to buy and regret when no hardware acceleration is available. 

So my question is if and how the qsv device would look like. I had both drivers installed but have heard that only the ihd driver works with qsv and not the i965. Also in the logs it says hevc qsv is available but it was never selected once that's why I anticipate i may haven't set up something properly but the question is what and how 

`10:14:33 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["vdpau", "cuda", "vaapi", "qsv", "drm"]
10:14:33 [DEBUG] (13) mdk: [MDK] VA-API error@782. vaPutSurface(display_, surface, pix , 0, 0, w, h , 0, 0, w, h , nullptr, 0, 0x00000000 | 0x00000020): 0xffffffff unknown libva error
10:14:33 [DEBUG] (13) mdk: [MDK] VA-API error@1018. va::vaCreateConfig(display_, VAProfileNone, VAEntrypointVideoProc, nullptr, 0, &proc_cfg_): 0xc the requested VAProfile is not supported
10:14:33 [DEBUG] (2) gyroflow_core::filesystem: FfmpegPathWrapper::new file:///home/test/GX011158.MP4, write: false
10:14:33 [DEBUG] (2) gyroflow_core::filesystem: url_to_path | url: file:///home/test/GX011158.MP4 -> Ok("/home/test/GX011158.MP4")
10:14:33 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: None
10:14:33 [DEBUG] (2) gyroflow::rendering: Possible encoders with Intel: [("hevc_vaapi", true), ("hevc_qsv", true), ("hevc_v4l2m2m", true), ("libx265", false)]
10:14:33 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_VAAPI 0, for: hevc_vaapi
10:14:33 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: HW formats: [VAAPI]
10:14:33 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: SW formats: [BGRA, ARGB, RGBA, ABGR, BGRZ, ZRGB, RGBZ, ZBGR, X2RGB10LE, GRAY8, NV12, YUYV422, UYVY422, YUV420P, YUV411P, YUV422P, YUV440P, YUV444P, P010LE]
10:14:33 [DEBUG] (2) gyroflow::rendering: video_codec: Some("hevc_vaapi"), processing_order: PreConversion
10:14:33 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 205800, render_frame_count: 10290`

AdrianEddy, 2023-10-26:
But what you're showing me is `Device for video processing ` and `Device for rendering` is a different thing. Show me a screenshot from this list in the UI

sanj3k, 2023-10-26:
Okay understood on it. Is there something else I can also try to provide like does vainfo state somewhere something resolution related or i don't know. Since i don't have the Notebook it would be a better to have most out of it when I get in contact with the person again

AdrianEddy, 2023-10-26:
In general hardware accelerated 10-bit encoding is a relatively new thing, so it's not very likely it will be supported by this old hardware. You'll have to check in the intel specs though

sanj3k, 2023-10-26:
According to the Quick Sync Video Wikipedia it says it's possible.

AdrianEddy, 2023-10-27:
From the `vainfo` you provided, it appears to support 10-bit decoding, but not 10-bit encoding (`VAEntrypointEncSliceLP` is for encoding)
You can force to convert to an 8-bit format (`-pix_fmt yuv420p` in encoder options) or encode 10-bit using CPU

AdrianEddy, 2023-10-31:
It doesn't look like I can do anything about it in Gyroflow, so I'm closing this issue. If you can confirm that you can in fact render 10-bit with VAAPI (eg by using ffmpeg) then I can reopen. But until then, I believe your hardware can't encode 10-bit, and you have to use CPU encoding

closed 2023-10-31 by AdrianEddy