gyroflow/gyroflow #753: Parse gyro metadata from mp4 files recorded on Android

UnlimitedIllusionsSoftware, 2023-11-07:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Is that possible to add support for https://developer.android.com/reference/android/media/MediaMuxer#metadata-track?
Media muxer writes data as described above (big-endian) to the "mett" box.
Example of the video:
https://drive.google.com/file/d/1BXm69h1aewEBCmbXnVQM822vfJcGE0Ar/view?usp=sharing

AdrianEddy, 2023-11-07:
Interesting, what app and phone was used to record this? I don't think this is standard in android default camera apps?

UnlimitedIllusionsSoftware, 2023-11-08:
It's third party app https://play.google.com/store/apps/details?id=com.software.illusions.unlimited.filmit, Samsung Galaxy Note 10 Lite.

AdrianEddy, 2023-11-16:
Added in latest dev build https://gyroflow.xyz/devbuild/ but this file doesn't stabilize well for me.
Very important is to have accurate lens profile, and since I don't have that phone, you have to create a lens profile
Also very important is to disable phone OIS is possible. If the OIS is enabled, Gyroflow won't be able to stabilize that video

Please calibrate an accurate lens profile and send me a new test file and that lens profile (and make sure it doesn't have OIS)

UnlimitedIllusionsSoftware, 2023-11-16:
Thank you so much, I will try to do that.

AdrianEddy, 2023-11-20:
Nice, so it works correctly

closed 2023-11-20 by AdrianEddy